iNavFlight/inav #8787: increase nav_extra_arming_safety options

sdellava, 2023-02-10:
## Current Behavior
Setting **nav_extra_arming_safety = off** allow to arm without restrictions:
- Throttle can be higher than 0 
- GPS can't be fix, so home point is not defined.

In some cases it is useful to arm even without the GPS fix, but is normally very dangerous to arm if the throttle is not under the idle level: motor/motors jump immediately to the throttle level and if it is very high, the effects can be very dangerous.

## Desired Behavior
Allow to selectively disable Throttle and GPS check.

## Suggested Solution
Add **nogps** and "nothrottle" to the nav_extra_arming_safety options to disable only the GPS check.

## Who does this impact? Who is this for?
Every INAV users

## Additional context
The new options have to be selectable via programming system too.

breadoven, 2023-02-10:
`nav_extra_arming_safety `only affects Nav related things, i.e. no GPS fix or Nav modes selected before arming. It doesn't bypass throttle position, or at least it shouldn't. You should never be able to arm with the throttle raised.

Also `nav_extra_arming_safety` OFF option has been removed from INAV 6.0.

sdellava, 2023-02-11:
Well, so there must be a bug. The follwing program logic allow to arm without GPS Fix and with Throttle in any position:

[image]

MrD-RC, 2023-02-11:
I'm guessing you are showing the **override arming safety** from the programming tab. The image did not work. This is not the same as nav_extra_arming_safety. It will arm in any circumstance, so should be used with extreme caution. You should certainly only use it with logic checks to make sure that it can only operate in certain conditions.

sdellava, 2023-02-11:
I've updated the image. So, my suggestion can be changed in splitting the "Override Arming Safety" in more detailed condition or simply add an "Override GPS Fix" condition, to allow ariming without GPS dix, but keep all other safety conditions

MrD-RC, 2023-02-11:
> I've updated the image. So, my suggestion can be changed in splitting the "Override Arming Safety" in more detailed condition or simply add an "Override GPS Fix" condition, to allow ariming without GPS dix, but keep all other safety conditions

Why? The `Override Arming Safety` is doing what it is supposed to do. What are you trying to achieve here, with this programming?

sdellava, 2023-02-11:
The goal is to program a switch on the tx to enable arm even if the GPS did not fix yet.

I've added a lc on the chan 4 that is true only if the throttle is low and this avoid the risk of arming at hight throttle, but it is not exactly the same of aving a "override gps fix" lc: FC still print a system message "wainting for GPS fix" and this is a little confusing.

MrD-RC, 2023-02-11:
Why not just use **set nav_extra_arming_safety = ALLOW_BYPASS**. On the rare occasions when you would want to arm without a GPS lock, yaw right when arming. That will do what you want. There's no need to use the programming tab at all for that.

sdellava, 2023-02-11:
Thank you for the suggestion. I will try it out today.

Actually, at least for me, it happens often that I have to arm without GPS. If you fly in the mountains taking off from a narrow valley the fix can take a long time. By taking off and simply gaining an altitude by staying vertical (using landmarks so as not to lose orientation), the GPS is able to receive many more satellites or at least better signals and thus is able to make the fix. At that point I land and arm normally to fix the home point.

Since it is an aprocedure that solves a specific issue I had tried programming a switch on the tx also to have clear evidenta (tactile) that I am in that condition.

closed 2023-02-14 by DzikuVx