iNavFlight/inav #8795: IMPORTANT: Motor keeps spinning, even after disarming

Tommy032017, 2023-02-13:
Controller: Matek F411 Wing
ESC: ZTW 30A, but problem also exists within other ESC's
Motor: A2212 6T

Yesterday, I was flying my pretty big plane at the model club and after full throttle, the motor keeps spinning in full throttle. No way to cut off the motor, even disarming did not power down the ESC and motor. I had to fly about 15 minutes with full throttle until the ESC cuts off the motor at 3.3V (luckily I had programmed the ESC to cut off the motor at 3.3V) and there was enough power for the receiver to do a safe landing.

Reproduce:
INAV 6
- Install (with full chip erase) [6.0.0-RC2](https://github.com/iNavFlight/inav/tree/6.0.0-RC2) or RC1
- Setting ESC to "standard" and ESC refresh rate to "50Hz".
- Do a throttle calibration in the ESC
- Start "Motor Test Mod" in INAV

**Motor seems to run normal within throttle up to 50%
After throttle up to 100% for 1-2 seconds, the motor keeps spinning full throttle, until you unplug battery or data cable of the ESC.**

INAV 3
- Install (with full chip erase) https://github.com/iNavFlight/inav/releases/download/3.0.1/inav_3.0.1_MATEKF411.hex
- Setting ESC to "standard" and ESC refresh rate to "50Hz".
- Do a throttle calibration in the ESC
- Start "Motor Test Mod" in INAV

Motor and ESC works completely normally and no problems at all.  
Setting "set motor_pwm_rate = 50" -> everything is perfect and no problems at all
Setting "set motor_pwm_rate = 400" -> same problems, described above within INAV 6

Conclusion:

Even if setting "set motor_pwm_rate = 50" in INAV 6, the motor keeps spinning in full throttle. 
There must be something different in handling pwm rate, compared to INAV 3.0.1. PWM-Rate seems to be incorrect.

INAV 3.0.1
Setting "set motor_pwm_rate = 50" everything perfect, no problems.
Setting "set motor_pwm_rate = 400" same problem within INAV 6

 See also attached files of settings
[INAV_6.0.0_cli.txt](https://github.com/iNavFlight/inav/files/10719848/INAV_6.0.0_cli.txt)
[INAV_3.0.1_cli.txt](https://github.com/iNavFlight/inav/files/10719849/INAV_3.0.1_cli.txt)
 for INAV 6 and INAV 3

breadoven, 2023-02-13:
A log file would be most useful here.

Tommy032017, 2023-02-13:
I did a logging in house, without a GPS lock and arming with allow_bypass. 
What I did:
- arming with stick allow_bypass
- push the throttle up to about 25%, then lower and higher a little bit -> everything seems to be ok
- increase throttle to about 50% and a little higher, lower again, but motor (ESC) did not respond anymore and motor is spinning with constant speed. 
- disarm, but motor is still spinning
- disconnected battery from FC

Pls. see attached logging. 
[inav_data_log_2023-02-13_222636.txt](https://github.com/iNavFlight/inav/files/10726513/inav_data_log_2023-02-13_222636.txt)

breadoven, 2023-02-14:
Really need the raw Blackbox log file itself.

Tommy032017, 2023-02-14:
Sorry, but the F411 has no SD-Card, UART2 is GPS, UART1 is RX Receiver. So I could not record a log.
But most interesting is, that with inav 3.0.1 and also betaflight (betaflight_4.4.0_STM32F411.hex), there are no problems at all.
Installing inav 6 RC1 or RC2, the problem occurs.
I tested it with my three identical F411 Wing boards and different ESCs and it is always repeatable. The wiring is exactly as described from Matek: ESC gets main power +/- from the F411 within the current meter and communication is GND and S1, without +5V (only G and S1).    
If the motor (ESC) spins without control (throttle about more than 50%), there is no more control over the FC and you you cannot stop the motor anymore, only if you restart the FC or disconnect the data cable from the ESC (or programmed power cut off of the ESC stops the motor). 
There must be something different in PWM handling for S1/S2, compared to 3.0.1 and betaflight 4.4.0.
Thank you.

breadoven, 2023-02-14:
OK looking at that log it does show the RC motor signal and Mixer output to the motor drops to 1000 but the Amps remain stuck at 2.7 odd Amps. So something appears awry between the mixer output and the signal sent to the ESC. Flagged as a bug.

Tommy032017, 2023-02-14:
Yes, exactly, the Amps remain stuck at 2.7A and this is the actual amperage the motor uses without prop at his constant speed.
Maybe it could help to find the cause, if you regard inav 3.0.1: There you can set ESC protocol and ESC refresh rate. If you set the refresh rate to 50Hz, everything will be fine and no problems at all. But if you set ESC refresh rate to 400Hz (not sure, if this is the same as CLI "motor_pwm_rate "), the behavior is exactly the same as the problem in inav6: Motor keeps spinning, even if Mixer output is set to 1000, or disarm. Maybe ESC refresh rate is not properly set to 50Hz in inav6. I also tried to use "set motor_pwm_rate = 50" manually in CLI, but without any improvement.

breadoven, 2023-02-14:
Looks like the code was changed middle of last year to make `motor_pwm_rate `only applicable to brushed motors. The pwm rate for Standard motors is now hard coded at 400Hz ... which seems to cause problems in your case. @DzikuVx ?

Tommy032017, 2023-02-14:
Ok, this explains the odd behavior. But I see it very critical to set the refresh rate hard coded at 400Hz, because it could end in a potentially dangerous situation. Most standard receivers uses 50Hz PWM and most ESC do not have a declaration about the pwm frequency. So testing everything in prior for any flight (of course without props!) is self-evident.
So my suggestion is to bring back the setting of motor_pwm_rate soft coded (50Hz or 400Hz, or even adjustable) and also implement it to the Configurator, as was done in inav3. 
In the documentation the values for motor_pwm_rate are now only for analog (brushed) motors. That confuse and does not consider, what happened to ESC refresh rate.  There should be also a command to change a "esc_refresh_rate".
I also would do a pop up and remark with some more information to this parameter: "If motor do not turn or even motor will not stop anymore.... set the correct ESC frequency"
Thank you again for helping and finding the cause.
Would be perfect, if there will be an implementation in next release candidate.

DzikuVx, 2023-02-14:
All modern ESC are able to work with 400Hz refresh rates. If one does not work, then I suggest downgrading INAV to older release to keep compatibility with some egacy ESC. 
But, as reported, happens with different ESCs, so doubt that is really the case. We also know that there were no other reports of such a behavior. 
And that puts this case in the gray area of "unknown" cause I'm afraid

DzikuVx, 2023-02-14:
On a different note, the attached blackbox log is damaged, I'm unable to open it.
The CSV file however is strange.
1 - timestamp is broken
2 - If ESC fails to understand 400Hz rate, then why it understands it when throttle is pushed up? ESC if rate is unknown do not start the operation at all.
3 - Why CSV file has 17 RC channels when INAV writes only 4?

In general, this is the first case anybody reported this kind of a problem. Was anybody else able to reproduce it? Or this is the only case.
On top of that, this change was already introduced so long ago that I honestly double is a cause of this issue. Might be wrong, but well.... mi spider senses are tingling that there is something else going on

Tommy032017, 2023-02-15:
At the moment I would say, that there is no problem within INAV. It seems Inav uses a fixed coded 400Hz ESC refresh rate for brushless motors. 

I now tried some new ordered ESCs and tested them without any problems!

**BUT, with some different ESCs, all from ZTW, the described problems still exists.** 

If the ZTWs are only for lower frequencies, the shouldn't initialize with a wrong (400Hz) frequency - but they do!
You can test them with no faults in lower throttle but if you increase up to full throttle, they won't stop anymore and keep running until power is cut off.
May be the attached video could help and ensure that it never happens to anyone.

Thank you. 

https://user-images.githubusercontent.com/26543724/218971708-1abad40f-edfe-4451-bbe7-7bd0e0bfb7cc.mp4

closed 2023-02-20 by DzikuVx