iNavFlight/inav #8798: WP "user actions" breaks waypoint navigation

ABLomas, 2023-02-13:
Short video demonstrating issue:
https://user-images.githubusercontent.com/8989151/218552591-65620b35-67e2-451c-ae52-334931ab7444.mp4
Text version - mission waypoint with user action set has it's altitude set to MSL+preset altitude, while AGL setting is selected.
This is not configurator issue - the same is on CLI:

```
# wp
#wp 4 valid
wp 0 1 548752733 255081618 5000 0 0 0 0
wp 1 1 548839392 254988921 5000 0 0 0 0
wp 2 1 548860868 255146849 5000 0 0 0 0
wp 3 1 548780634 255179036 18400 0 0 3 165

# ...modify altitude:
wp 3 1 548780634 255179036 5000 0 0 3 165

#wp 4 valid
wp 0 1 548752733 255081618 5000 0 0 0 0
wp 1 1 548839392 254988921 5000 0 0 0 0
wp 2 1 548860868 255146849 5000 0 0 0 0
wp 3 1 548780634 255179036 5000 0 0 3 165

save
# wp
#wp 4 valid
wp 0 1 548752733 255081618 5000 0 0 0 0
wp 1 1 548839392 254988921 5000 0 0 0 0
wp 2 1 548860868 255146849 5000 0 0 0 0
wp 3 1 548780634 255179036 18400 0 0 3 165
```

It's also unavoidable probably by setting MSL altitude flag, so in fact target altitude can't be set (or i did not found suitable workaround).

Tested - in reality airplane climbs a lot (as shown in configurator) before performing user action (while it should stay at the same level as previous waypoints).

ABLomas, 2023-02-13:
Oh, there's more - i changed one bit (MSL -> 0):
```
# wp
#wp 4 valid
wp 0 1 548752733 255081618 5000 0 0 0 0
wp 1 1 548839392 254988921 5000 0 0 0 0
wp 2 1 548860868 255146849 5000 0 0 0 0
wp 3 1 548780634 255179036 5000 0 0 2 165
```
And now CLI shows correct alt (?), but planner still shows the same 184m (which is preset alt + ground level over sea level). Maybe this issue is actually in configurator, not firmware...

stronnag, 2023-02-13:
I'm sort of missing the point here. The use of P3 is defined in the wiki https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages#p3-bitfield-usage.  So based on terrain database:

P3 =3  (AMSL)

[image]

and P3 ==2 (relative to home)

[image]

I believe the firmware will behave as the illustrations and terrain analysis (I guessed a home based on 130m AMSL).

The difference of 4m is due to accuracy or otherwise of terrain DB.

ABLomas, 2023-02-13:
Yes, i believe now it's a problem with INAV-configurator (which sets and reads bit0 incorrectly) and not firmware itself.
I will close this one and re-open new in other branch.

closed 2023-02-13 by ABLomas

breadoven, 2023-02-13:
Configurator hasn't been updated to use a Bitfield for P3. It just reads 0 or not 0 to set the MSL option, Any user action would need to be added manually by editing the P3 value taking into account the required MSL setting for Bit 0.