iNavFlight/inav #8801: High throttle in automatic modes

tiriad, 2023-02-14:
## Current Behavior
Automatic nav modes throttle well above `nav_fw_cruise_thr` setting e.g. Cruise.
Automatic modes with ascending throttle up even more e.g. RTH.

## Steps to Reproduce
1. 
`fw_level_pitch_trim` =  8
`nav_fw_max_thr` = 1500
`nav_fw_pitch2thr` = 3
`nav_fw_cruise_thr` = 1300
`nav_fw_allow_manual_thr_increase` = ON
`nav_fw_climb_angle` =20
`max_throttle` = 1600
2.
WP mode
[image]
Cruise mode. Straight line, level without touching the sticks and at the minimum. Throttle well above `nav_fw_cruise_thr` 

3.
RTH mode
[image]
Throttle above `nav_fw_max_thr` in OSD, and throttle far above  nav_fw_cruise_throttle - (nav_fw_pitch2thr * pitch_angl. Also Pitch angle > `nav_fw_climb_angle`

## Expected behavior
Take the value of the variable `nav_fw_cruise_thr` at `Cruise` and avoid excessive acceleration on ascents.
Limit the value of  Autothrottle on OSD at `nav_fw_max_thr` 
Limit the value of Pitch to `nav_fw_climb_angle`

- FC Board name and vendor: OmnibusF4V3
- INAV version string: v6.0RC1

breadoven, 2023-02-15:
Why do you think the throttle is too high ?

For the CRUZ case, 45% throttle is correct for a `nav_fw_cruise_thr `of 1300 if you also have `max_throttle `set at a low value of 1600.

tiriad, 2023-02-15:
> Why do you think the throttle is too high ?
> 
> For the CRUZ case, 45% throttle is correct for a `nav_fw_cruise_thr `of 1300 if you also have `max_throttle `set at a low value of 1600.

[image]

In angle mode a 28% throttle has a consumption of 3,19 amps. 
I guess 1300 is equivalent to 30% throttle and with my configuration to Cruz and this is not the case. The throttle I want in Cruz is the same as 30% in Angle and in climb with RTH a value of 35% with `nav_fw_pitch2thr` = 3.
I do not want more consumption. 
What are the steps to do that?

Regarding the pitch value I can't find an explanation for its value to be greater than `nav_fw_climb_angle` unless `fw_level_pitch_trim` is not being applied correctly and affect more values.

Jetrell, 2023-02-16:
Have you tried increasing `nav_fw_pitch2thr_threshold`, to widen the pitch2thr deadband, and reduce auto throttle hunting ?

Or have you tried a test. By setting `fw_level_pitch_trim = 0 ` and adjusting the FC board angle, by using `align_board_pitch`? 
To ascertain if auto throttle is building upon a combination of `nav_fw_level_pitch_trim` + `nav_fw_pitch2thr`? If that's what your implying.
 
`fw_level_pitch_trim = 8°` is a rather extreme value... If their is a bug. It may have been over looked. Due to most people only using `fw_level_pitch_trim` for slight board offset correction.

Can I assume these settings were only for your tests?
`nav_fw_pitch2thr = 3` (_3 = 0.3% throttle change per degree of pitch change.  Equaling only 6.6% increase at 20° climb_angle_)
`nav_fw_max_thr = 1500`

Because if not, its a disaster waiting to happen, if you run into strong turbulent wings in the mountains. 
Its one thing to chase efficiency. But another to endanger your model, if it needs the extra thrust and propulsion velocity to fight turbulence, if you failsafe.

tiriad, 2023-02-16:
> Have you tried increasing `nav_fw_pitch2thr_threshold`, to widen the pitch2thr deadband, and reduce auto throttle hunting ?

I could do it if I only had the problem in level mode but also in climb.
 
> Or have you tried a test. By setting `fw_level_pitch_trim = 0 ` and adjusting the FC board angle, by using `align_board_pitch`? To ascertain if auto throttle is building upon a combination of `nav_fw_level_pitch_trim` + `nav_fw_pitch2thr`? If that's what your implying.

I used `align_board_pitch` option when `nav_fw_level_pitch_trim` was not available and was implemented for that purpose. And I don't remember when it was like that I had these problems.
I have only realised that my pitch in climb corresponds to `fw_level_pitch_trim` + `nav_fw_climb_angle`

> `fw_level_pitch_trim = 8°` is a rather extreme value... If their is a bug. It may have been over looked. Due to most people only using `fw_level_pitch_trim` for slight board offset correction.

It migh be possible

> Can I assume these settings were only for your tests? `nav_fw_pitch2thr = 3` (_3 = 0.3% throttle change per degree of pitch change. Equaling only 6.6% increase at 20° climb_angle_) `nav_fw_max_thr = 1500`

In principal it is for test and efficiency but I am trying this because I have a 2200kv motor with a 7 inch propeller and `nav_fw_max_thr = 1500` is the limit of my esc.
> Because if not, its a disaster waiting to happen, if you run into strong turbulent winds in the mountains. Its one thing to chase efficiency. But another to endanger your model, if it needs the extra thrust and propulsion velocity to fight turbulence, if you failsafe.

I should change the engine maybe but with this workaround I could do it as well.

Why is the throttle re-scaled in automatic mode to max_throttle?
If I select `nav_fw_cruise_thr=1300` I want 30% throttle, `nav_fw_max_thr=1500` 50% throttle, `max_throttle=1600` 60% throttle so I don't burn my esc upper this value.

MrD-RC, 2023-02-16:
> Why is the throttle re-scaled in automatic mode to max_throttle?
> 
> If I select `nav_fw_cruise_thr=1300` I want 30% throttle, `nav_fw_max_thr=1500` 50% throttle, `max_throttle=1600` 60% throttle so I don't burn my esc upper this value.
> 

I totally agree with this. I'm not a fan of how INAV "scales" the throttle at the moment. All scaling should be in the output only. At the moment some seems to scaled, others transposed, and others just capped. 
> 
>

breadoven, 2023-02-16:
> Why is the throttle re-scaled in automatic mode to max_throttle? If I select `nav_fw_cruise_thr=1300` I want 30% throttle, `nav_fw_max_thr=1500` 50% throttle, `max_throttle=1600` 60% throttle so I don't burn my esc upper this value.

It isn't rescaled only in Auto mode, this applies always including in ACRO mode. The purpose of the throttle display is to show the percentage of the **useable** throttle range. It's not intended to show the percentage of the max possible pwm range, i.e. 1000 to 2000us, because this wouldn't be particularly useful in a lot of cases since it would rarely ever reach 100% which would make the use of percentage a bit pointless.

The throttle percentage should probably also reflect the stick position, i.e. 50% throttle should be roughly mid stick. In your case you should have rescaled the transmitter throttle channel to output a max of say 1700 (1600 + some margin) at max stick otherwise most of the stick movement above mid stick will be dead movement that has no affect on the throttle. (I realise you're not supposed to do this but in this case with the max throttle so restricted it doesn't make sense using the normal 1000 - 2000 range).

Also, why are there 2 OSD throttle options ? They both show the same percentage scaled throttle value regardless with the only difference being the symbol shown before the numbers. Don't see the point of this. The combined Throttle/Auto Throttle option should work in all cases surely. If you are going to have another throttle option maybe it would be more useful for it to just show the raw RC output value in us (or a percentage based on 1000 - 2000us).

MrD-RC, 2023-02-17:
@breadoven I'd agree that there's no need for the current two throttle elements. Throttle/Auto Throttle is the main one. However, rather than RC Throttle for the second. I'd rather see unscaled throttle. Whether that's RC or nav commanded.

The issue is what is being asked by the pilots is not what they are expecting to see. And what they expect to see depends on their preference and how well they understand the systems in the back end. Some may be happy with how it is now. But I'd argue that most are confused by it. This issue is a prime example of that. 
   - If the throttle is limited to 50%, why is is going above 50%?
   - If the cruise throttle is set to 1300 (30%), why is it cruising ag 45% throttle?

These are just two examples of why the current throttle display doesn't work for most people. It actually makes things harder to set up too. Because if you're flying along at 50% throttle, see the amp draw is nice, it's a decent speed, so decide to make this the cruise throttle. When you enter 1500 for the cruise throttle, when you enable the Cruise mode, you will see 52%.

breadoven, 2023-02-17:
> @breadoven I'd agree that there's no need for the current two throttle elements. Throttle/Auto Throttle is the main one. However, rather than RC Throttle for the second. I'd rather see unscaled [...]

Well the other way of looking at it I guess is that the settings are confusing, i.e. using pwm microseconds in all cases rather than throttle percentage where possible (actually `throttle_idle ` is the one setting that does use percentage). `nav_fw_max_thr `would always need to be in microseconds , but `nav_fw_cruise_thr `could be a percentage so it matches the OSD throttle display. Not sure how practical this would be to code but the idea makes sense. Same with the Nav Launch throttle settings,

Either way the base OSD Throttle display should be reconsidered unless someone can explain its propose.

It did occur to me that one solution to this specific issue is to limit the throttle output from the transmitter to 1600, set `nav_fw_max_thr `= 2000 and `nav_fw_max_thr `= 1600. Haven't tried this but wouldn't this scale the OSD throttle display to 100% at 2000us whilst limiting the max throttle command to the ESC to 1600 in manual and Auto modes ?

MrD-RC, 2023-02-17:
I agree that the settings should be changed to percentages where possible. But, changing the settings to percentage alone does not stop the issue, because of the scaling. 

It also doesn't stop the confusion of the throttle being shown above the maximum throttle.

I still think that scaling should only be applied in the output stage. Though I can see the reasoning behind some wanting to se a scaled throttle on the OSD. I think it should be made clearer though. 

As for the non-auto throttle in the OSD now. The only reason I can think of having it is so that you know what the stick is set to before exiting an auto throttle flight mode. Though feel should get someone close enough to where they want the throttle to be.

breadoven, 2023-02-17:
> As for the non-auto throttle in the OSD now. The only reason I can think of having it is so that you know what the stick is set to before exiting an auto throttle flight mode. Though feel should get someone close enough to where they want the throttle to be.

The throttle percentage shown is the same for both OSD Throttle options, only the symbol changes. Nothing in `getThrottlePercent `changes between Auto and manual throttle control, it's just based on `rcCommand[THROTTLE]`.

https://github.com/iNavFlight/inav/blob/9c32094f5ec469bcb387a7c0cd7559fde9eda1d3/src/main/io/osd.c#L1079

MrD-RC, 2023-02-17:
That is completely pointless then. I always believed it to work as I described.

MrD-RC, 2023-02-17:
It would be easy to get a non-scaled throttle though. Just a Boolean parameter added to the `getThrottlePercent()` method and handle it in there.

breadoven, 2023-02-17:
Looks like the OSD throttle values were different:
https://github.com/iNavFlight/inav/pull/6162

But then got changed to what exists today:
https://github.com/iNavFlight/inav/pull/6510

Not sure why the different calculation method for Auto and Manual throttle was removed in #6510.

At least it confirms having the 2 current OSD Throttle options is pointless.

MrD-RC, 2023-02-17:
Nice to know I wasn't going completely mad too.

tiriad, 2023-02-17:
Thank you for your work and for coming to these conclusions, I found something strange in the behaviour. In order to configure the nav_fw_*_thr settings it is necessary to unscale  values as the configuration inputs are 1000-2000 uS and to find the custom values it is necessary to see them in the osd in manual flights.

MrD-RC, 2023-02-17:
I need to do a bit of testing. But the above PR should provide two different working throttle displays.

RomanLut, 2023-04-28:
PR does not solve original problem of user confusion.

Display scaled throttle percentage value next to Cruise throttle input in configurator; then in would be clear.

[1600     us]  Cruise throttle (56%)

Ideally it would be better to replace Cruise throttle input with percentage input, which under the hood converts from us to percenage and and vice versa to keep backward compatibility.

MrD-RC, 2023-04-28:
Ideally nothing should be scaled until it is output. 

There should be no scaling in the mixer or by navigation at all. Throttle should be treated as 1000 to 2000 until it is output to the ESC(s) or the OSD. At the point of output it can be scaled if required or requested. Then there would be no issues at all. The whole throttle setup currently is a mess.

0crap, 2023-05-12:
> Ideally nothing should be scaled until it is output.
>
> There should be no scaling in the mixer or by navigation at all. Throttle should be treated as 1000 to 2000 until it is output to the ESC(s) or [...]

Totally agreed!

Until recently this mess was not so visible.
But now we even get crap related to this in the OSD:
[image]
Who ever thinks that was a good idea?

Probably because I added only one line to my diff.
`set max_throttle = 2000`
This should be the default like everything else in the world sees it, except INAV.
Sjeezzz, I hope INAV fixes this properly one day...
</rant>

MrD-RC, 2023-05-12:
@0crap actually the scaled throttle in INAV 6 is what you would have seen in older versions of INAV; regardless of what element you chose. The reworked unscaled OSD element now shows pure throttle input until the auto throttle engages.

0crap, 2023-05-12:
> @0crap actually the scaled throttle in INAV 6 is what you would have seen in older versions of INAV; regardless of what element you chose. The reworked unscaled OSD element now shows pure throttle input until the auto throttle engages.

I'm into INAV since v2.5.1 and never seen that weird "I" sign in front of my OSD throttle icon.
Just more clutter that adds totally nothing.
The "S" with the percentage is good enough.
(As it always has been shown on INAV v5 and lower.)

MrD-RC, 2023-05-12:
@0crap the scale symbol is to show that your are using a scaled throttle OSD element. The other OSD throttle element is not scaled. There needs to be a symbol to distinguish between the two throttle type. Now that they do actually display different data.

0crap, 2023-05-12:
@MrD-RC "There needs to be a symbol to distinguish between the two throttle type."
This is the problem in a developers mind, it does NOT need a different symbol.
What does the pilot care if he is using scaled or unscaled throttle? None!
Can he now magically fly it better?
99.9% doesn't even know the difference.

Keep it simple, stick at 0%, OSD element 0%, stick at 100% OSD element at 100%.
All this scaled unscaled whatever is not on the pilots mind.
Probably good intentions by development, but really..... it's under the hood tech stuff.

So like you said a few posts above, it's a mess. We agree on that.
On this OSD stuff we probably agree to disagree.

MrD-RC, 2023-05-12:
Actually the there have been two throttles for the OSD in INAV for a long time. A PR in 3.0 broke them, so that they both acted the same. This only showed the scaled throttle. There were plenty of people who hated this because the cruise or launch throttle on the OSD would not match the value they set in configurator. They asked for a throttle that showed the proper value. This is why the change came about. So there are once again 2 different throttles available for the OSD. If people want to have both throttles visible, for example on a tuning OSD. They need to be able to tell the difference. 

I agree, this is not the best solution. But it was the best that could be done in the time available. And a better solution will be worked on for future releases. Oh, this idiot developer who actually added a throttle that tracked with the input stick, not a scaled value was me. The scale symbol is one of the least invasive symbols available, being a pixel wide for the most part. Like I said. With how the current system works. It was the best that could be done in the time available. For something that pilots have been asking for, for a while. But, in future releases, where will hopefully not be a need for stuff like this.

rts18, 2023-05-13:
@MrD-RC  I was also wondering about the "I" symbol next to the throttle. 
`I get your reasoning. And I'm grateful for the throttle scaling fix.`
But that scaling symbol looks messed up.
Even a full stop would have looked less invasive. 

Thanks for making us aware that both throttle selections display auto. I'll use the other one for now.

closed 2024-03-22 by b14ckyy