iNavFlight/inav #8803: JETIEXBUS implementation does not work with 24 channels

coretrekthomas, 2023-02-16:
## Current Behavior
JETIEXBUS works fine when setup to transfer 16 channels (default)
JETIEXBUS stops working when 24-channel multimode is activated (see screendump).
-- Telemetry still works
-- Control signal stops working

## Steps to Reproduce
1. Turn ON "24-channel multimode" in the Jeti transmitter under the menu: Advanced Properties/Wireless Modes/Trainer
2. JETIEXBUS control signal stops working

3. Turn OFF "24-channel multimode" in the Jeti transmitter under the menu: Advanced Properties/Wireless Modes/Trainer
4. JETIEXBUS control signal starts working

## Suggested solution(s)
Since iNAV only supports 16 channels internally we suggest that iNAV could receive 24 channels from, EXBUS, but discard channels 17-24 instead of stopping working completely alternatively support 24 channels internally;-)

We have complex setups where we need all 24 channels sent to the Jeti Receiver, channels 17-24 are used as PWM outputs to control retracts, lights, brakes, etc on complex models with 24 functions, but the first 16 channels are sent to iNav and iNav is used as normal with those 16 channels.

- FC Board name and vendor:
Mateksys H743 WING
- INAV version string:
6.0.0RC2

[image]

egilrove, 2023-02-16:
Jeti has sold 24 channel transmitters for many years, so this is quite common.

DzikuVx, 2023-02-19:
INAV internally work with 16 channels, so this is expected

closed 2023-02-19 by DzikuVx

coretrekthomas, 2023-02-20:
I understand that INAV internally works with 16 channels - and that is totally fine.

What I ask is if INAV could ignore the channels 17-24 from EXBUS instead of crashing - then it would solve the problem.

Thanks.