iNavFlight/inav #8807: set ahrs_inertia_comp_method = ADAPTIVE  results in error that wont go away

iflyforfood, 2023-02-17:
## Current Behavior

After updating to V6 RC2 from 5.1, and running the 5.1 CLI through the tool per the V6 RC2 release notes, the CLI keeps giving me a "set ahrs_inertia_comp_method = ADAPTIVE ERROR: Invalid name message". 

It's the only error that wont go away

I reflashed twice via a speedybee dongle. This FC refuses to flash via my PC.

## Steps to Reproduce

Import 5.1 diff file to v6 CLI after running it through tool.

## Expected behavior

set ahrs_inertia_comp_method = ADAPTIVE
No errors

## Suggested solution(s)

?
## Additional context

https://pastebin.com/FL4a5MsU
---

INAV/ATOMRCF405NAVI 6.0.0 Oct 20 2022 / 14:00:23 (d1f6f19a)
GCC-10.2.1 20201103 (release)

closed 2023-02-20 by iflyforfood

MrD-RC, 2023-05-30:
There was no parameter for ** ahrs_inertia_comp_method** in 5.1. It was introduced in 6.0. This parameter works fine in 6.0.

cmanley, 2023-05-30:
Thanks Darren. I deleted my comment after I realized I mixed up imu and ahrs but you replied so fast.

cmanley, 2023-05-30:
B.t.w. I noticed that pitch/roll values in the configurator are very erratic with my Matek F405 WMN board powered only from USB, but when powering from a battery too, then they stabilize. That was driving me nuts trying to figure out what was wrong (recalibrating, rechecking every imu/ahrs setting I could find, etc). It may be worth mentioning in the docs somewhere.