iNavFlight/inav #8814: H7 support

wdliu356, 2023-02-21:
I'm using Lumenier H7 as the flight controller and it can be recognized as Matek H743 by INAV. All the sensors are recognized well but the angle mode is not working. I tested that as follows:
1. Arm the drone.
2. Roll and pitch the angle by hand
It turned out that the angle mode is not trying to level the drone. Is there any possible reason why this is not working? 
By the way, the drone can be manually controlled to roll and pitch with a remote controller.

b14ckyy, 2023-02-23:
what does the 3D model in the Setup tap show when you move the drone? is it upright? does it move correctly the same way as the physical drone?

wdliu356, 2023-02-23:
The 3D model moves correctly in the same way as the physical drone, which is the main reason why it makes me super confused. I tried INAV 4.1 for MATEKH but I still have the same issue. For reference, I tried other firmware such as BetaFlight and Ardupilot. For BetaFlight, it's working super well but the issue seems to be happening with Ardupilot as well(still testing with Ardupilot).

b14ckyy, 2023-02-23:
what exactly happens if you enable ANGLE mode?

wdliu356, 2023-02-23:
I armed the drone and turned on angle mode. Now the drone is on the ground horizontally. Then I use my hand to rotate the drone (roll and pitch) without giving any other input from the controller. I expected the drone to give some response trying to get itself to level again but nothing happened. I think this is just showing the angle mode is not working well right?

MrD-RC, 2023-02-23:
Airmode will not activate until the throttle had been raised above certain level. Until then, the motors won't respond.

wdliu356, 2023-02-23:
Are you talking about angle mode?

MrD-RC, 2023-02-23:
Anything on a quad uses stabilisation. For that to happen at zero throttle, airmode has to be active.

Contrary to what BetaFlight will have you believe, airmode is not a flight mode. It is a modifier that sits on top of flight modes. When BetaFlight pilots say they're flying in "airmode", they're actually using acro with airmode enabled on top. It doesn't help that the BetaFlight OSD shows "AIR".

Until you raise the throttle above the threshold (was 1350, but I'm sure it has been lowered). Airmode will not be active, so you will not see any stabilisation at low throttle.

wdliu356, 2023-02-23:
Oh ok, sounds reasonable, I'll do some tests to see whether it works.

closed 2023-03-01 by DzikuVx