iNavFlight/inav #8830: Wind estimator. OSD doesn't display unit character sometimes.

Jetrell, 2023-02-27:
I've experienced issues with the `wind estimator display`.
It now displays an asterisk in place of the units symbol, when no value is present... But I've had flights when the Unit `km/h` doesn't appear at all... Even after much course changing, to make the estimator update. 
This change was added to the master mid January, before RC1.

I'm not sure what this OSD change was about. I found it hard to decipher what it was trying to convey. Because it doesn't line up with the lost data method, used on all other OSD elements. When hyphens are displayed in place of Bad, None or Stale data.
[image]

- FC Board name and vendor: Matek F765-WSE
- INAV version string: 6.0.0 RC2

MrD-RC, 2023-02-27:
The asterisk is when the wind data is stale. It would be better if it cycled between the units and the asterisk every couple of seconds.

MrD-RC, 2023-02-27:
Just to confirm. Are you saying there are times when you see **🚩---***

Jetrell, 2023-02-27:
Once the plane was thrown, it sometimes appear as - e.g.  `24.7 km/h`.. Which is what it should do.
But in one flight it would display the wind speed with the asterisks - e.g `24.7*`..  Even when the estimator appears to be updating the wind speed, after making turns.

Jetrell, 2023-02-27:
Thanks Darren.
I tested it again today. And having them flash back and forth between the asterisk and the speed unit, was more intuitive. 
I didn't experience the issue of the asterisk remaining either, once the GPS heading and wind speed was attained... It showed the km/h every time in today's tests.
Maybe it was a timing issue, that the character cycling fixed... Who knows.