iNavFlight/inav #8835: Mag align helper

ltwin8, 2023-02-28:
## Current Behavior
graphical mag align helper is not very useful, leaves room for misinterpretation, does not state what the default gps orientation is (cw270flip like m8q5338??) where is the arrow and so on.

## Desired Behavior
implement a mode like the various auto tune mode on fixed wings, instruct pilot to fly northways for eg. 100m STRAIGHT, instruct pilot to fly east fo 100m STRAIGHT and then assign the axis that has icremented accordingly to the fitting axis X Y and Z

## Suggested Solution
implement a solution described above that is only required to be used once, maybe with an (uncalibrated mag) and then snap to the closest 45deg (im not that good at linear algebra but i think it should be possible to calculate magnitude of changing vector and its phase)

## Who does this impact? Who is this for?
everyone using GPS

## Additional context
I fly a small craft with FC arrow facing to front, M8Q-5883 facing to back and thus the setup should be CW270FLIP as per manual but substract 180deg, so CW90FLIP? however it is sadly not the case

Jetrell, 2023-03-01:
When `Default` is selected under the drop down tab. It takes the value to `CW270FLIP`.. And will then move the compass image to that selected position.
I agree, that it would be more intuitive for the GPS to have a direction arrow... But as it is. The PLUG at the rear of the GPS/Mag, is in the opposite direction to which the arrow points. Just like the SAM M8Q+5883.

[image]

I personally find the greatest thing lacking in this tool... Is its inability to take into account the FC board alignment.
I have many quads that have their flight controllers mounted with the arrow facing in all directions, and even up side down (due to the USB plug or motor/ESC header plug).  This really makes the magnetometer alignment far more complicated for new users. Because the Magnetometer is referenced to the FC's IMU.. Not the multirotor or fixedwings air frame.

@kernel-machine Any possibility of an update one day?

ltwin8, 2023-03-02:
I used it exactly with that assumption however it does not work at all correct.If inav code had at least some documentation I would do it myself, but I do not get why it is so hard to add a mode to help the user, even to ask user to go outside with laptop or phone, go into open field, point quad nose to north, confirm, prompt to point to East and so on…Facing N E S W separate the full orientation of the mag can be obtained…

kernel-machine, 2023-03-02:
> @kernel-machine Any possibility of an update one day?
Not soon, probably in the future but i don't know