iNavFlight/inav #8836: iNAV 6 iFlight Chimere7 V2 Pro problem with throttle

DemoFPV, 2023-02-28:
Hi,

I have a problem with lunching of new Chimera7 V2 Pro. Look like all modes, GPS, Mag are OK. But when I try to lunch the drone just shot up like crazy. A cannot control throttle from remote. Throttle channel from receiver work OK, motor test also is OK. Spin from 7% and smooth to 100%. On Beteflight 4.4.0 all work OK.

breadoven, 2023-02-28:
Can you post a Blackbox log please.

DemoFPV, 2023-02-28:
Now I can have from iNAV 5 because I try to check also with this version.
[https://drive.google.com/file/d/1NHtziEOLX1IXN0qvHNgKhlopWj5XyZPv/view?usp=sharing](url)

breadoven, 2023-03-01:
Not sure what that second log is supposed to be showing (first log is empty). There don't seem to be any props attached ? Throttle just follows the throttle stick movement. You need to describe the problem better in more detail.

DemoFPV, 2023-04-02:
Hello again, No I install stable iNAV 6 and still the same problem.
Look like I have problem with props??

[https://drive.google.com/file/d/1987Wha86u_Qub7-teAkH60-ea_xSn9sh/view?usp=share_link](url)