iNavFlight/inav #8838: HUD home position constraint

rts18, 2023-03-01:
I was testing @MrD-RC's new ESP32 radar updates. Really nice work !!
When I ventured out to test the rest of the HUD features as shown here. 
[image]

I stumbled across one that is not found in the GUI. Only the CLI. And doesn't require any of the above to be selected.
`osd_hud_homepoint = ON`
It allows the home point to be displayed on the HUD, with the distance from home being displayed below it.
[image]

Now for the problem. It isn't constrained by `osd_hud_margin_h` as the other radar HUD functions are. It will move all the way to the edges of the screen and merge with the characters position there.
It only appears to function if the crosshair is activated. Which is disappointing. Because it would be optimal if it could function independently of the ESP32 radar. So both can be used on alternate OSD screens and not clutter up the same screen.

b14ckyy, 2024-03-22:
Moving to Configurator repo.

b14ckyy, 2024-03-22:
OK this covers basically 3 topics. 

- I agree that its time that we have HUD settings in the OSD setup tab. Will look into that (maybe you can open that as a short request in the configurator repo so we can keep track of that)
- the HUD margin seems to be a bug. Will add the label. 
- The crosshair dependency is there because the crosshair is the reference point. I wonder if the HUD elements could be made part of the OSD layout settings in the future to change them dynamically. But for that we use the linked ticket.

rts18, 2024-03-23:
>  The crosshair dependency is there because the crosshair is the reference point. I wonder if the HUD elements could be made part of the OSD layout settings in the future to change them dynamically. But for that we use the linked ticket.

@b14ckyy 
@mmosca Attempted to resolve this issue with the release of FormationFlight. 
This would allow us to place FormationFlight on Alt1, HUD on Alt2 and Home radar on Alt3 OSD. Without them all clashing together on every OSD screen where a crosshair is placed.
However, it doesn't appear to have moved forward in a while. It would be very useful indeed.

https://github.com/iNavFlight/inav-configurator/pull/1812
https://github.com/iNavFlight/inav/pull/9199

mmosca, 2024-03-24:
I may revisit this in 8.0

Right now we are in the stage where we are talking about our grievances with the configurator and what we could do to improve it, if motivation doesn't tank when we touch the javascript code, there may be some room for changing this

rts18, 2024-03-26:
@mmosca Thank you