iNavFlight/inav #8839: Unstable altitude in position hold

akurekin, 2023-03-01:
HI, I have got an issue with altitude hold with my 4inch quad.
I am using MAMBA F405 mini 4th edition flight controller and BrotherHobby 1504.5 motors.
When I switch to PosHold mode the quad starts drifting up and down, sometimes jumps up by several centimeters. I need to use the throttle to keep it at the same height. Please help. The configuration and flight logs are enclosed.

Thanks
 
[blackbox_log_2023-02-28_215953.TXT](https://github.com/iNavFlight/inav/files/10861968/blackbox_log_2023-02-28_215953.TXT)
[INAV_5.1.0_cli_20230131_194400.txt](https://github.com/iNavFlight/inav/files/10861976/INAV_5.1.0_cli_20230131_194400.txt)

Jetrell, 2023-03-01:
I have the same flight controller... And its not just this FC.. All my small quads have there times when the altitude holding ability will wander, or even jump within a meter.

Its mostly caused by poor initial baro calibration. Or a poor GPS, `EPH/EPV` (HDOP)... _Provided your `nav_mc_hover_thr` is set correctly._
I find its best to have a short hover flight.. Then land and disconnect the battery and reboot the FC.. The precision is generally better every other after, in the same session..  Although, if you get a day with low satellite coverage, its never going to be as good as a day when you have a high Sat coverage.

If you're interested to compare. These are my Profile setting on a 3.5" quad. Using the Mamba F405 mini MK4.
```
# profile
profile 1

set mc_p_pitch = 39
set mc_i_pitch = 50
set mc_d_pitch = 16
set mc_cd_pitch = 70
set mc_p_roll = 36
set mc_i_roll = 46
set mc_d_roll = 18
set mc_cd_roll = 95
set mc_p_yaw = 40
set mc_i_yaw = 55
set mc_cd_yaw = 35
set mc_p_level = 24
set mc_d_level = 65
set dterm_lpf_hz = 150
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 370
set yaw_lpf_hz = 10
set heading_hold_rate_limit = 60
set nav_mc_pos_z_p = 55
set nav_mc_vel_z_p = 95
set nav_mc_vel_z_i = 46
set nav_mc_vel_z_d = 15
set nav_mc_pos_xy_p = 60
set nav_mc_vel_xy_p = 41
set nav_mc_vel_xy_i = 16
set nav_mc_vel_xy_d = 97
set nav_mc_heading_p = 55
set mc_iterm_relax_cutoff = 5
set d_boost_min =  0.900
set d_boost_max =  1.500
set d_boost_max_at_acceleration =  4700.000
set d_boost_gyro_delta_lpf_hz = 62
set antigravity_gain =  2.200
set antigravity_accelerator =  5.000
set tpa_rate = 20
set tpa_breakpoint = 1650
set rc_expo = 35
set roll_rate = 64
set pitch_rate = 58
set yaw_rate = 60
set fpv_mix_degrees = 20
```

akurekin, 2023-03-02:
Hi, thank you very much for your reply that inspired me to do more experiments. From the log file it does not look like there is a barometer temperature drift. Also, I do not think that GPS has any effect on altitude hold as the altitude is controlled by barometer + accelerometer measurements. The accuracy of GPS is too poor for this task. 
After some experimenting I managed to get an appropriate altitude lock/accuracy by changing the config parameter from inav_w_z_baro_p = 0.35 by default to inav_w_z_baro_p = 1.6 . This improved the altitude stability, it may drift by plus minus half a meter but now it stopped drifting uncontrollably up and down. This parameter adds more weight to the baro measurements and less weight, I presume, to the accelerometer measurements. I can only guess that in my case I have got too much noise in accelerometer data. It is not clearly seen in the logs but the problem with altitude instability has developed after I crashed my drone. The crash probably affected the bearings and added vibrations. I think that INAV is too sensitive to the vibrations when in altitude hold mode. My drone remains rock solid in ACRO mode, even at strong wind conditions. 
I have also checked the log file and found large spikes at the spectrum plot at around 50Hz, 100Hz, 150Hz, etc. frequencies in mcPosAxisP(2) and mcVelAxisP(2) measurements. But there are no spikes in the spectrum of  acc(2) data. Can this be explained?

Thanks.

Jetrell, 2023-03-03:
>I do not think that GPS has any effect on altitude hold as the altitude is controlled by barometer + accelerometer measurements. The accuracy of GPS is too poor for this task.

`inav_w_z_gps_v` is used by the sensor fusion algorithms to assist `inav_w_z_baro_v` and `inav_w_z_baro_p` for vertical velocity and position control. 
GPS altitude data can make the altitude control smoother than just relying too heavily on the barometer data.
For this same reason, its not advisable to set `inav_w_z_baro_p` too high... This is why sensor fusion exists. GPS altitude may not provide the outright accuracy of a barometer... But its a second opinion for the navigation altitude controller to work with.

>But there are no spikes in the spectrum of acc(2) data. Can this be explained?

My reasoning is that gyro Z axis vibrations (noise) will also effect the Z axis accelerometer element... So raw gyro log data should also give an example of the noise the vertical velocity computations have to deal with.

>I think that INAV is too sensitive to the vibrations when in altitude hold mode. My drone remains rock solid in ACRO mode, even at strong wind conditions.

Acro mode is only using the Gyro/Acc to control attitude.. While the navigation controller is using all the sensors to provide precise 3D position control.

akurekin, 2023-03-03:
"North and East are fused from accelerometer and GPS data, while Up is fused from accelerometer + barometer for multicopters ..." ( https://github.com/iNavFlight/inav/wiki/INAV-blackbox-variables )  - it looks like that GPS data are not used for altitude hold in drones. 

I have also checked gyro measurements from blackbox logs and I did not find any spikes in the spectrum in any axes.

akurekin, 2023-03-04:
If my assumption is right and the problem is due to vibration, I would like to ask the community what is the acceptable level of vibration for INAV? If the vibration affects the flight so seriously, I would like to have automatic vibration level check performed by INAV before the flight and in flight. It should not be so difficult to implement.

wawanlost, 2023-03-05:
> HI, I have got an issue with altitude hold with my 4inch quad. I am using MAMBA F405 mini 4th edition flight controller and BrotherHobby 1504.5 motors. When I switch to PosHold mode the quad starts [...]

have you covered the baro with foam, if it gets wind directly from the propeller, has the throttle hold been set correctly? you can also check filtering ..! As far as I know GPS doesn't affect altitude spikes...

akurekin, 2023-03-20:
I think, I managed to solve the puzzle! The problem was with quad motors that were damaged during the recent crash. I am not sure if the shaft was bent or bearings failed but after I replaced the motors with new ones everything went back to normal - stable altitude hold and position hold. This is a very important result as the vibrations at my quad also caused toilet bowl effect and now it does not take place. This effect is usually explained  by bad magnetometer calibration or the battery wires placed too close to the GPS unit, but it turned out that it can also be caused by vibrations. I have looked at accelerometer data before and after motor replacement and could not notice any difference. It is possibly because the vibrations were produced at the frequencies that are too high for accelerometer recordings, but sufficient to add a small bias to acceleration measurements. 

Thanks for your help.

akurekin, 2023-03-21:
Thanks to everyone who replied and left comments. The problem has been resolved.

closed 2023-03-21 by akurekin