iNavFlight/inav #8840: DJI O3 air unit - OSD battery voltage

RobinatiNav, 2023-03-02:
Hi, would it be possible to manage somehow to see OSD battery voltage like in the video? I see 0,0V now on DJI goggles 2. Of course I can choose to see bat voltage in iNav OSD but I would like to see it using dji OSD. It can be done in Betaflight (shown in video) and ardupilot, too. Maybe it is possible in iNav but I do not know how.

https://www.youtube.com/watch?v=KIis4hB_duQ&t=2s
[image]

[image]

RobinatiNav, 2023-03-02:
Thank you Marcelo :)

mmosca, 2023-03-03:
It will make it to 6.0 after all. 

Should be included in the next rc.

closed 2023-03-03 by mmosca

RobinatiNav, 2023-03-03:
Well done Marcelo :)

RobinatiNav, 2023-03-06:
Hi Marcelo, it works! Many thanks. You maybe do not know but you fixed 3 things in one. Not only voltage is present in native dji OSD but arming command passes through, so when armed O3 air unit changes its state from low power to full power, and also auto recording works when armed, too. Good job :)