iNavFlight/inav #8844: INAV6 RC2 : USER1 without function on MAMBAH743_2022B

Pairan, 2023-03-02:
## Current Behavior
USER1 mode is not working in TARGET MAMBAH743_B2022 

[image]

## Steps to Reproduce
1. simply attach the vtx to the 9v pin and choose USER1 to be active/inactive
2. Power won't switch off

## Expected behavior
The power should become switchable as described in the manual (FC)

## Suggested solution(s)
enable mode USER1 to control the 9v pin to power a connected VTX/WALKSNAIL/O3 on/off

---

### version
 - INAV/MAMBAH743_2022B 6.0.0 Feb  9 2023 / 11:19:13 (92b237e5)
 - GCC-10.2.1 20201103 (release)

DzikuVx, 2023-03-02:
Are you sure it's `MAMBAH743_B2022` not `MAMBAH743` target? They differ in PINIO assignment and 2022B has a mapping exactly like requested by Diatone

Pairan, 2023-03-02:
It’s the Mamba H743 v2 stack … the website says it like this:

MCU: 480MHz STM32H743
GYRO: DUAL ICM-42688-P
Target：MAMBA H743_2022B

so I would assume it to be the right target.

DzikuVx, 2023-03-02:
Then Diatone must have changed something again... I'll talk to them

DzikuVx, 2023-03-02:
I have samples, but they might be pre-production batch

Pairan, 2023-03-03:
In case you have a test build, I can try it out on my quad

DzikuVx, 2023-03-04:
[inav_6.0.0_MAMBAH743_2022B.zip](https://github.com/iNavFlight/inav/files/10888491/inav_6.0.0_MAMBAH743_2022B.zip)
 @Pairan please test this hex

Pairan, 2023-03-04:
just a sec ...

Pairan, 2023-03-04:
Sadly it didn't work. The Walksnail Airunit wasn't shut down and kept working while enabling/disabling `USER1`.

Pairan, 2023-03-05:
this didn't fix it. Guess we'll have to wait for Diatone ...

The Airunit connected to 9v pad kept on while switching USER1 on/off

DzikuVx, 2023-03-07:
@Pairan I spoke with Diatone and looks like current target is correct. I will retest something

Pairan, 2023-03-07:
Thanks a lot, Pawel - If you need some testing then let me know. Test Rig is currently on RC3

GootyS, 2023-05-09:
Does iNAV support the dual gyro Mamba H743? Not having much luck with BF an this board.

closed 2023-10-27 by DzikuVx

marcdornan, 2023-11-07:
This is not working on Mamba H743 V2 (4th Gen) on iNav 6.1.1 using MAMBAH743_2022B target, It seems to be non functional as Pairan reported on March 2023. I tested it by flashing Betaflight and it worked fine: 9V is applied to the VTX power pin when User 1 is set up on the modes tab and a channel is mapped.

ioakimov, 2023-12-05:
As mentioned in #9455 
USER2 (instead of USER1) is switching power on VTX.