iNavFlight/inav #8847: Remark in CLI.md necessary for better understanding

hannes-m, 2023-03-03:
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****
Current dokumentation in CLI.md:

set | Change setting with name=value or blank or * for list

requested change:

set | Change setting with name=value or blank or * for list REMARK: NOT ALL OF THE VARIABLES ARE SHOWN WITH THE WILDCARD - USE "GET" INSTEAD

(This is the necessary change according to your last answer)
(Using Version 6.0)

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

DzikuVx, 2023-03-03:
Actually, docs nowhere says that `set` allows to list any config variables. `set`, like the name implies sets a values. Not list variables

closed 2023-03-03 by DzikuVx

hannes-m, 2023-03-03:
Hi Pawel,
here is the screenshot of the original Inav doc showing that the listing of variables is possible with "set":

[image]

stronnag, 2023-03-03:
The firmware behaves exactly as the documentation:

`set` lists the variables
`set *`  lists the variables
`set foo=bar` sets variable `foo` to value `bar`

hannes-m, 2023-03-03:
Hi Jonathan,
Thx for your reply...
...for better understanding: I wanted to point out that "Set *" is not showing all the possible variables...e.g. all the "align_board_(roll, pitch, yaw) are not listed.

stronnag, 2023-03-03:
There are here .... thusly:
```
$ cliterm 2>&1 > /tmp/vars.txt
set
exit
$
```
then
```
$ grep align /tmp//vars.txt 
align_opflow = CW0FLIP
align_mag = CW270FLIP
align_mag_roll = 0
align_mag_pitch = 0
align_mag_yaw = 0
align_board_roll = -25
align_board_pitch = 40
align_board_yaw = 0
osd_temp_label_align = LEFT
osd_switch_indicators_align_left = ON
$ 
```

hannes-m, 2023-03-03:
Different response:
Configurator: 6.0.0
Firmware: 6.0.0 [MATEKH743]

CLI command: set *
Result:
[image]

MrD-RC, 2023-03-03:
@hannes-m that is just a partial screen grab. These are the results from `set *` on 6.0 [set star.txt](https://github.com/iNavFlight/inav/files/10880188/set.star.txt). Lines 106-108 has the `board_align` stuff. It is working.

[image]

Though I do think using `set` to perform a `get` is a bit backwards. @stronnag is that residual nonsense from CleanFlight?

stronnag, 2023-03-03:
It is probably an historical artefact. Seems mainly harmless to me.
The documentation is accurate, the FC works as documented; the only issues  demonstrated are PEBCAK, which is beyond the development team's remit.

hannes-m, 2023-03-03:
THX a lot...
sorry that I used up so much of your time...
I copied it to notepad++ and searched for it...yes it is listed...
(used to ardupilot and missionplanner I have to synchronize myself to the CLI).