iNavFlight/inav #8848: RTH by magnetometer in case of GPS failure

DangerD1024, 2023-03-03:
Is there any way to add RTH that will turn fixed wing to the home direction and just fly to it, than pilot will take control once it will fly closet.
Yes plane can drift away due to some wind, but that's okay

Jetrell, 2023-03-04:
@DangerD1024  https://github.com/iNavFlight/inav/pull/8347

wawanlost, 2023-03-05:
> Apakah ada cara untuk menambahkan RTH yang akan memutar sayap tetap ke arah rumah dan terbang ke sana, daripada pilot akan mengambil kendali setelah terbang dari lemari. Ya pesawat bisa hanyut karena angin, tapi tidak apa-apa

Sounds like the first GPS rescue.... this has been implemented on the Bayangtoys X16 Altitude Hold non-GPS

DangerD1024, 2023-03-05:
> @DangerD1024 #8347

That's even better))

closed 2023-03-05 by DangerD1024