iNavFlight/inav #8868: Aocodarc f7 mini

Ultraexpert, 2023-03-09:
Inav 6.0rc3

Neither the v1 or v2 firmware work. No motors or servos will work. I have tried all the settings in the mixer on both versions. For some reason my esc gets so hot after I try to test the motor. 
Please fix this problem, there is working 4.1 firmware only and it's not officially supported.
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Ultraexpert, 2023-03-09:
So after playing with settings I could get motor on s1 work ok. Servo on s5 work ok but sometimes will freeze at maximum. If I apply vtail template the servos on s6 and s8 will randomly engage. None of the servos will sync with gyro .

Ultraexpert, 2023-03-09:
I am using on an airplane. With 3 servos and a motor

The code that works on the 4.1 inav is as follows

BUSDEV_REGISTER_SPI_TAG(busdev_mpu6500,     DEVHW_MPU6500,      MPU6500_SPI_BUS,    MPU6500_CS_PIN,     MPU6500_EXTI_PIN,       0,  DEVFLAGS_NONE,  IMU_MPU6500_ALIGN);

const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM3, CH1, PB4,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S1   UP1-2   D(1, 4, 5)
    DEF_TIM(TIM3, CH2, PB5,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S2   UP1-2   D(1, 5, 5)
    DEF_TIM(TIM3, CH3, PB0,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S3   UP1-2   D(1, 7, 5)
    DEF_TIM(TIM3, CH4, PB1,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S4   UP1-2   D(1, 2, 5)

#if (defined(AOCODAF722MINI_8MOTOR))
    DEF_TIM(TIM4, CH1, PB6,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S5   UP1-6   D(1, 0, 2)
    DEF_TIM(TIM2, CH2, PB3,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S6   UP1-7   D(1, 6, 3)
    DEF_TIM(TIM2, CH1, PA15,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S7   UP1-7   D(1, 5, 3) - clash with S2
    DEF_TIM(TIM4, CH2, PB7,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S8   UP1-6   D(1, 3, 2)
#else
    DEF_TIM(TIM2, CH1, PA15,  TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR, 0, 0),   // S7   UP1-7   D(1, 5, 3) - clash with S2
    DEF_TIM(TIM4, CH2, PB7,   TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR, 0, 0),   // S8   UP1-6   D(1, 3, 2)
#endif

    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_LED, 0, 2),                            // LED          D(2, 6, 0)

    DEF_TIM(TIM9, CH2, PA3,  TIM_USE_PPM, 0, 0),                            // PPM, RX2

    DEF_TIM(TIM5, CH3, PA2,  TIM_USE_PWM, 0, 0),                            // TX2 & softserial1
};

sensei-hacker, 2023-03-09:
Please don't run three different issues for the same problem at the same time.
It makes it difficult for others to follow.

I'll probably catch you back on Discord, but if not:

Can you please let us know whether your board has 1 SH 1.0 mm plug socket, or two?
That will tell us which hardware you have.  Then we'll know which target you need to flash, and which wiring diagram applies to your board, which will be a good start.

closed 2023-03-10 by Ultraexpert