iNavFlight/inav #8872: ESC Telemetry not working!

samnikos7, 2023-03-10:
I'm using FF F35_LIGHTNING FC with iNav 5.1, Skystars Talon32 Slim40 ESC on a Zohd Dart250 wing using Dshot600 protocol. Everything works perfect except ESC Telemetry (RPM, Temp etc). I've already tried to change the AUTO TELEMETRY (ON-OFF) from BlHeliSuite 32, use another RX UART of the FC with no luck. I don't receive any data (debug, OSD).

MrD-RC, 2023-03-10:
Just to confirm. You have run a wire from the T pad on the ESC to a UART Rx pad, and enabled the ESC Telemetry peripheral on the ports page?

samnikos7, 2023-03-10:
Yes, of course. 
The ESC is sending data, I can read them using BlHeliTv tool and rs232 usb.
[image]

closed 2024-03-22 by b14ckyy