iNavFlight/inav #8881: Turtle Mode on SpeedBeeF405V3 (SpeedyBee Stack ESC 4 in one)

koklimabc, 2023-03-12:
## Current Behavior
Choose To DSHOT600 and save but It ended up to change back to 'MultiShot'

## Desired Behavior
Enable DSHOT600 to choose in tact

## Suggested Solution
At SpeedBeeF405 to be allow to choose pwm_motor_protocol to DSHOT600 to saved in selection and not change to MUltiSHOT after saved.

## Who does this impact? Who is this for?
Unable to choose for Turtle Mode and It is significantly must in UAV.

## Additional context
None

tang520-tang, 2023-03-24:
I have the same problem,have you solved it?If you solve it,can you let me refer to your method

MrD-RC, 2023-03-24:
DSHOT is not supported on that FC. It is documented in both the target outputs wiki and the document for that specific board https://github.com/iNavFlight/inav/blob/master/docs/boards/SPEEDYBEEF405V3.md

closed 2023-03-28 by DzikuVx