iNavFlight/inav #8882: iNav 6.0 RC3 produced worrying dive with fixed wing - after switching from RTH to cruze

helicocrasher, 2023-03-12:
installed the 6.0 RC3 on my nano Goblin plank today.
Was absolutely blown away by the horizon drift elimination, what a treat thanks to all that contributed.

But there was one strange moment when the plane produced a dive in 3d cruze mode, after coming out of RTH climb.
From Blackbox: For 3s the auto-throttle was at 1258ms  (plane requires cruise _thr is 1450) and the elevator full down, generating a sink rate of up to 4.5m/s loosing more than 10m height towards the set target height.   
Something seems weird here.

Blockbox file is attached. 
The sequence is in Log 2/2 and starts at 2:16.666 when switching RTH off and getting out of the climb.

Thanks for looking into that.
 
[blackbox_log_2023-03-12_194137.TXT](https://github.com/iNavFlight/inav-configurator/files/10951991/blackbox_log_2023-03-12_194137.TXT)

MrD-RC, 2023-03-13:
Do you have a DVR recording of this? Also, could you attach your diff (not dump) as a text file.

stronnag, 2023-03-13:
Notable that RTH shows no desire to fly in the direction of home. Didn't we see this before?

breadoven, 2023-03-13:
At 2:16 PosHold mode was selected which correctly set the Target altitude to the current altitude around 52m. Then Cruise was selected with the Target altitude remaining at 52m but the plane decided to dive regardless, to 40m before starting to reduce the dive rate shortly before RTH was selected. So this just ignored the set Target altitude. The same thing happened again when RTH was cancelled with Cruise active again except the descent rate is even higher even though the target altitude seems to be set correctly. The dive seems to reduce when the actual altitude is 10m below the target altitude.

Not obvious why this is happening, needs further checking.

helicocrasher, 2023-03-13:
Thanks guys,

@MrD-RC: attached the "diff all" file. FPV recording to come - short edit session.
@stronnag: RTH is set to climb first so no worry about the missing desire for home.
@breadoven: agreed, it was the first switch off of RTH that scared me, as the plane was rather low and approching houses. The second switch off of RTH I did not analyze yet.  

[NG_iNav6.0.txt](https://github.com/iNavFlight/inav/files/10960448/NG_iNav6.0.txt)

breadoven, 2023-03-13:
I'm thinking this might be a PID problem related to the fact the Altitude PIDS aren't reset when you select AltHold directly after RTH so the I term carries over from RTH. During this RTH climb to 120m `fwAltI `becomes very -ve (is this right given it's climbing ?). This -ve I term is countered by a large +ve P term (large target altitude error given current altitude is only 50m) resulting in a climb pitch command. However, when RTH is switched off and Cruise AltHold becomes active the P term becomes very small whilst the I term holds the previous large -ve value which results in a large pitch down command. Having said all that it's not obvious why this hasn't happened before because it doesn't appear anything has changed regarding AltHold initialization.

One notable thing that happens when AltHold is initialized after RTH is `setupFixedWingAltitudeController` is called. Unfortunately this function contains nothing more than a "TODO" comment. Maybe this is part of the problem which might explain why selecting AltHold sometimes results in a FW abruptly changing altitude. I've noticed this many times although in my case it's never been bad enough to consider it much of a problem (probably similar to the change from RTH to Cruise at 2:44 in the above log).

Not sure why you wouldn't always want to reset altitude PIDS when changing between modes given they may have completely different altitude targets ?

Jetrell, 2023-03-13:
I agree with @breadoven assessment.

@helicocrasher And reason you have lowered `nav_fw_pos_z_p` in both identical profiles?  That setting is responsible for how fast the plane with attempt to compensate for altitude error.
 And being that you have `nav_fw_dive_angle = 25`... That would likely contribute to it diving quickly, when its SLOWLY attempting to convert the error, to the desired altitude set point, after it came off the top of a climb.

helicocrasher, 2023-03-13:
The reasoning of @breadoven is convincing, thank you. 
Resetting I-term when switching the set point of a control loop - I would consider good practice especially whe the control error to the new set point is near zero (just saying from control theory/practice - I am not in the code).

@Jetrel thank you.l  I do not think that I have lowered the **nav_fw_pos_z_p** it is on the default=35 (I just flashed a spare controller).
And yes **nav_fw_dive_angle = 25**, I have increased a long while ago without noticing issues. I guess I will take it back to 15 for less excitement

breadoven, 2023-03-15:
Think this goes all the way back to https://github.com/iNavFlight/inav/issues/322.

This may have fixed the issue on a multirotor via `setupAltitudeController ` but it won't have done anything useful for a fixed wing given `setupFixedWingAltitudeController `is "TODO". In fact it's not definite that this fix would have entirely worked for a multirotor given it didn't reset the Alt I term when perhaps it should have given the RTH target altitude could be totally different to the AlHold target altitude.

helicocrasher, 2023-03-16:
I did some flight testing today, to check this behavior - switching from RTH to cruze when the RTH height is still much higher. The dive reported can be reproduced, somtimes it is a little bit less, but a 15m drop is quite often happening.
iNav 5.1 has this behavior too. So it does not seem to be newly introduced in iNav 6.0.
If there is interest into DVR and black box logs, I am happy to provide

breadoven, 2023-03-16:
The dive happens if you test using the HITL sim. If you also test by switching RTH off before selecting Cruise, the dive doesn't happen because the Altitude controller gets reset when navigation switches to Idle between mode selections.

I've made some changes to the logic used to reset controllers when switching between modes. Need to test to see if they improve things.

helicocrasher, 2023-03-18:
@breadoven cool.
Did another test session. Yes it looks like its flawless when I switch from  RTH during climb to angle mode first and then to cruze.
It could be that when the time in angle mode is too short, that it still misbehaves.
Session did not end well. No logs available before I get the plane out of the tree.

[https://youtu.be/QDIytOVCcxw](url)

helicocrasher, 2023-03-29:
@breadoven and all.
I have managed to get the plane out of the tree and have the blackbox log now attched.

Log nr 1, at time 7:33, something goes wrong. The intention was to switch fast from RTH to angle and then to cruise mode. The log shows a direct switch to cruise.
Then the plane starts to dive.
New obesrvation --> The target height gets decreased several times - which I do not understand. 
From time 7:42 Elavator is pulled back and throttle is at max still the plane does not climb.
At time 7:50 the plane hits the tree with 29m/s.  

Question did the iNAV6.0 release already receive the above mentioned changes.

Thanks`!

[blackbox_log_2023-03-29_191057.TXT](https://github.com/iNavFlight/inav/files/11103804/blackbox_log_2023-03-29_191057.TXT)

breadoven, 2023-03-30:
I had a look at the log you posted @helicocrasher. The reason it keeps reducing the target altitude is because you kept trying to adjust the altitude with the pitch stick (understandable) but didn't maintain the adjustment or release the stick long enough between adjustments to allow the altitude to recover by itself. The problem with altitude RC adjustment in Althold modes is it resets the target altitude to the current altitude (+ some margin in the direction of the desired adjustment) if the target altitude is too far away from the current altitude when the adjustment was attempted. This was the case here because the steep dive after RTH was deselected meant the altitude difference became large enough to cause adjustment to reset to current altitude every time adjustment was attempted but not held. The main problem does seem to be the steep dive which is caused by the RTH climb when the PID I term builds up to a relatively large value that opposes the equally large P term. When RTH was switched to CRUISE the P term suddenly became small (target alt set to current alt so small error) but the I term remained large because the PID controller wasn't reset ... giving a large pitch down command. RC adjustment struggled to overcome this large pitch down command mainly because it took a long time for the large opposing I term to unwind. It did actually start to recover and climb at one point but you pitched up again which reset the target altitude lower. Not entirely sure why the I term does this during the RTH climb, doesn't seem right.

#8896 should fix the above problem because the PID controller would have been reset switching between modes, resetting the I term and pitch command to 0 which should allow RC adjustment to work as expected. This change isn't in 6.0.0 but should be in 6.1.0.

Have to admit I've never thought RC altitude adjustment worked well on a fixed wing. Very sluggish to respond. I'd never rely on it when flying close to the ground, only when well away from it.

helicocrasher, 2023-04-03:
Thank you @breadoven 
for looking into the BB log.
The explanation sounds plausible to me.  
So to repeat my understanding in my words it was two effects. First the I-Term that was not reset and not completely decayed yet plus second continously adjusting the height due to pitch stick pullup input.  

Ok https://github.com/iNavFlight/inav/pull/8896 not in iNav 6.0 good to know, I will (try) to avoid aborting RTH climb until iNav 6.1.

Yes height adjustments with fixed wing felt sketchy at times, especially when loitering in combination with the up and down caused by horizon drift. As the horizon drift is now solved, height adjustments might get more usable also closer to the ground.

helicocrasher, 2023-04-25:
@breadoven 

Just checked iNAV6.1 RC1. 
The transition now works like a charm.
Blackbox data graphics iNAV 6 vs iNAV 6.1 RC1 attached.

Thank you

[image]

[image]

DzikuVx, 2023-04-26:
then let's close, thank you very much!

closed 2023-04-26 by DzikuVx