iNavFlight/inav #8884: speedybee f405 v3, accelerometer, gyro and baro got crazy after several flights

konikav, 2023-03-13:
After rebooting, the position of the copter in space does not work. Sometimes also the altitude rises sharply up to 80 meters, and then goes down. 
The copter did several normal flights at the first day of flying. The next time it launched into space with the throttle stick at more than 20%. The next time it cant take off, it just jerked around. This behavior does not appear immediately, but after several restarts. So far installed INAV 5.1, judging by the configurator everything is fine, but so far there was no opportunity to check in the field.

[https://www.youtube.com/watch?v=3Llf_8Z_Pas](url)

jianwwpro, 2023-04-04:
also have the problem

jianwwpro, 2023-04-04:
try
`set throttle_idle = 8`

throttle idle > 7

konikav, 2023-04-07:
Today we had the same problem on other two fc. Temporary installed 5.1 version, looks like works without issue, but it would great to install version 6

whsMCU, 2023-04-18:
I'm experiencing the same symptoms. ㅠ.ㅠ

konikav, 2023-04-18:
Inav 5.1 also does not get rid of problems. After firmware upgrade to 5.1 it flies fine for a while, but after a few flights (two or three batteries) and a few minor crashes the problems appear again. In this case the horizon line is ok, the copter just spontaneously starts oscillating and trying to fly up on its own, even if I force the stick down to 0. It looks like some kind of accumulated error. Also after the crash I changed props, I thought they were out of balance which made the pd controller go crazy.

whsMCU, 2023-04-18:
I bought one more SPEEDYBEEF405V3 FC for verification, but I'm embarrassed because I have the same symptoms.

I guess I have to choose another FC. ㅠ.ㅠ

When I asked another user with the same FC, he said he didn't have any of those symptoms... so I don't think I'm lucky to pick.

shirase, 2023-04-30:
Try disable blackbox logging.
`feature -BLACKBOX`

konikav, 2023-04-30:
Disabling blackbox did not help me, i tried. So I had to flash betaflight, it works well with this board.

shirase, 2023-04-30:
Try my fork https://disk.yandex.ru/d/XY6MJOjHOMkLpw and enable blackbox and inset sd card.

whsMCU, 2023-05-03:
> Try my fork https://disk.yandex.ru/d/XY6MJOjHOMkLpw and enable blackbox and inset sd card.

We need to check for intermittent abnormalities.

It works well where the SD card has not been automatically recognized.~!

Which part did you modify?

shirase, 2023-05-03:
Speedybee f405 v3 have problem with sd card and OSD. Because it use same SPI but different SPI flags #8733. 
I fixed that problem PR #9028. 
And now if i insert SD cart, gyro works fine, blackbox works fine. But gyro got crazzy without SD card.

shirase, 2023-05-04:
I found problem. SDcard use BUS_SPEED_INITIALIZATION when init. But if no sdcard, bus speed is stayed in INITIALIZATION state. But OSD use same bus and works on INITIALIZATION speed. When OSD writes many data, it make big loop time, and gyro do not works normaly. Need to restore BUS_SPEED_STANDARD after sdcard init.

sgr41, 2023-09-26:
I just got a new F405 V3, actually 2 of them and they both have the same issue as the video. They accl/gyro is either going crazy or is super slow. I have attached a video. It works perfectly fine with betaflight but iNav makes it go crazy.

https://github.com/iNavFlight/inav/assets/12378318/0e9d76f8-b8e4-404b-8b3f-5ca0800fc183

closed 2023-10-27 by DzikuVx