iNavFlight/inav #8889: Stabilised Throttle does not obey arm switch

MrD-RC, 2023-03-16:
## Current Behavior
In some cases, it is necessary to use the new `SERVOS output_type` when setting up a model. This is especially true when setting up a flight controller with limited outputs. Setting the `output_type` to `SERVOS` and then setting a servo mix as Stabilised Throttle would be a perfect solution. The issue is that currently, Stabilised Throttle ignores arming. I have not tested. But this also implies that the safety checks are also ignored.

## Steps to Reproduce
1. Props off!!
2. Set up arm switch and mode
3. Test using standard motor output. Result: motor will not spin until all safety checks are satisfied and the craft is armed. 
4. Now set Output Type to SERVOS and add a servo mix for the motor using Stabilised Throttle. 
5. Do not arm, and raise the throttle. Result: the motor will spin. 

## Expected behavior
Stabilised Throttle implies that it can be controlled by the flight controller, which should include all arming procedures and auto throttle operations. It should work the same as the throttle on a motor output. Only, it outputs as a servo. I would understand RC Throttle ignoring the arm, as that is directly from the Throttle channel. But stabilised throttle should obey all checks. 

## Suggested solution(s)
Make Stabilised Throttle act the same as a motor output. That is the behaviour implied by its name. 

## Additional Information 
FC independent. This is certainly true of INAV 5.1 and 6.0. No doubt true in earlier versions too. It is just more prominent and pressing to fix now that `output_type = SERVOS` exists.

DzikuVx, 2023-03-16:
It's true for every version oF INAV. And yeah, time to fix it probably

Jetrell, 2023-03-16:
@MrD-RC Interesting that you brought up this issue.
A friend is building an RC jet boat, with INAV stabilization control.
And wants to use the stabilized throttle in conjunction with yaw heading control.. So the jet boat will be easier to keep running straight up  the rapids.
As you would be aware, both Jet boats or Swamp (air) boats, both require the motor to be proportionally active to direct the thrust, for steering control..  He had asked about being able to still steer, if the FC wasn't armed.

I just thought I'd mention this rover/boat case. . Even though its still a work in progress for him.

MrD-RC, 2023-03-17:
My first thought is that if you're looking to control something that relies on propulsion for control. It should be armed to control it. Same with a quad. The only difference is that the boat won't fall from the sky.

PR #8916 (2023-03-26): Pull all servos to min when disarmed if thr is mixed in
Fixes #8889
closed 2023-03-29 by DzikuVx