iNavFlight/inav #8894: Failsafe on Serial MSP disconnect

synersignart, 2023-03-17:
## Current Behavior
there are 3 failsafe method
Lost of RC Link -this can be done on the RC radio setup to set the aux position would be on Disconnect
Low  Battery - this can be set on Programming in Logic condition that if the Voltage is this level puts flight mode to RTH
Lost of Serial MSP - can any one tell me what to make this work ??? (currently the drone retains last MSP commands)

## Desired Behavior
an option to activate MSP serial read on constant basis if a disconnect happens a logic condition can be triggered putting the drone on RTH mode 

## Suggested Solution

- on Multiwii is if MSP serial RC reads less than 900ms or ( below RC_Min)  it switches command over to RC link if there is no RC link too, it switches to 1500ms which a flight mode of RTH can be set to

- on PX4 or Ardupilot if Mavlink disconnects which can be set on failsafe menu 

##Suggestion
- we make use of the similar style of multiwii or we use similar to PX4 and Ardupilot that  an serial telemetry can be set to constant monitor. but with the option of switching the feature on or off in the GUI

## Who does this impact? Who is this for?
anyone who uses MSP Serial for both RC controls and Telemetry because for reason they prefer using the joystick  or having a GCS computer to fly the drone.

ricardodeazambuja, 2023-04-14:
Hi @synersignart, would you have more details about this issue? We are having something that seems related. Since we started using the official inav 6 release, the MSP connection to our quadcopters stop working after a while and we need to power off/on to get it back (we are using GPS, BARO, etc from MSP, so lots of messages). When we use the master from around Nov 2022, this problem disappears. Thanks in advance for any extra info!

stronnag, 2023-04-14:
* If you are using MSP RX for RC control, it will invoke the specified failsafe procedure (RTH etc) if 
  - A channel value is below the specified minimum
  - Input is not provided at 5Hz 
  - i.e. same conditions as for "radio" RX failsafe
* If you wish to fall back to RC link on MSP RX failure, consider compiling your own firmware with `USE_MSP_RC_OVERRIDE`

MSP RX failsafe demonstrably working with INAV 6.0, INAV 6.1 and INAV 7.0 with the [SITL](https://github.com/iNavFlight/inav/blob/release_6.1.0/docs/SITL/SITL.md) and [fl2sitl](https://github.com/stronnag/bbl2kml/wiki/fl2sitl) (which has some clues for MSP RX and failsafe).

closed 2024-03-21 by b14ckyy