iNavFlight/inav #8895: I have another problem, which is that when I arm the plane, it writes that it cannot be armed because of the MSP

besarridvan, 2023-03-17:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

besarridvan, 2023-03-17:
I have another problem, which is that when I arm the plane, it writes that it cannot be armed because of the MSP

b14ckyy, 2023-03-17:
Please fill out the bug ticket submission form properly. Without any additional information it will be impossible to give a adequate answer.

DzikuVx, 2023-03-20:
INAV does not block arming because of MSP

closed 2023-03-20 by DzikuVx