iNavFlight/inav #8897: Gps arming error despite appropriate locked SATs.

Joe1602, 2023-03-18:
## Current Behavior
Matek f405 WMN is giving an arming error for no gps fix but is reporting through telemetry adequate SAT lock and an hdop of 1.5.  Will not allow arming through the previous over ride of 10 on/off switchs. Previously the board was on 6.0fp and did not seem to have the issue.

## Steps to Reproduce

1.Appears to be random.
2.Have matek f405 WMN on 6.0 rc 3.
3. Attempt to power up craft.
4. Have had instances where the error clears as expected after gaining 7 SATs, clears after sitting an extended period (10+ minutes), and does not clear at all.

## Expected behavior
Allowed arming after gaining minimum required SAT lock.  

## Suggested solution(s)
I assume something has changed in the gps arming flag since the 6.0 fp release.

## Additional context
Can provide the dump on request, was a fresh flash of 6.0 rc 3 set with defaults for delta wing.

---

- FC Board name and vendor: matek f405 wmn
- INAV version string: 6.0 rc 3

MrD-RC, 2023-03-18:
Are you sure that they all have 3D lock? It is possible to have more than 6 sats, but not all have a 3D lock.

Joe1602, 2023-03-18:
I'm not sure how I can check that with INAV, I know the betaflight configurator will tell the quality of each sat but I'm not seeing that in the INAV configurator.  I would think with a reported HDOP between 1.3 and 1.7 that a sufficient lock has occurred but if there is a better way for me to troubleshoot this I'm all for it.

Jetrell, 2023-03-18:
@Joe1602 The GPS tab in the Configurator, will help you trouble shoot.. But it doesn't provide the signal strength of each satellite, only EPH and EPV... Signal strength would be helpful, when you have localized RF noise from VTX's or HD camera's, bombarding the GPS receiver 
I've experienced the same as you, on my HD quads, that have noisy VTX's. When the signal strength on a given day, from the available satellites is low. Even if 7 or 8 satellites are present.. HDOP doesn't give much indication under these conditions. 
_As can be seen below, with a 2.2 HDOP and a 3D fix_

[image]

While out flying, look to see if the PPS led on your GPS is blinking. This informs you of a valid fix.

Jetrell, 2023-03-19:
@Joe1602 The GPS tab in the Configurator, will help you trouble shoot.. But it doesn't provide the signal strength of each satellite, only EPH and EPV... Signal strength would be helpful, when you have localized RF noise from VTX's or HD camera's, bombarding the GPS receiver
I've experienced the same as you, on my HD quads, that have noisy VTX's. When the signal strength on a given day, from the available satellites is low. Even if 7 or 8 satellites are present.. HDOP doesn't give much indication under these conditions.

While out flying, look to see if the PPS led on your GPS is blinking. This informs you of a valid fix.

Joe1602, 2023-03-19:
If you guys are saying this could be normal Im not going to argue. I've got a number of setups across INAV, ardu, and betaflight and I've never experienced a flight controller reporting SATs greater than required but still giving an error.

Jetrell, 2023-03-19:
>  I've never experienced a flight controller reporting SATs greater than required but still giving an error.

I wouldn't call it normal.. But I've experienced this too with both AP and INAV, over the last 12 months.. It doesn't happen to me on a regular occurrence... But is frustrating when it also occurs with low noise/high component separation on my analogue models.  

Here is an example on the **same model**.. 
On one specific day a few months ago., I had 8 SATS, with the HDOP of 2.8. And it couldn't get a 3D fix for over 10 minutes.
On the following day. I plugged in the FC and got 32 SATS, with a HDOP of 0.8. And got a 3D fix within 30 seconds.

Out of interest. What country are you in?

Joe1602, 2023-03-19:
This in the USA. Only reason it draws a flag in my head is due to this being the only model and board that I'm having an issue with.  May not be of much help but here is a video .
https://youtu.be/Ww4_wl7FxWU
I've had more than one instance where the arming flag is still present at 6-8 sats but then quickly clears; this is the first and only occasion where a board is telling me I have 14+ sats and still reporting a arming flag consistently.

Jetrell, 2023-03-19:
Walksnail Avatar emits stacks of 1500Mhz RF noise from the VTX unit. Which raises the RF noise floor in the vicinity of the GPS... Effectively reducing accuracy, or drowning it out completely, if the two are within 10cm or so.

Remembering the GPS patch antenna is unidirectional. with only 180° coverage.. So don't be surprised, if the GPS signal drops off even more in flight, as the quads bank angle increases.

breadoven, 2023-03-20:
A GPS fix is determined by setting `max_eph_epv` rather than HDOP. You can compare the values in the Blackbox log although only by stripping out the GPS data separately. This shows HDOP and eph values. Although you'd need to arm using the arming bypass method to bypass the GPS fix arming blocker obviously.

DzikuVx, 2023-03-20:
Like @breadoven explained, ARMING is blocked by HDOP not sats count