iNavFlight/inav #8899: Jeti Ex Bus might cause iNav freezing

RobinatiNav, 2023-03-19:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

RobinatiNav, 2023-03-22:
The main reason why I'm investing my time to iNav is that I want to keep the model in particular altitude and horizontally stable. I use Jeti REX10A receiver. One of the flight modes is Horizon. In this mode only Roll movement is stabilised and pitch is set to fly slightly nose down. I don't like that mandatory descending. Loiter or Cruse mode would be perfect. I spent over 100€ for FC to only found out it is not reliable using EXBUS protocol which is supposed to be working 100%. What a shame. Now I have expensive OSD. Fortunately, I was able to configure my Jeti Tx to keep Altitude like I want even the Horizon mode tries to push model down.....

MrD-RC, 2023-03-22:
That is not true regarding Horizon. Horizon is stabilised on all axis is also self levels. If you find that your aircraft descends hands off in Horizon. Your level trim is not set correctly. The easiest way to set this is by enabling the **AUTO LEVEL** tuning mode. Fly hands off until it keeps level. Then disable the tuning mode.

RobinatiNav, 2023-03-22:
> That is not true regarding Horizon. Horizon is stabilised on all axis is also self levels. If you find that your aircraft descends hands off in Horizon. Your level trim is not set correctly. The [...]

Hi Darren, it is misunderstanding I was talking about JETI REX10 Assist flight mode called Horizon.

egilrove, 2023-03-22:
Also using Jeti ExBus  Robin.  Great system, but have same problems. Can I send you a PM? Adress?

RobinatiNav, 2023-03-23:
Hi, of course.Regards, Robert Bulko.Odoslané z iPhonuDňa 23. 3.

RobinatiNav, 2023-03-23:
***@***.*** pozdravom, Robert Bulko.Odoslané z iPhonuDňa 23. 3.

RobinatiNav, 2023-03-23:
> Also using Jeti ExBus Robin. Great system, but have same problems. Can I send you a PM? Adress?

Hi, of course. robert.bulkoatgmailldotcom

RobinatiNav, 2023-03-24:
Maybe somebody from devs could check following, please:

https://github.com/iNavFlight/inav/blob/master/src/main/rx/jetiexbus.c
[EX Bus protokol v1.21 EN.pdf.pdf](https://github.com/iNavFlight/inav/files/11059971/EX.Bus.protokol.v1.21.EN.pdf.pdf)
[JETI_Telem_protocol_EN_V1.07.pdf.pdf](https://github.com/iNavFlight/inav/files/11059974/JETI_Telem_protocol_EN_V1.07.pdf.pdf)

RobinatiNav, 2023-03-24:
I do not understand, no one takes it seriously?

stronnag, 2023-03-24:
> I do not understand, no one behind iNav takes it seriously?

You do not understand how open source works. None of the developers uses this radio, we cannot debug hardware we don't have. You have the option to use a better supported TX/RX.

RobinatiNav, 2023-03-24:
> > I do not understand, no one behind iNav takes it seriously?
> 
> You do not understand how open source works. None of the developers uses this radio, we cannot debug hardware we don't have. You have the option to use a better supported TX/RX.

What do you mean by saying I do not understand? Is it OK for you to keep faulty protocol and not saying about it? I said it slready. If the EXBus protocol is not 100% working just delete it. It is bezter to not supoort it at all that keep the faulty protocol.

RobinatiNav, 2023-03-25:
Conclusion:

Once upon a time there was an enthusiast developer who implemented EXBus protocol into the iNav (what a greet man) By the time something strange happened and EXBus protocol has become not reliable inside the iNav. It was discussed here but never solved and it was left behind somehow forgotten....
Now, a modeller flying Jeti stopped by and decided to try iNav. He found it was promising software which could enhance his previous flying experience, so he's bought necessary hardware to give it a try but his invitation into the iNav world was a bad experience in a form of a crash. iNav has failed to deliver expected.
Investigation took a place and what was found is crazy. The problem with EXBus protocol was not new but people inside the iNav has failed to inform about it... Should I continue?

I think we could close this issue in here. We all know what I am talking about.

MikeW19, 2023-03-25:
Hi together,
I would prefer not to close the issue. I think iNav is amazing and we should take the change to make it even better. Besides I guess we don’t need to mention, it is a non-commercial project, developed by a couple of enthusiasts in their free time. Therefore, nobody can expect to get a bugfix at once. 
I also encountered two crashes two weeks ago. Initially I wasn’t sure whether I stalled my model in the final turn but last week I encountered a frozen telemetry on the ground. I used a Matek F411 WTE in combination with a Jeti Rsat 2 EX and inav_5.1.0. The radio link was still working, but I didn’t receive any telemetry and control stick inputs had no effect.

This didn’t look like a lost UART package more like a segmentation fault (and might relate to different compiler settings).
I haven’t locked into the code yet. Therefore, it would be interesting to get to know more details regarding other crashes. It would also be great if some of the developers could point out some functions that might be worthwhile to have a closer look at.
Best Regards,
Mike

RobinatiNav, 2023-03-25:
> Hi together, I would prefer not to close the issue. I think iNav is amazing and we should take the change to make it even better. Besides I guess we don’t need to mention, it is a non-commercial [...]

Hi Mike, I agree with you but we are standing at wrong side of the barricade now. I was told devs don't have Jeti hardware nad Jeti users are small minority, so they are not willing to invest their time into us. :(

MikeW19, 2023-03-25:
Hi Robin,
I am sure there is no barricade. It is just a matter of free time and priorities and I am pretty sure there are plenty of people interested in a working Jeti telemetry. Besides, I don’t think you will need hardware to debug. I just started reading the pdf’s provided by Jeti. There is an identifier encoding the length of the transmitted package. If this gets messed up, we might end up with a segmentation fault. So, we can already search the code for lines that access the memory. I just started looking into the code and already found the following block. 

// Store in frame copy
jetiExBusFrame[jetiExBusFramePosition] = (uint8_t)c;
jetiExBusFramePosition++;

So the next step will be to check for the size of the allocated array jetiExBusFrame and see if jetiExBusFramePosition will always stay below the allocated memory size.

You see the issue is already under investigation;)

RobinatiNav, 2023-03-25:
> Hi Robin, I am sure there is no barricade. It is just a matter of free time and priorities and I am pretty sure there are plenty of people interested in a working Jeti telemetry. Besides, I don’t [...]

Well, God bless you and I mean it. ;)

MrD-RC, 2023-03-25:
Robin, that is a pretty unreasonable thing to say to be quiet honest. Of course, we are concerned about bugs which can lead to a lost aircraft. This could also lead to injuries. That is the last thing any if us want.

But, how can any developer work on fixing an issue with an RF system if they do not have that RF system? It is not that we are unwilling to spend time on this (I personally have very little at the moment). But I doubt many of the INAV developers have a few thousand to spend on getting a Jeti system needed to look in to the problem.

RobinatiNav, 2023-03-25:
> Robin, that is a pretty unreasonable thing to say to be quiet honest. Of course, we are concerned about bugs which can lead to a lost aircraft. This could also lead to injuries. That is the last [...]

Darren, I am sorry I had to go so far. I did not find any other way how to get your attention. I did not want much anyway. Only hex file compiled in GCC 9.3.1 because after I read almost everything I could it was a chance and also I do not want other to loose their planes or hurt someone which is even more important. So, thank you that you care.

egilrove, 2023-03-25:
Hi all.

I am also a Jeti user (ExBus), mainly flying scale models. 

Came across iNav some time ago, and see a great potential in it for us as a support and failsafe system for our models. A different way of setting up the models, but it is just great to have some support while my model is passing in front of the sun, and when I feel that I am too far out for the comfort.

Even greater as a failsafe function, making my model return back home instead of crashing if I loose it.

INav telemetry back to the transmitter is so useful, and the Jeti/iNav combination is perfect for us.

Unfortunately, bench tests has showed me that iNav is not yet stable in ExBus mode.

I have great respect for iNav as a system, the way it is developed, and the guys that are working on it.

But I also have a prayer for you. 

PLEASE DO NOT LEAVE US JETI USERS BEHIND IN A DITCH.....!  

Many RC pilots out there are using Jeti, but has not yet discovered iNav.  A lot of us want to be a part of the iNav community.

I understand that there are busy times releasing 6.1, but when it all calms down, I strongly hope that ExBus will remain in iNav, and will be reliable.

I also think we Jeti users should be patient, polite and respectful.

At last, I want to directly ask you iNav guys what we Jeti users have to do to be welcome in your iNav world? 

We are knocking on your door. What do you demand from us to let us in?

Egil

egilrove, 2023-03-25:
> Robin, that is a pretty unreasonable thing to say to be quiet honest. Of course, we are concerned about bugs which can lead to a lost aircraft. This could also lead to injuries. That is the last [...]

Hi DARREN
Trying to catch the essence of all this.........

Just asking.......  
If some of us Jeti users come together and equip one trusted iNav developer with a Jeti transmitter and a Jeti receiver, could that secure our future in the iNav world?

Could that guy be you?

Could that be our contribution?

Egil

RoadyFPV, 2023-03-26:
I could provide a REX 3

RobinatiNav, 2023-03-26:
> Hi Robin, I am sure there is no barricade. It is just a matter of free time and priorities and I am pretty sure there are plenty of people interested in a working Jeti telemetry. Besides, I don’t [...]

Hi Mike, what if there is nothing wrong with the code and compiler itself is the problem?

MikeW19, 2023-03-27:
Hi Robin,

I don’t think a different compiler is going to help us. It might tackle a symptom but it is not going to address the root cause. A different compiler might generate a different memory layout which might cover up a segmentation fault for one specific target. But I wouldn’t dare to fly with it.

So far, I wasn’t really able to reproduce the error on the ground. That’s why I think a jeti setup is not really required for debugging at the moment. I encountered the error twice in flight and only once on the ground. And for sure I am not keen on testing it again while being airborne. 

The system was stable for quite some time. All of a sudden, the controller stopped working. The servos were frozen and I didn’t receive any telemetry. 

I am not sure if we encountered a segmentation fault due to an invalid data package (there is a checksum), a variable that wasn’t properly initialized or whether the system got stuck in a while loop. 

I had a look at the jetiexbus.c, checking all lines of code where data is saved to the memory. What I have seen looks pretty solid. I will have a look at the jetibus.c to see how telemetry packages are being generated.

Speaking of telemetry. In the reference design, the EX bus is connected to the TX and RX via a pull up resistor. Here it’s only connected to the TX line. So I guess it is worthwhile to see how the TX is switched to RX and back and have a look if there is a possibility to get stuck while polling the RX port (I would assume there should be a timeout or the system should go into a failsafe but don’t know it yet). 

Besides, any input from the developers is welcome:)

MikeW19, 2023-03-27:
I will see if I can reproduce the crash by moving the throttle stick up and down. 

Right now, I am far away from understanding the code. I just had a look at 

serialWriteBuf(jetiExBusPort, jetiExBusTelemetryFrame, jetiExBusTelemetryFrame[EXBUS_HEADER_MSG_LEN]);

void serialWriteBuf(serialPort_t *instance, const uint8_t *data, int count)
{
    if (instance->vTable->writeBuf)
    {
        instance->vTable->writeBuf(instance, data, count);
    }
    else
    {
        for (const uint8_t *p = data; count > 0; count--, p++)
        {
            while (!serialTxBytesFree(instance))
            {
            };
            serialWrite(instance, *p);
        }
    }
}

I am not sure if there is a possibility to get stuck in the while loop

while (!serialTxBytesFree(instance)){};

uint32_t uartTotalTxBytesFree(const serialPort_t *instance)
{
    uartPort_t *s = (uartPort_t*)instance;

    uint32_t bytesUsed;

    if (s->port.txBufferHead >= s->port.txBufferTail)
    {
        bytesUsed = s->port.txBufferHead - s->port.txBufferTail;
    } 
    else
    {
        bytesUsed = s->port.txBufferSize + s->port.txBufferHead - s->port.txBufferTail;
    }

    return (s->port.txBufferSize - 1) - bytesUsed;
}

currently I don't understand how the UARTs work / how serialWriteis called and therefore how txBufferHead and txBufferTail get modified. But maybe there is someone that can enlighten me, otherwise I will continue tomorrow;)

RobinatiNav, 2023-03-27:
A little conspiracy 🗡️ 

 there is a guy on Youtube called [RT-RC](https://www.youtube.com/@RTRCTV) He said in one of his videos, he asked Pawel Spychalski for implementing the Jeti telemetry into the iNav. He provided Jeti DS-12 radio to Pawel and Pawel was able to get job done in a couple of weeks. I was this information in the RT-RC's video published on July 2021. The RT-RC was very excited bout the result as you can see in the video but it is about a 3 mont ago he sold his Jeti stuff and moved to FrSky....

I am not going to continue ;)

Refference: https://www.youtube.com/watch?v=bQhVUxdAXlQ&t=1s

MrD-RC, 2023-03-27:
This is getting ridiculous!! What does Rob's choice of transmitters have to do with anything. He was kind enough to loan Paweł his transmitter for the Jeti implementation. Later on he decided to switch brands. Rob is mostly flying line of slight and slope soaring at the moment. So isn't even using INAV. I'm finding all this mud slinging pathetic if I'm being perfectly honest!

There is an issue. Stick to facts about the issue. There is no need for any of this muck raking. It will do nothing with regards to solving the problems.

RobinatiNav, 2023-03-27:
> This is getting ridiculous!! What does Rob's choice of transmitters have to do with anything. He was kind enough to loan Paweł his transmitter for the Jeti implementation. Later on he decided to [...]

Darren, calm down please. I am not insulting anyone only. We will find solution. I can find something else between the lines. The JETIEXBUS worked fine before the telemetry was implemented. After it was added the JETIEXBUS could become unstable. Maybe, that was the reason why Rob did not want iNav to control his planes and decided to move to another brand. I am not saying it was like that. It is just a brainstorm.

I have many questions. I don't know why the telemetry goes through the one wire together with RC link. I asked Jeti how it works because in their published documentation they use Tx and Rx and 2,4kOhm resistor. I am waiting for the answer.

This happened in a similar time the GCC 9.3.1  was updated to GCC 10.x.x

I am trying chronologically find out what and when could lead to JETI EXBUS instability.

MrD-RC, 2023-03-27:
Yes, you are insulting the generosity of Rob here. Stop trying to imagine things between the lines. They only exist in your head.

Stick to the facts of the issue only! This is not a gossip magazine!

RobinatiNav, 2023-03-27:
> Yes, you are insulting the generosity of Rob here. Stop trying to imagine things between the lines. They only exist in your head.
> 
> Stick to the facts of the issue only! This is not a gossip magazine!

Thank you Darren, I just bought T-shirt from your website to somehow support your work. Never mind. I see thing in a different light.

RoadyFPV, 2023-03-28:
There was no issue in bf since 2018 marv-t does a good job. Bf use now gcc10 too with the same result.

Here resistor information.
https://github.com/betaflight/betaflight/pull/6767

RobinatiNav, 2023-03-28:
> There was no issue in bf since 2018 marv-t does a good job. Bf use now gcc10 too with the same result.
> 
> Here resistor information. [betaflight/betaflight#6767](https://github.com/betaflight/betaflight/pull/6767)

So, marv-t managed to remove the resistor and made it running via one wire only. Well done.
It was working fine but now it freezes for " unknown reason" in Betaflight and iNav. Do I get it right?

Well, I think we should get somehow marv-t's interest to solve this issue.

RobinatiNav, 2023-03-31:
Well, for me this is done. I decided not to allow iNav to control my plane but I am going to use it for OSD and some nav telemetry it will be the same like [RT-RC](https://www.youtube.com/@RTRCTV). He knows why he made such a decision and know too. At the end, for me it is enough. [RoadyFPV](https://github.com/RoadyFPV) Thank you very much. :)

happylandings, 2023-04-22:
hi Christoph, could you please share matekH743.hex compiled in GCC 9.3.1 please?

RoadyFPV, 2023-04-23:
Hi sry no RC Version. 6.1 Release no problem, remind me pls

happylandings, 2023-05-13:
hi Christoph, a small reminder. Could you please share latest iNav 6.1 matekH743.hex compiled in GCC 9.3.1 please? :)

RoadyFPV, 2023-05-14:
Lets see what i can do :)

RoadyFPV, 2023-05-14:
[inav_6.1.0_MATEKH743_GCC9.3.1.zip](https://github.com/iNavFlight/inav/files/11471271/inav_6.1.0_MATEKH743_GCC9.3.1.zip)

happylandings, 2023-05-14:
> inav_6.1.0_MATEKH743_GCC9.3.1.zip

Thank you very much :)

happylandings, 2023-05-21:
I tested INAV 6.1 compiled in GCC 9.3.1 and it works fine. I noticed one thing. In previous version 6.0 when landed OSD always showed altitude about -8 m but now with 6.1 it is pretty accurate.

happylandings, 2023-06-19:
Hi Christoph. Could you please share latest iNav 6.1.1 matekH743.hex compiled in GCC 9.3.1 please? :)

RoadyFPV, 2023-06-19:
I've already prepared something 

[inav_6.1.1_MATEKH743_GCC9.3.1.zip](https://github.com/iNavFlight/inav/files/11785941/inav_6.1.1_MATEKH743_GCC9.3.1.zip)

JBKingdon, 2023-08-13:
FYI I think BF has now run into the same problem: https://github.com/betaflight/betaflight/issues/12756

RoadyFPV, 2023-08-14:
> There was no issue in bf since 2018 marv-t does a good job. Bf use now gcc10 too with the same result.
> 
> Here resistor information. [betaflight/betaflight#6767](https://github.com/betaflight/betaflight/pull/6767)

see post above

RoadyFPV, 2023-08-14:
https://github.com/betaflight/betaflight/issues/12140

on jan i noticed the bug in BF too

RoadyFPV, 2023-09-03:
@DzikuVx 

Pls take a look at these: https://github.com/betaflight/betaflight/issues/12756

maybe we can use the same solution to fix the freeze issue?

MrD-RC, 2023-09-03:
Using an old SDK is a bandaid, not a solution. The SDKs are updated to fix bugs and improve other aspects.

JBKingdon, 2023-09-03:
> Using an old SDK is a bandaid, not a solution. The SDKs are updated to fix bugs and improve other aspects.

Agreed, but it may be the best solution available in the short term. If the problem is caused by something in the sdk or even the compiler toolchain then it will likely take a substantial amount of work to track down. We'll be in better shape if the sdk change is only exposing a problem in our respective code bases, but even so we will need to be able to reproduce the failure on the bench to be able to debug it effectively. On the BF side I don't think we are able to do that yet, do you have a good repro for iNav?

RoadyFPV, 2023-09-03:
I linked BF to show that it is definitely a gcc problem. First inav changed to gcc10 and this caused the issue. then betaflight had the same problem since gcc10 was used.

RoadyFPV, 2023-09-10:
i asked jeti if the transmitter frequency affects the output of the exbus signal at the receiver. this was affirmed. it affects the generation speed of the exbus at the reciever putput. this confirms the faster freezeing of the fc with 100Hz instead of 50Hz transmitter frequency.
it looks like the fc is getting out of sync and freezing.
also, disabling the telemetry option in the configurator seems to fix the problem as well.

could this help to isolate the problem?
and how is this related to gcc10?

RoadyFPV, 2023-10-22:
#13130 on Betaflight could be solve the freezing issue.
i have done 7 test today and have no freeze.

@MrD-RC @DzikuVx  could we port this to inav?

DzikuVx, 2023-10-25:
@RoadyFPV yes we can :)

DzikuVx, 2023-10-25:
@RoadyFPV #9407

RoadyFPV, 2023-10-25:
@DzikuVx thats great, how can i test it?

closed 2023-10-26 by DzikuVx

RoadyFPV, 2023-10-29:
@DzikuVx i could not find any documentation on how to connect my flight controller to the receiver. some time ago i created a document. shall we add this?

i have now flashd inav 7 dev to my h743 and have done 3 test runs the fc dont freez up. is the jeti fix already in this version?
[How to setup your RX and FC for Jeti Ex-Bus.pdf](https://github.com/iNavFlight/inav/files/13197608/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)

egilrove, 2023-10-29:
iNAV will freeze if 24ch mode is activated in Jeti Transmitter.

Heard this will be fixed in 7.0.

RoadyFPV, 2023-10-30:
@egilrove please don't mix anything up here. INAV dont freeze when u try 24ch mode.

see     https://github.com/iNavFlight/inav/issues/8803     this is the right behavior

i was able to observe the same thing some some time ago, but I have not felt the need to report this. 
Why? 
INAV support only 16ch. its better to force the user to choose the right setting in the transitter. 

@DzikuVx  @MrD-RC 
Now we have strange behavior i noticed tody. maybe caused by this changes u told?
In the configurator, channel 17 jumps back and forth randomly between 1000 and 2000. The speed of this, changes with the stick movements.
As soon as I activate the 24ch mode in the transmitter this behavior disappears and channel 17 is solid at 1500. if i switch back to 16ch mode ch17 stays solid at 1500 too.

I think its not a jeti protocol issue, I cannot observe such behavior with betaflight.

egilrove, 2023-10-30:
Ok....

happylandings, 2023-11-04:
> i have now flashd inav 7 dev to my h743 and have done 3 test runs the fc dont freez up. is the jeti fix already in this 

Hi Christoph, have you tried latest 7 RC1 yet? Is it really reliable? I mean using JETIEXBUS. Thanks for sharing your experience.

RoadyFPV, 2023-11-04:
> > i have now flashd inav 7 dev to my h743 and have done 3 test runs the fc dont freez up. is the jeti fix already in this
> 
> Hi Christoph, have you tried latest 7 RC1 yet? Is it really reliable? I mean using JETIEXBUS. Thanks for sharing your experience.

Hello, yes I have already tested 7 RC1. 3 test runs are inconspicuous. it's best to try it yourself and decide whether you trust it. i think the betaflight guys have done a great job to find the error. i have already carried out several tests under bvetaflight and could no longer observe any freezing.

happylandings, 2023-11-05:
> > > i have now flashd inav 7 dev to my h743 and have done 3 test runs the fc dont freez up. is the jeti fix already in this
> > 
> > 
> > Hi Christoph, have you tried latest 7 RC1 yet? Is it really reliable? I mean using JETIEXBUS. Thanks for sharing your experience.
> 
> Hello, yes I have already tested 7 RC1. 3 test runs are inconspicuous. it's best to try it yourself and decide whether you trust it. i think the betaflight guys have done a great job to find the error. i have already carried out several tests under betaflight and could no longer observe any freezing.
> 
> It would be useful if several Jeti pilots could test it and give feedback

Thank you Christoph, you are right some more Jeti pilots should try it. I am gonna wait until those who decided to do so will give us a positive result. I am sorry to say it but I do not trust DzikuVx's words. I watched his latest promotional video and he said the problem with JETIEXBUS was not connected to the GCC 10.2.1 but it was about the telemetry. He said the problem was solved (thanks to devs from Betaflight) and from now JETIEXBUS should work. "Should" is not enough for me. Anyway, tell me why it was working without any problem using GCC 9.3.1?

[20 more replies]

PR #8948 (2023-04-05): Revert "Switch H7 to -Ofast optimizations"
Probably 
fixes #8409
fixes #8905 

Could resolve #8899 for H743WLITE