iNavFlight/inav #8903: Add Iflight succex-e f7 v2 flight controller support

Ovalleader12, 2023-03-22:
Fc name: iflight succex-e f7 v2
Bf target name: IFLIGHT_SUCCEX_E_F7

fermiums22, 2023-04-04:
In order to officially add it, they ask to send a prototype to them. I myself compiled the firmware for myself. But this is a difficult path, and there are no guarantees.
[IFLIGHT_SUCCEX_E_F7.zip](https://github.com/iNavFlight/inav/files/11151927/IFLIGHT_SUCCEX_E_F7.zip)

Ovalleader12, 2023-04-04:
Dude then you do much

Ovalleader12, 2023-04-04:
how do i go about flashing this? i see that its a c and an h file

fermiums22, 2023-04-05:
I'm not going to collect all the firmware versions for you.
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20or%2011%20with%20Linux%20Subsystem.md

closed 2023-10-27 by DzikuVx