iNavFlight/inav #8923: INAV 6.0.0 and Matek 405TE Mar 22 2023 with Jeti Receiver

boomerschnecke, 2023-03-29:
Strange behaviour with INAV 6.0.0 and Matek 405TE Mar 22 2023 with Jeti Receiver

Pulling ROLL function for longer time (f.e. 2sec) the Roll Servos have very long delay to return back to neutral postion.
Pulling Roll function shortly everything is ok 

Some with elevator servo

Also RTH Mode is missing 

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

boomerschnecke, 2023-03-29:
The behaviour depends on throttle position ! Only visible at full throttle

RoadyFPV, 2023-03-31:
Servos freeze compiled with ggc10 **and** with gcc9....
The issue does not seem to be related to this: https://github.com/iNavFlight/inav/issues/8899

boomerschnecke, 2023-03-31:
I am actually using SBus. So not connected to JetiEX-Bus issues !

RobinatiNav, 2023-04-01:
Well, I believe the ideal solution from iNav comunity view is to use FrSky. Forget about solving Jeti connected issues.....  ;)

RoadyFPV, 2023-04-02:
I flashed INAV5.1, the problem is there too.
I think thats not a JETI issue and should be renamed...

RoadyFPV, 2023-04-02:
I have found why the servos freeze...
u enter accidentally by stick wiggeling the OSD menu and the servos freeze until u leave it.

boomerschnecke, 2023-04-02:
Thanks for information.
Could you please explain how to enter this OSD menue mode by sticks activation in order to avoid it? Where is this procedure discribed? How to make a workaroud ?

RoadyFPV, 2023-04-02:
https://www.mrd-rc.com/tutorials-tools-and-testing/inav-flight/inav-stick-commands-for-all-transmitter-modes/

boomerschnecke, 2023-04-02:
Thank you. Didn´t know that, but is this only to be activated if unarmed? Otherwise if this could be activated all the time this would be a big problem?

RoadyFPV, 2023-04-02:
You should not be able to access the menu when armed.

RoadyFPV, 2023-04-02:
u can close the issue?

closed 2023-04-02 by boomerschnecke