iNavFlight/inav #8925: iNav 6.0 - Cannot disengage Manual Flight Mode

AdamGFPV, 2023-03-29:
**Description:** When setting up flight modes, if there is a range added for **Manual** mode, there is no way to disengage manual mode, which means **Acro** is not accessible. When the value for the channel falls outside of the specified range to activate manual, manual mode remains active no matter what. (Please see attached screen shot).

**Note:** If i setup an identical model in 5.1 and 6.0, the issue does not occur in 5.1, but it does occur in 6.0.

**Note 2:** This issue does not occur on all of my flight controllers on iNav 6.0. 

**Hardware:** Matek H743 WLITE

**Steps to replicate:**
1: Set FC in DFU and flash inav 6.0 with full chip erase
2: Connect and select plane without a tail (wing/delta)
3: On serial tab, set 'Serial Receiver Provider' to 'CRSF'. Save and reboot.
4: Go to modes tab, can see Acro is enabled as no flight modes have a range added. Add manual mode and set small range e.g. min: 1125 - Max 1300. Hit Save. (now have to reboot FC as it stops responding to receiver).
5: Go to Modes tab, Manual flight mode will be enabled no matter what value the channel is on, even when outside of the specified range for manual. Therefore it's impossible to activate Acro mode.

Again, I have done the above steps exactly as described on the same FC with both iNav 5.1 and iNav 6.0 and the problem only occurs on 6.0.
[image]

Please let me know if any further info is required or if a video screen recording would help.

Thanks
Adam

DzikuVx, 2023-03-30:
Duplicate of https://github.com/iNavFlight/inav/issues/8905
Closing

DzikuVx, 2023-03-30:
Now, the question is what happens if you restart Configurator?

DzikuVx, 2023-03-30:
@AdamGFPV please perform this scenario:

1. Reproduce
2. Shutdown Configurator without rebooting FC
3. Start Configurator
4. Connect
5. Check the state

AdamGFPV, 2023-03-30:
> I assume sensors have finished calibrating ? Manual remains active if disarmed and sensors are still calibrating.

I believe they only calibrate when powered by a battery? This is a new FC I have not soldered a power connector on yet. It is just powered by USB.

AdamGFPV, 2023-03-30:
> @AdamGFPV please perform this scenario:
> 
> 1. Reproduce
> 2. Shutdown Configurator without rebooting FC
> 3. Start Configurator
> 4. Connect
> 5. Check the state

I've just tried this again to be double sure. The issue is still present.

Would it help if i made a video screen recording?

AdamGFPV, 2023-03-30:
Ok so after a ACC calibration the issue is no longer present. 

The strange thing is that this is not the case on iNav 5.1. ACC Cal is not needed

I am also sure that when I first found this issue, I had already calibrated the ACC when it occurred.

Thanks

closed 2023-03-31 by breadoven

preisedj, 2023-05-03:
I have a similar thing happening with the same FC and INAV 6.0.  The system stays in Manual mode even after selecting any other Flight Mode.  In the Modes tab, the other flight modes light up as I step through them, but the Manual mode continues to be lit and overrides any other mode selection throughout.
But once the GPS locks in, this condition clears and everything performs as it should.