iNavFlight/inav #8929: no OSD

DimitriyUmniy, 2023-03-30:
Hello, I have such a problem. FC MATEK F405 VTOL, system DJI goggles version 2, unit VISTA. No OSD. Installed Ardupilot and Betaflight, everything works fine OSD shows.

RobinatiNav, 2023-03-30:
This might help: https://www.youtube.com/watch?v=elq6AF5gdXE

mmosca, 2023-03-31:
What version of INAV?
What does your ports tab look like?
Is your DJI unit rooted?

DimitriyUmniy, 2023-03-31:
> Какая версия ИНАВа? Как выглядит вкладка портов? Ваше устройство DJI рутировано?

I tried on version inav 5.1 and 6.0.  dji goggles rooted firmware 06. Vista rooted tried version 06 and 08. Ports inav MSP displayport

DimitriyUmniy, 2023-03-31:
> Это может помочь: https://www.youtube.com/watch?v=elq6AF5gdXE

yes i did it like in the video

mmosca, 2023-03-31:
Can you post a diff all from your flight controller?

DimitriyUmniy, 2023-04-01:
> Можете ли вы опубликовать diff все с вашего полетного контроллера?

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/MATEKF405TE_SD 6.0.0 Mar 22 2023 / 10:21:39 (084101ac)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 100 0 -1
smix 1 2 0 100 0 -1
smix 2 1 1 100 0 -1
smix 3 2 1 -100 0 -1

# Outputs [servo]
servo 1 1000 2000 1550 -100
servo 2 1000 2000 1450 -100

# safehome

# features
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map
map TAER

# Ports
serial 3 33554432 115200 115200 0 115200
serial 4 2 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color
mode_color 6 8 0

# Modes [aux]
aux 0 0 0 1700 2100
aux 1 1 1 1650 2100
aux 2 53 5 1850 2100
aux 3 10 3 900 1250
aux 4 3 5 1725 2100
aux 5 21 2 1750 2100
aux 6 42 4 900 1200

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 48 1 V
osd_layout 0 1 29 7 V
osd_layout 0 3 8 6 V
osd_layout 0 6 46 19 V
osd_layout 0 7 29 20 V
osd_layout 0 9 24 4 H
osd_layout 0 10 6 10 H
osd_layout 0 11 36 5 V
osd_layout 0 12 1 6 H
osd_layout 0 13 1 0 V
osd_layout 0 14 42 1 V
osd_layout 0 15 25 6 H
osd_layout 0 20 9 12 V
osd_layout 0 21 9 13 V
osd_layout 0 22 30 17 V
osd_layout 0 23 28 18 V
osd_layout 0 28 23 11 H
osd_layout 0 30 0 10 V
osd_layout 0 32 20 6 V
osd_layout 0 37 25 13 V
osd_layout 0 38 1 9 H
osd_layout 0 41 1 9 H
osd_layout 0 46 3 10 H
osd_layout 0 55 1 9 H
osd_layout 0 102 12 4 H
osd_layout 0 110 48 2 V
osd_layout 0 125 2 1 V
osd_layout 0 129 8 19 V
osd_layout 1 20 18 1 V
osd_layout 1 21 7 1 V
osd_layout 1 30 0 11 V

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set gyro_main_lpf_hz = 25
set gyro_main_lpf_type = PT1
set dynamic_gyro_notch_enabled = OFF
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_y = 4
set gyro_zero_z = 10
set ins_gravity_cmss =  983.015
set acc_hardware = ICM42605
set acczero_x = -2
set acczero_z = 7
set accgain_x = 4086
set accgain_y = 4090
set accgain_z = 4092
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set baro_hardware = SPL06
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set align_board_yaw = 5
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set ahrs_inertia_comp_method = ADAPTIVE
set disarm_kill_switch = OFF
set applied_defaults = 3
set inav_w_z_baro_p =  0.350
set nav_wp_radius = 5000
set nav_wp_max_safe_distance = 500
set nav_emerg_landing_speed = 300
set nav_rth_climb_first = OFF
set nav_rth_altitude = 5000
set nav_auto_disarm_delay = 2000
set nav_fw_bank_angle = 45
set nav_fw_loiter_radius = 4000
set nav_fw_control_smoothness = 5
set nav_fw_launch_max_angle = 180
set nav_fw_launch_spinup_time = 1000
set nav_fw_launch_timeout = 20000
set nav_fw_launch_max_altitude = 12000
set nav_fw_allow_manual_thr_increase = ON
set osd_video_system = DJIWTF

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 70
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set max_angle_inclination_rll = 550
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.200
set nav_fw_pos_z_p = 15
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# battery_profile
battery_profile 1

set nav_fw_max_thr = 1850

# save configuration
save

# end the command batch
batch end

#

mmosca, 2023-04-01:
diff looks ok, and if you have msp-osd installed on both the goggles and the vista is connected to uart 4, it should work.

Can you try adding more elements to alternative layout 1? I suspect the fc may be using it instead of default layout, since you have a switch to change layouts.

neprsitady, 2023-04-01:
I have the same issue as OP. Cant make any OSD or HD OSD work on Goggles 2 now with 4.4

dump all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb 26 2023 / 16:01:38 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 004500223338510539393832
signature 

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A14
resource LED 2 A13
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
[... 1242 lines cut ...]
# rateprofile 3
set rateprofile_name = -
set thr_mid = 35
set thr_expo = 55
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 5
set pitch_rc_rate = 5
set yaw_rc_rate = 5
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 15
set pitch_srate = 15
set yaw_srate = 20
set throttle_limit_type = SCALE
set throttle_limit_percent = 85
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# restore original rateprofile selection
rateprofile 0

DimitriyUmniy, 2023-04-02:
> diff выглядит нормально, и если у вас установлен msp-osd на обоих очках, а Vista подключена к uart 4, это должно работать.
> 
> Можете ли вы попробовать добавить больше элементов в альтернативный макет 1? Я подозреваю, что ФК может использовать его вместо макета по умолчанию, поскольку у вас есть переключатель для изменения макетов.
> 
> Так как системное сообщение и координаты GPS могут не отображаться, если у вас нет блокировки.

Thanks a lot.  but I can try later.  had to leave

DimitriyUmniy, 2023-04-03:
> diff выглядит нормально, и если у вас установлен msp-osd на обоих очках, а Vista подключена к uart 4, это должно работать.
>
> Можете ли вы попробовать добавить больше элементов в альтернативный макет [...]

I added.  I tried to switch to all 3. nothing changes.  also no OSD

mmosca, 2023-04-03:
I am out of ideas. Maybe try a different uart.

julianschweizer, 2023-04-14:
Same here.

DJI Airunit (Old One), Googles V2, Matek F411-WTE FC with INAV 6.0. All is updated to latest Version with fresh install/flash. Config is tripple checked. OSD (wtf.os) is showing for a few seconds then disappearing. If i move elements in the configurator osd shows for a few seconds and disappears again. 

Ideas welcome.

DimitriyUmniy, 2023-04-14:
> То же самое.
> 
> DJI Airunit (старый), Googles V2, Matek F411-WTE FC с INAV 6.0. Все обновлено до последней версии с новой установкой / прошивкой. Конфиг трижды проверен. Экранное меню (wtf.os) отображается несколько секунд, а затем исчезает. Если я перемещаю элементы в конфигураторе, экранное меню отображается на несколько секунд и снова исчезает.
> 
> Приветствуются идеи.

I also noticed that when loading the OSD, I have flickering, but it is not displayed at all, even for a short period of time

mmosca, 2023-04-20:
This sounds more like a hardware/wiring issue than a firmware issue. I will convert this to a discussion.