iNavFlight/inav #8931: Gyro problem SKYSTARSF722HD

mauroS123, 2023-03-31:
Pleae add driver for SKYSTARSF722HD  gyro.   
Need driver for BMI270.
Thanks

- FC Board name and vendor: SKYSTARSF722HD
- INAV version string:

DzikuVx, 2023-04-05:
This target already uses BMI270

closed 2023-04-05 by DzikuVx

mauroS123, 2023-04-05:
Yes, but after restart the board can't detect gyro.
Same problem like previous INAV   https://github.com/iNavFlight/inav/issues/8355#issuecomment-1255585823
then, JulioCesarMatias has released a mod version for this board that working 100%.
Now on INAV 6 I have again the same problem. After flashing the board,  gyro seems to works, but after a restart no more works again

PooPeePantz, 2023-06-03:
I also have the same problem using inav 6.1 with my new Skystar F722 HD Mini.  It also has the BMI270 gyro, and exhibits the same problem described above.

I humbly request someone fix this.

jgchua, 2023-08-01:
Having the same problem as well. Here's the exact [board](https://skystars-rc.com/product/skystars-f722hd-pro-flight-controller-km55a-esc-flystack/) we are trying to use in iNav.

I think the issue is this board is not compatible with the iNav existing targets. This exact board here uses SKYSTARSF722HDPRO target in Betaflight instead of the SKYSTARSF722HD.

Hope this can be supported too someday. Thanks!

ve7mjc, 2024-01-01:
I am unable to use this FC also and would like to amplify some information in hopes we can get it working in iNav.

Problem (at the least):
- Does not recognize GYRO/ACC (AUTO does not work even once unlike OP experience)

Flight Controller Identifying Remarks:

- Marketed as "Skystars F722 Mini HD Pro"
- 20x20 sized, black PCB, with "Skystars\nF722\nMini HD Pro" sticker atop MCU
- PCB Silkscreen, bottom-side: Mini F7HD Pro

What concerns me is that the manufacturer (Skystars) advertises on their website, and subsequently dozens of reseller listings that this product is iNav compatible, and goes so far as listing the the targets:

- Betaflight Target: SKYSTARSF7HDPRO
- INAV Target: SKYSTARSF722HDPRO

The referenced Betaflight target exists - but the advertised INAV target DOES NOT exist in the iNav repository (checked for 6.x and 7.x including unstable).

I was able to flash Betaflight to this FC and it autodectects the following: GYRO=BMI270, ACC=BMI270, BARO=BMP280. Unfortunately I cannot get my Mag to work in Betaflight (and yet, it works in iNav! - perhaps a clue?)

The confirmation of BMI270 versus MPU6000 is important here as Skystars also inconsitently advertises MPU6000 and BMI270 inconsistently and makes no reference on or accompanying the product itself.

I am pretty disappointed in Skystars here - I suspect this has nothing at all to do with iNav project. Are they intentionally misrepresenting this product and assuming the iNav devs will just deal with it?

I am a software dev, I would be willing to attempt a custom target build if anyone could point me in the right direction.

jgchua, 2024-01-01:
Hi Matthew,

I was able to get it working with inav by using the SKYSTARSF722HD target,
adjusted the accelerometer alignment, and leave the imu model to mpu6000 in
the config instead of the bmi270 stated in the specs.

Everything is now working even with the latest firmware update.
- cruise mode
- rth
- alt hold
- pos hold etc.