iNavFlight/inav #8935: Unable to flash ATMOMRCF405NAVI to 6.0

gleeloyd, 2023-04-01:
## Current Behavior
I have a new model with ATMOMRCF405NAVI flight controller installed. It came with INAV 5.1 pre-loaded and INAV Configurator 5.1 works fine with this flight controller.

## Steps to Reproduce

If I start Configurator 6.0 and hit Connect using the same COM4 port as 5.1, it behaves as if it is connected (Connect button turns to red "Disconnect") and it goes straight to CLI. CLI reports nothing and I notice the status info under the logo in upper left says "FC Firmware: Not connected". So then I disconnect and proceed to Firmware Flasher. First time in the correct FC Target was found (ATMOMRCF405NAVI) and it continues to load correctly each time I run Configurator 6.0. I then pick the 6.0 Firmware version and select "No reboot sequence" and "Full chip erase". I then click Load Firmware (Online) which causes the Flash Firmware button to become enabled. But after clicking Flash Firmware, essentially nothing happens. I've followed all the steps under Recovery / Lost communications that I can, however the doc for my FC says nothing about BOOT pins or a BOOT button. Not sure what else to do and frankly a bit fearful of wiping out my 5.1 setup.

## Expected behavior
Should be able to flash the 6.0 firmware

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: ATMOMRCF405NAVI
- INAV version string: 5.1

MrD-RC, 2023-04-02:
Don’t enable “No reboot sequence”. Just enable “full chip erase”

gleeloyd, 2023-04-02:
I tried that the first few times with the same result. "No reboot sequence' is one of the suggested steps under Recovery/Lost Communications.

gleeloyd, 2023-04-02:
The first few attempts were done without "No reboot sequence" with same result. That setting is one of the steps under Recovery / Lost communication.

gleeloyd, 2023-04-10:
This continues to be a problem for me. I use COM4 in Configurator 5.1 and it works in 6.0 Configurator to connect and successfully run the version command in CLI. I then disconnect, and follow this sequence:
1) Go to Firmware Flasher. The correct Target board is pre-filled.
2) Choose a Firmware version (6.0.0)
3) Turn Full chip erase ON - all other selections are left OFF.
4) Click Load Firmware (Online)
5) Click Flash Firmware.

The page immediately jumps to the bottom. 
After a few seconds this message appears in the upper left beside the timestamp:
"Failed to flash COM4"

COM4 is fine in 5.1, and 6.0 allows me to connect on COM4, so why can't the firmware flasher use it?

gleeloyd, 2023-04-10:
I was able to flash. I needed Zadig to install the needed driver to enable a DFU device to be seen by Configurator

closed 2023-04-10 by gleeloyd