iNavFlight/inav #8941: No OSD w/ DJI 03 Air/Goggles 2/ Matek w743 wlite

bonem1dr, 2023-04-02:
Still cant get my osd to work. I am running the most recent release Inav 6.0.0 with DJI 03 air unit, goggle 2, matek h743 wlite fc and have tried both using the msp display port option with bf43compt and DJI FPV VTX options with no luck. Only thing different I see is when I use bf43compt in my goggles it allows me to use canvas mode in normal or hd, but still no OSD at all...

mmosca, 2023-04-02:
post your diff.

RobinatiNav, 2023-04-02:
> Still cant get my osd to work. I am running the most recent release Inav 6.0.0 with DJI 03 air unit, goggle 2, matek h743 wlite fc and have tried both using the msp display port option with [...]

Did you enable OSD in configuration menu? I have the same setup and it works. Here is my setup. I have 2 screens and 3 pos switch (ALT1, ALT2, and NO OSD)

[image]
[image]
[image]
[image]

bonem1dr, 2023-04-02:
Thats how I have it but I didn't try the osd alt 2 switch.
would love to try it but my FC failed and my plane death spiraled into a
lake and was a total loss...

thanks for the info tho!

RobinatiNav, 2023-04-03:
> Thats how I have it but I didn't try the osd alt 2 switch. would love to try it but my FC failed and my plane death spiraled into a lake and was a total loss... thanks for the info tho!
> […](#)

I am sorry you lost the plane. I crashed twice already. Both time with Matek H743-WLITE. First time I partly damaged PDB board and second time with a new PDB. There is something strange happening with iNav and maybe it is related to Matek H743-WLITE too

closed 2023-04-03 by mmosca