iNavFlight/inav #8942: speedybee F7 v3 doesn't work correctly with DSHOT. Some motors are not spining.

AlexZ550Proton, 2023-04-02:
It looks the defect (One motor doesn't spin after flashing iNav #7368) is reoccured in INAV 6.0.

I have Y6 drone based on Speedybee F7 V3 and 2 ESCs (speedybee 4in 1 50A). It works perfect in BF with DSHOT 300/600. 

Once I flushed it with INAV 6.0, the motor 1 was not spinning in DSHOT 300. I thought it migth be a hardware issue and switched back to BF4.4, the drone motors worked fine again. 

Flushed back on INAV and found that motors are spinning normal in STANDARD or MUTLISHOT, but all DSHOT are not but different symptoms. DSHOT 150 - 2 motors spining of 6, DSHOT 300 - motor 1 is not spinning, DSHOT 600 - 1 motor spinning only. 

I saw a notice for Speedybee F405 V3 but it looks it is applicable for F7 as well  https://github.com/iNavFlight/inav/blob/86f11d786d514571c8a48cf04e680b190ec5df08/docs/boards/SPEEDYBEEF405V3.md

AlinPTR, 2023-10-25:
Same issue.
Motors are spining all at first plug the , 2-3 after second plug.
Fc speedybee f7 v3
Esc bheli32 50A
Flashing  inav 6.0.0 doesn't solve the issue.

DzikuVx, 2023-10-27:
will be fixed in INAV 7

closed 2023-10-27 by DzikuVx

DerWolle82, 2023-11-04:
I tested it in INAV 7.0 RC1 and the 4th motor does not spin. I tried other modes... No spinning at DShot, Multi or Standard. Flashed back to latest INAV 6.1 and all motors work fine again. Any advice here?

Setup: FC Speedybee F7 V3 Stack (with 50A ESC)

AlexZ550Proton, 2023-11-04:
> I tested it in INAV 7.0 RC1 and the 4th motor does not spin. I tried other modes... No spinning at DShot, Multi or Standard. Flashed back to latest INAV 6.1 and all motors work fine again. Any advice here?
> 
> Setup: FC Speedybee F7 V3 Stack (with 50A ESC)

Compare diffs, maybe motor ports are incorrect in the latest version

samsab1, 2023-11-09:
hello same thing herre: motor N)4 doesn't spin with INAV 7--> i had to go back to betaflight...too bad
any chance to fix that pb?
best regards from france

samsab1, 2023-11-09:
[INAV_7.0.0_cli_20231109_153634.txt](https://github.com/iNavFlight/inav/files/13309303/INAV_7.0.0_cli_20231109_153634.txt)
Hello
here is the dump
i have once again flash with INAV 7 using the last Release 7.0.0 rc2 firmware; once again motor 4 does not spin
strange, very strange; by the way i have flashed an other quad with speedybee f5 V3 to inav 7, and it works like a charm...

samsab1, 2023-11-09:
i have to add that:  Dshoot600 has been set  as speedybee advise to set. i did not try another setting. should I?

samsab1, 2023-11-09:
motor 4 does not work what ever is choosen: dshoot multishoot, standard: N°4 is not responding

DzikuVx, 2023-11-09:
@samsab1 does this sentence:

> by the way i have flashed an other quad with speedybee f5 V3 to inav 7, and it works like a charm...

means that it's working on a different quad with Speedybee f405 V3?

samsab1, 2023-11-09:
ya, speedybe F5V3 with inav 7 works very well (altitude mode........hum.......very good)

DzikuVx, 2023-11-09:
@samsab1 please test this hex. 
[inav_7.0.0_SPEEDYBEEF7V3.zip](https://github.com/iNavFlight/inav/files/13310495/inav_7.0.0_SPEEDYBEEF7V3.zip)

samsab1, 2023-11-09:
no change, N°4 still do not spin, whatever shot mode used

DzikuVx, 2023-11-09:
@samsab1 the same with Multishot?

samsab1, 2023-11-09:
yes, no difference

DzikuVx, 2023-11-09:
Then are you sure you have everything wired correctly? pin and output assignment is exactly like with INAV 6

samsab1, 2023-11-09:
it works at 100% with betaflight, with exactly the same wiring

samsab1, 2023-11-10:
hello
issue has been closed, any chance to get a solution later?
best regards

samsab1, 2023-11-10:
hello
here are some screenshoot
[image]
one esc is missing with inav 7
[image]
all esc are active with betaflight
[image]

DzikuVx, 2023-11-12:
@samsab1 issue will be fixed in RC3. We've found the reason

samsab1, 2023-11-12:
Thanks PAvel
what i do not understand is the fact that this esc should be nativly  flashed with blheli-s ex works  and mine seems to be flashed iwith blheli32.
do you know how to flash it with blheli-s in order to flash it in the step with blujeay?
best regards

samsab1, 2023-11-13:
hello, thanks for your answer
I think I made a big mistake with the ESCs supplied with the F405 V3 which allow an upgrade to bluejay. the BL32 50A ESC supplied with the F7V3 are indeed in BL32 and do not require an upgrade to bluejay
thanks again

samsab1, 2023-11-20:
hello
i was testing inav 7 RC2 with last firmware of speedybee f7v3 : none motor spin and seems to be detected
[image]
[image]

MrD-RC, 2023-11-20:
@samsab1 update to RC3.

samsab1, 2023-11-20:
i have a big doubt:
**RC3** concerns  inav configurator or the FC speedybee F7V3?
if it concerns FC, i guess i have upgraded to last FC firmware who was 3 hours old 
my INAV is INAV 7 RC2--> is it ok ? i do not see wher to download INAV 7 RC3, does it exist?
[image]

MrD-RC, 2023-11-20:
@samsab1 you need to install RC 2 configurator and flash RC3 firmware. The configurator is correct. But you need to update the flight controller.

samsab1, 2023-11-20:
by the way here is the diff all

diff all

# version
# INAV/SPEEDYBEEF7V3 7.0.0 Nov 19 2023 / 20:22:41 (2e9b1c7d)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Timer overrides

# Outputs [servo]

# safehome

# features
feature -BLACKBOX
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports
serial 5 2 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 1700 2100
aux 1 2 3 1700 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set looptime = 500
set gyro_main_lpf_hz = 110
set gyro_main_lpf_type = PT1
set gyro_use_dyn_lpf = ON
set gyro_dyn_lpf_min_hz = 85
set gyro_dyn_lpf_max_hz = 300
set gyro_dyn_lpf_curve_expo = 3
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 100
set setpoint_kalman_q = 200
set gyro_zero_x = 1
set gyro_zero_y = 13
set gyro_zero_z = 3
set acc_hardware = BMI270
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = BMP280
set blackbox_rate_denom = 2
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set applied_defaults = 2
set gps_ublox_use_galileo = ON
set airmode_type = THROTTLE_THRESHOLD

# mixer_profile
mixer_profile 1

set model_preview_type = 3

# Mixer: motor mixer

# Mixer: servo mixer

# mixer_profile
mixer_profile 2

# Mixer: motor mixer

# Mixer: servo mixer

# profile
profile 1

set mc_i_pitch = 90
set mc_d_pitch = 27
set mc_p_roll = 36
set mc_i_roll = 80
set mc_d_roll = 25
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_type = PT3
set d_boost_min =  1.000
set d_boost_max =  1.000
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 75
set rc_yaw_expo = 75
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# profile
profile 2

set mc_i_pitch = 90
set mc_d_pitch = 27
set mc_p_roll = 36
set mc_i_roll = 80
set mc_d_roll = 25
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_type = PT3
set d_boost_min =  1.000
set d_boost_max =  1.000
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 75
set rc_yaw_expo = 75
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# profile
profile 3

# battery_profile
battery_profile 1

set throttle_idle =  6.000

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
mixer_profile 1
profile 2
battery_profile 1

# save configuration
save

samsab1, 2023-11-20:
iu have reflash with online firmware RC3
now i find back 4 motors, but N°1 do not spin
[image]

samsab1, 2023-11-21:
hello
situation:
i just flash this F7V3 this morning with last firmware RC3; all motors are spining, but only in dshot300, and not in dshot600 as it should be possible with the speedybee f7v3 stack (esc bl32).
is it normal?

samsab1, 2023-11-21:
in fact I have never seen so many difficulties and randomness with INAV
with this fc speedybee F7V3, I have almost all possible behaviors
Dshot600: no motor, then 3 motors, then 1 motor, then finally 4 motors
same in Dshot300 or 150

as I write, Dshot600 programmed, 4 motors are working, but I don't know why or for how long

Doug777777, 2024-03-10:
Hi,  I am building my first drone - F7 V3 , iNav 7 - Only 2 of the 4 motors spin with dshot - I switched to standard - all 4 spin but the drone just flips upside-down - is this to be expected ?  Old fart so I did iNav because its more stable flight Vs beta flight from what I understand.   Any suggestions appreciated.

sensei-hacker, 2024-03-10:
If the board has trouble with DShot z try Multishot. It's faster anyway. Please let us know how that works.

"Flips upside-down" means the motors aren't actually mounted in the positions they are set as.  Use the motor wizard to tell INAV where you mounted each motor.

FraserJB, 2024-09-11:
I have a REKON Y6, FC is a SPECTER F722 running HGLRCF722 in INAV (note, was HGLR-HGLRCF722AIO_X6 under Betaflight). 5 of 6 motors are spinning. All are on the same mixer timers, all work fine under Betaflight (have swapped back to BF and returned to INAV to confirm). DSHOT 600 & 300 same result. (Does not run on DSHOT 150 or any of the other protocols (MULTISHOT, etc). INAV version 7.1.1.

Any ideas if I can get this working?

Phil-MC, 2026-10-01:
To anyone still fighting with the speedybee f7 v3....
[sensei-hacker](https://github.com/sensei-hacker) is correct. The board runs perfectly well using MULTISHOT. 
I have seen this behaviour before and it seems to be related to SPI BUS conflicts possibly caused by the DMA of the DSHOT digital protocol and the BlackBox recording traffic. It doesn't happen with the analogue (calibrated) MULTISHOT.
I fly one of these on INAV 9.1.1 without an issue so far.