iNavFlight/inav #8944: 3D OSD map/Radar

sdellava, 2023-04-03:
## Current Behavior
The OSD map/radard feature show the UAV relative position to the home or vice versa in a 2D map.

## Desired Behavior
Home (and others waypoints) is a 3D point, so it could be possible to show the relative position of the home the the OSD, taking in consideration the current UAV orientation in the space and adapt the logical home point view.
Basically it works like artificial horizon, keeping the home points (and other WPS) logically fixed in the field of view. 
  
  ## Suggested Solution
In this picture you can see a simpliefied view of how the 3D map mode should appear in the OSD.  Of corse in 3D mode the map do not provide distance information but this information is already available as numeric field.

[image]

If the he home is outside of the FOV it it shown on the OSD border, like this:

[image]

## Who does this impact? Who is this for?
All map users

rmaia3d, 2023-04-03:
I'm not I understood correctly your idea, but wouldn't what you are suggesting also require the OSD to have some kind of access to 3D terrain elevation databases?

sdellava, 2023-04-03:
No, It is clear that if the home point is behind an obstacle the OSD still represent it in front of the osbstacle. Let see this example:

[image]

But before to arrive in that situation, hopefully the pilot had this view:

[image]

So it has the clear percption that it is dangerous to descent much more.

No surface information is required to represent the map in 3D view.

sdellava, 2023-04-06:
To simplify the work of the devs, I defined the mathematics required to implement the algorithm.

This image shows the overall scenario in a simplified way 

[image]

This is the view from above
[image]

and this is the view from the side

[image]

Questo è un esempio reale, con informazioni prese da google map:
[image]

In this excel file, I used aerospace trigonometric formulas to calculate the distance and angles between the two points and then estimate the relative position of Home in an XY grid.

Parameter to provide:
- P1 and P2 latitude, longitude and altitude
- Pitch and heading
- Portion of the FOV (horizontal and vertical) to be represented in the OSD grid (in example 80 deg of the 160 total camera FOV)
- dimension of the OSD grid representing the portion of the used FOV 

In red the position X,H in the grid of the Home in the OSD 3D map.

[3D MAP Math.xlsx](https://github.com/iNavFlight/inav/files/11170709/3D.MAP.Math.xlsx)

rmaia3d, 2023-04-10:
Check this video around the 2:00 mark. It shows the OSD recording from a DJI Avata, and I believe it shows the H icon (which I took as meaning home) having the behavior you are proposing to be implemented into Inav. Isn't it?

https://youtu.be/sA6vngripF8

sdellava, 2023-04-10:
Yes. Exactly. I was not aware that Dji did it. Nice.

I was thinking that this feature can have other interesting uses. Being
able to show the home point means that we can show any points in the space.
So we can create a route of wp and they will be visible in sequence with an
additional  arrow pointing to the next wp.
This use can lead to "distance race" where recording the osd many pilots
can compete to run the same route in the shorter time.

rts18, 2023-04-17:
@sdellava It would seem the most difficult part of this feature would be persuading all the digital manufactures to implement OSD canvas mode into their systems.  
I guess you're aware that both analog and digital OSD, use character display port.

Most of the requirements for such a feature already exist. But I hazard a guess, that the radar HUD code is as good as it can be.

@rmaia3d It appears the Goggle 2 and 03 already have canvas mode. But limit its use to the current provision of the open source platforms, and only to a level that suits their sales.
I could envision HDzero getting onboard quickly. and Walksnail being the last to attempt it. As yet, they still don't support iNAV custom OSD.

sdellava, 2023-04-17:
@rts18 I'm not sure I understand your concern. I fly INAV and Walksnail and use radar mode which obviously uses characters to represent home point and uav position.

My proposal is to change the display of the home point, or other wp, according to the actual view.

The actual view is always done by "writing" characters in the space made by X columns and Y rows of characters.  Each point p(X,Y) in the grid corresponds to an angular range (horizontal and vertical). Thus the processing of the 3D map is simply to calculate at which point p(X,Y) it is appropriate to display the home symbol, depending on its location and the current view.

The only problem I really see with digital systems using the MSP protocol to get the data and display the OSD is that the refresh rate will not be high and this may make the 3D mode useless for free style. But I don't think anybody cares about a 3D map in free style. The use of this feature is certainly useful in Long Range and for aircraft.

rts18, 2023-04-17:
>I'm not sure I understand your concern. I fly INAV and Walksnail and use radar mode which obviously uses characters to represent home point and uav position.

I was simply conveying the limitations of a character based OSD, comparing to the higher visual accuracy of a pixel drawn canvas display.
The present method jumps from character to character, and not smoothly from pixel to pixel. This limitation constrains HD vertical accuracy, from between 18 to 22 steps, depending on the system, and even less for analog. 

> So we can create a route of wp and they will be visible in sequence with an
additional  arrow pointing to the next wp. 

Isn't that what this present setting already does.
* `osd_hud_wp_disp`
 `osd_hud_homepoint`

sdellava, 2023-04-17:
@rts18 more clear now .... and .... I totally miss the **hud** !!!

So yes, I guess it it what I was looling for and therefore, for the moment, I close this feature request.  

May be in the future I'll reopen if I'll find some additional need. 

Thanks all for the contribution.

rmaia3d, 2023-04-18:
> @sdellava It would seem the most difficult part of this feature would be persuading all the digital manufactures to implement OSD canvas mode into their systems. I guess you're aware that both [...]

Just a correction, Walksnail does fully support the Inav OSD, much better than the O3 compatibility mode. Only thing that Walksnail does not support (for now), concerning the Inav OSD, is custom fonts. But the default font includes all current icons and glyphs that Inav uses. The dev team at Walksnail is already working on adding custom font support and it will come in a future firmware update.

Also, I'm not sure what your concern regarding a "true canvas mode" for this feature is. Granted, the character positioning has a "lower resolution", going pixel by pixel would be far smoother, but the current character grid implementation does not impede such a feature.

sdellava, 2023-04-18:
@rmaia3d thank for your clarifications.

I was aware about the "true canvas mode" but I'm not sure it will be implemented by WS. Maybe HDZero mostly used for race, but WS is much more interesting for long range (with both quads and airplanes) and freestyle. In this two cases the "pixel mode" could impact on the overall video quality requiring much more bandwith to transfer data, than the simple "characther mode" with low refresh rate.

Anyway, for long range,  that was my primary focus, the overly precision is not required. We just need an indication where the home is, just to simplify the orientation in from a new point of view.