iNavFlight/inav #8945: MultiRotor : (on landings and) after disarm the QUAD tilts over instead of cutting down the motors

Pairan, 2023-04-03:
## Current Behavior
The 7" Quad is flashed with regular INAV 6 and since then has changed dramatically in behaviour after disarming as after landing properly. I looks like the quad doesnt figure that it has been disarmed and tries to compensate ( ... and overshoots).

Mostly it flips over and the props spin aggressively for 1-2secs ... and a such most likely kill the props. 

I didn't tune the rates or such. The defaults for a 7" quad was applied and nothing more (but flight modes and adjustment options for in-flight and such).

## Steps to Reproduce
1. Launch the quad
2. decide to land
3. try to disarm near the ground
4. hope that it goes well (this time)

## Expected behavior
After disarm I would have expected to simply shut everything down and DO NOTHING more than this.

## Suggested solution(s)
Simply shut down motors on DISARM as it used to be ...

## Additional context
I had think I also had it with INAV6 RC2 but I took it as me beeing clumsy on landings -.- But as it now happened several times yesterday I thought of reporting back in. Can anybody confirm this behaviour ?

- FC Board name and vendor: **MAMBAH743_2022B** from DIATONE
- INAV version string: latest INAV 6(will provide that later)

breadoven, 2023-04-03:
What reason does it give for the disarm, Switch or Landing ? Do you let the quad actually settle on the ground for 1-2s before disarming or try and disarm the moment it touches the ground or slightly before even ?

Pairan, 2023-04-03:
I want to land- not to disarm at 1m alt and let it fall into the dirt. And even in that height it would Tipp over and not simply shut down. 

Only way to land properly for now is to grab the quad and disarm. Everything else will risk to spoil the gear

Jetrell, 2023-04-03:
@Pairan Can you leave your Diff All settings? And what flight mode were you in when you disarmed.

Can you clarify, from both your posts.. I'm wasn't sure which one it is.. Disarming near the ground, or landing then disarming, which was causing the issue ?

>try to disarm **near** the ground

>I want to land- not to disarm at 1m alt

Pairan, 2023-04-04:
The effect is the SAME on both landing and disarming near ground or just disarming! The thing is that the disarm doesn't cut the motors immediately and then tries to balance the movement after that. 
Hard to describe ... guess I'll make a video of it later today to make the effect clear

Diff coming up:
```
# version
# INAV/MAMBAH743_2022B 6.0.0 Mar 22 2023 / 10:11:50 (084101ac)
# GCC-10.2.1 20201103 (release)

# Mixer: motor mixer
mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# features
feature -TX_PROF_SEL
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper
beeper -RUNTIME_CALIBRATION
beeper -ON_USB

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map
map ERTA

# Ports
serial 1 2 115200 115200 0 115200
serial 2 33554432 115200 115200 0 115200
serial 3 0 115200 115200 0 115200

# Modes [aux]
aux 0 0 0 1500 2100
aux 1 2 1 1475 2100
aux 2 11 1 1125 1250
aux 3 10 1 1375 1475
aux 4 42 2 1100 1325
aux 5 43 2 1325 1675
aux 6 44 2 1675 2100
aux 7 13 4 1800 2100
aux 8 26 0 1700 2100
aux 9 27 1 1400 1475
aux 10 47 6 1600 2100

# Adjustments [adjrange]
adjrange 0 0 5 1125 1275 14 3
adjrange 1 0 5 1275 1425 15 3
adjrange 2 0 5 1400 1550 16 3
adjrange 3 0 5 1575 1750 10 3
adjrange 4 0 5 1700 1875 11 3
adjrange 5 0 5 1850 2025 12 3

[... 172 lines cut ...]
set dterm_lpf_hz = 80
set dterm_lpf_type = PT3
set mc_iterm_relax = RPY
set d_boost_min =  0.800
set d_boost_max =  1.200
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 75
set rc_yaw_expo = 75
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set bat_cells = 6
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 315
set battery_capacity = 4200
set throttle_idle =  5.000
```

Pairan, 2023-04-04:
So far I didn't let INAV time to figure, if we might have landet (landing in Acro) as it uses to wind up and then flips. Will have to check that on some grassy place so it won't kill the props (in case).
Task: Landing by descending and finally lower the thrust : INAV should figure "we" have landed? right?

I land by descending and slowly balancing the ground effect until I disarm. Will try to raise throttle after disarm next time.

Perhaps I report in this evening

breadoven, 2023-04-04:
> What happens if you raise the throttle after arming to enable airmode, lower the throttle back down and then disarm ?

By this I mean start with the quad disarmed. Then arm, raise the throttle until airmode activates, then lower the throttle, then disarm. Do it on a flat surface ideally. Does it still flip ?

Pairan, 2023-04-04:
Nope ... I recorded several landings and it didn't flip these times. Sometimes it had a hard time to detect it had landed and winded up some motors BUT then I disarmed and it was okay. Did nine attempts and nothing bad happened. 

Guess for now we can close the ticket as I can't get it to do what it did yesterday. 

Anyways ... thanks for helping and educating me on this :)

Pairan, 2023-04-09:
[image]

there‘s still something going on- after Position hold and landing I disarmed and then the quad started to flip and tried to get home! I was 1m away- it didn’t stop although I had tried to disarm several times. 

Two props melted due to the heat on the motor shaft, my hand got burned by one motor and I had to rip out the power lead while two motors spinned 100%
My left hand cam to close to a prop and for now it doesn’t hurt- just bleeding and feeling a bit odd. 

Guys… perhaps I can pull a log from the h743 but for now I will get back to iNav 5 where disarm stops everything.

breadoven, 2023-04-10:
A log would be very useful. Almost sounds as if disarming caused it to Failsafe and RTH. In that case without a Kill switch set up it would be impossible to disarm. Log should confirm if this was the case.

Pairan, 2023-04-10:
Sounds reasonable- I will clean up the quad later and see, what I can pull off the memory then

Pairan, 2023-04-10:
That's what I pulled of the quad - perhaps it can give any clue

[blackbox_log_2023-04-10_194606.TXT.zip](https://github.com/iNavFlight/inav/files/11192646/blackbox_log_2023-04-10_194606.TXT.zip)

breadoven, 2023-04-10:
Well according to the log, 12/12, you simply selected Failsafe mode and RTH mode at the same time since they're both activated by the same switch (as confirmed in the Diff posted above). So the quad tried to RTH because that's what is told to do .... according to the log. However, shortly after selecting Failsafe the log file doesn't seem to record correctly, a few broken records but mainly blank, so something appears a little odd where that is concerned. Not sure why Failsafe and RTH modes are on the same switch given Failsafe is set to RTH anyway. This makes no sense. This is the the second time I've seen this erroneous setup recently. Has it been recommended somewhere ?

And since there log continued to record something it appears it never disarmed. There was no disarm flag at the end of the log file either.

breadoven, 2023-04-11:
I take it this didn't climb during the RTH because instead it flipped upside down when RTH started. This is how it appears from the log at least. Is this what happened ?

Pairan, 2023-04-11:
I would say that is what happened- ... one side of the quad gets more thrust and as such it flips ... then the "tries to climb" perhaps

Jetrell, 2023-04-25:
I had this happen today too.
I've never tried to disarm just above the ground when a quad was in the _RTH landing phase_. 
But I did that today. And it disarmed, then flipped over to the right.

It sounded like the motors shut off on disarm. But the attitude control was still active.
I check that log file and it showed only a little after the disarm, but not the flip..  Because the flip over wasn't instantaneous. There was about a 0.2 second delay after the motors stopped on disarm, before the flip over occurred. 

MAMBAF405_2022A
INAV 6.1

Jetrell, 2023-04-26:
@breadoven @MrD-RC Concerning this issue and the SAVED SETTINGS after disarm.. Does this save reboot the FC ?
The reason I ask, is because I've noticed in 6.1, the ESC communication connection tune occurs after the FC is disarmed. Just like it does after the initial FC bootup.
And I was thinking that the flip over could be connected to the eeprom save and reboot while the MC is airborne. #8907

MrD-RC, 2023-04-26:
No, the save when your see settings saved does not reboot.

breadoven, 2023-04-27:
I did wonder if resetting the PID accumulators immediately before disarming would solve this tipping problem. Effectively kills airmode so less likely to have motor surges happening at the point the motors are shut off. Can't do any harm so worth trying if this is a common issue.

Jetrell, 2023-04-27:
> I did wonder if resetting the PID accumulators immediately before disarming would solve this tipping problem. 

To be honest. I'm seeing discrepancies that don't seem to indicate error accumulation.

In this landing the quad was disarmed 10cm above the ground in Angle mode.. Note the G shock as it touch down. There was no flip over.

[image]

While in this RTH_LANDING phase, the quad was disarmed about 20cm above the ground. Without Air mode enabled. (I never permanently enable it).
It stayed in `navState` RTH_LANDING after disarming.. And I had the goggle lifted watching this landing... It actually hit the ground with the motors stopped, before the left motors start up again and flip it over, then shut down.. The flip over was after the log stopped recording.. Pity the logger didn't keep recording for a few seconds after disarming.

[image]

I'm seeing a lot of weird stuff lately, like i touched on in the earlier posts.. And I'm still trying to track down these issues.... Without them being consistently repeatable.. I'm thinking it may even be related to the CRSF's RF link to the FC.

e.g. today, I did a FS test, by turning off the radio. It flew most of the way back home.. Then I turned the TX back on.. I regained pitch, roll and yaw control. But the throttle got stuck a 47%. The logs showed the `rcCommand[throttle]` to be at that value, even though I'm 100% certain I had it pull low, to try to stop it climbing.
The RF link was recovered immediately, according to the OSD.. But the log showed it took 8 seconds for me to regain manual throttle control again.
[image]

Jetrell, 2023-04-29:
It appears to be related to - `set stats = ON`... I noticed you have it enable too @Pairan 

@MrD-RC and @breadoven If you guys get the chance at some point. Would you be able to have a look how this feature is tied into the Disarming process.
The function seems to effect both Multi Copter and Fixed wings, But the results are worse with MC's.
And its only seems to be an issue 1 time in 5.... Like its caused by a race condition in the scheduler at disarming.
The effect is worse with Dshot than PWM protocols.

Pairan, 2023-04-29:
Yeps, it’s activated. Both on fw and mr

Jetrell, 2023-05-13:
@breadoven @Pairan  Did either of you get a chance to confirm whether `set stats = OFF` would fix the disarm _flip over_ / _reboot_ issue in your cases?

Pairan, 2023-05-13:
Nope … after I removed the rth + failsafe combo it didn’t occur any more for me. Even installed a switch for the “kill switch mode” 

yesterday I upped mine to v6.1 

I didn’t have any flips so far. But that doesn’t guarantee they’re gone. Just nothing new on that from my side

steviegeek, 2023-06-24:
I experienced exactly this issue today on a 7" quad. Are we any closer to a solution please ?
This was using CRSF and a H743 FC
ARM switch was OFF
All NAV modes were OFF
Throttle was low
craft inverted, props racing drilling hole in field :-) challenging to get the battery disconnected.

# version
# INAV/MATEKH743 6.0.0 Mar 22 2023 / 10:48:04 (084101ac)
# GCC-10.2.1 20201103 (release)

oldrootbeer, 2023-06-24:
I can confirm this as I was present and it was a very dangerous situation. Disarm should mean disarm.

Jetrell, 2023-06-25:
@steviegeek @oldrootbeer Don't stay on 6.0.0 with any H743 controller.. There was a fix implemented in 6.1 to alleviate most of this problem with those controllers.

oldrootbeer, 2023-06-25:
> There was a fix implemented in 6.1 to alleviate most of this problem with those controllers.

Thank you. You say "most of this problem"? Is 6.1 safe wrt disarming?

breadoven, 2023-06-25:
> I experienced exactly this issue today on a 7" quad. Are we any closer to a solution please ? This was using CRSF and a H743 FC ARM switch was OFF All NAV modes were OFF Throttle was low craft inverted, props racing drilling hole in field :-) challenging to get the battery disconnected.
> 
> # version
> # INAV/MATEKH743 6.0.0 Mar 22 2023 / 10:48:04 ([084101a](https://github.com/iNavFlight/inav/commit/084101acd78ae5bfc6221458acb6f8ee64fadb6a))
> # GCC-10.2.1 20201103 (release)

Would be useful if you could provide a Diff and a log would be useful also.

Jetrell, 2023-06-26:
> You say "most of this problem"? Is 6.1 safe wrt disarming?

6.1.0 and 6.1.1 are safer options.. But there are still some combinations of user settings that can cause problems.
Like @breadoven said, providing a DIFF file of your settings may be useful to find any potential issues.

steviegeek, 2023-06-26:
@Jetrell 
@breadoven 

I've attached the DIFF ALL file, unfortunately I do not have logs (sdcard corrupt/damaged)
Thank you for your interest in this problem, very much appreciated.

[DIFF ALL 230626.txt](https://github.com/iNavFlight/inav/files/11865976/DIFF.ALL.230626.txt)

steviegeek, 2023-06-28:
@Jetrell 
@breadoven 

Hi guys, will you have the chance to cast your eyes over this DIFF file?
all suggests are most welcome
Thank You
Steve

Jetrell, 2023-06-28:
@steviegeek At present, there are a couple of settings that I've found to cause a conflict and a possible disarming flip-over problem.. But in your case, neither are present.
So load 6.1.1... It should be fine.. But as always.. Use caution for the first couple of flights.

steviegeek, 2023-06-29:
Thank You,
I'll do that update and test it, very carefully 🙂
Probably on Saturday morning at a club field for safety
Much appreciate your help 👍

breadoven, 2023-06-29:
I assume you're not using ELRS ? It's just that you're arming channel is 7 (Aux 2) rather than the 5 that is required for ELRS. Not sure if this is true for Crossfire or not.

I'm guessing this is the same problem as above ... when you went to disarm or did something else Failsafe occurred which makes normal disarm impossible. Was there any indication of failsafe in the OSD ?

And I assume you've added the Killswitch mode since you had this problem ?

steviegeek, 2023-06-29:
@breadoven 
It's CRSF, channel 7 has been my standard arming channel on everything since I got CSFR 5-6 years ago
All CSFR channels are full resolution so no need for a dedicated arming channel like ELRS

FS shows in OSD almost two minutes after it has been inverted and trying to reach Australia 
It landed in ANGLE + ALTHOLD, I turned of ALTHOLD but would still not shutdown motors.

A 'sanity mode' of 'high motors' + 'no altitude change' = 'shutdown'  would at first thoughts seem to be a good idea but the actual criteria would need more detailed thought.

breadoven, 2023-06-29:
Just to confirm:

1. You landed in Angle + Althold ... then tried to disarm ? At this point it flipped and didn't disarm so you shut off Althold but this made no difference, motors continued to run ?
2. FS was showing in the OSD for 2 mins after you tried to disarm and the quad flipped over or it only appeared 2 mins after you tried to disarm ?

steviegeek, 2023-06-29:
**_You landed in Angle + Althold ... then tried to disarm ? At this point it flipped and didn't disarm so you shut off Althold but this made no difference, motors continued to run ?_**

Yes, tried to disarm as soon as it contacted the ground.

**_it only appeared 2 mins after you tried to disarm ?_**

Yes, OSD video available if it helps

breadoven, 2023-06-29:
The OSD video would be useful thanks.

steviegeek, 2023-06-30:
Landing video here..........
https://www.dropbox.com/s/pw5xq5pbc2qnbhy/Falcon7%206.0%20landing%20issue.mp4?dl=0

Pairan, 2023-06-30:
Yeah … that’s how mine went down too.

breadoven, 2023-06-30:
Hmmm, looking at the video it does appear that this simply crashed into the ground and rolled over inverted (It "landed" with 20 mph ground speed after all). This wasn't a case of it flipping when disarmed after a normal vertical landing since it didn't disarm. If you did correctly disarm it then this would appear to be the only issue here. The Failsafe happened because the Rx signal was lost after a minute or so, so INAV worked as expected in this case unless there was no good reason for the signal dropping I guess (Tx turned off, broken Rx/antenna ?).

Exactly when in the video do you think you attempted to disarm the quad ... before the initial impact with the ground or after it ended up inverted ?

Jetrell, 2023-06-30:
> This wasn't a case of it flipping when disarmed after a normal vertical landing since it didn't disarm. 

I agree.. 
Not at all like I've experienced, as I posted here - https://github.com/iNavFlight/inav/issues/8945#issuecomment-1525723136

steviegeek, 2023-06-30:
The part of this incident that caused me concern was that when it was inverted on the ground (however that happened) the motors continued to run and I was not able to shut them down.
Which brings me to ..
**_A 'sanity mode' of 'high motors' + 'no altitude change' = 'shutdown' would at first thoughts seem to be a good idea but the actual criteria would need more detailed thought._**
Should (is there already) some code that covers this? if not then should we start a new request for it?

In addition to 'disarm' Ardupilot has a 'Motor Emergency Stop' which overides all else. Could INAV include something similiar please for safety reasons ?

Thank you

breadoven, 2023-06-30:
INAV has Killswitch mode. You have it configured to use according to your Diff (Channel 13 ?). This should disarm in all circumstances including during Failsafe (when the normal disarm switch is inhibited).

However, the problem here is Failsafe was not active when you tried to disarm, you also had a valid RC signal according to the video so there's no reason why the normal Disarm switch shouldn't have worked. Given that the AltHold mode switch seemed to work it's hard to see why the Disarm switch also didn't work before the Failsafe kicked in. From the way INAV behaved it appears it never received the Disarm switch command although without a log it's hard to know. A Transmitter log might show the channel status if you have one ?

steviegeek, 2023-06-30:
@breadoven
@Jetrell 

Firstly thanks for confirming how the kill switch mode works, 'disarm in all circumstances'
I'd been struggling as the documentation indicates it modifies how the disarm switch works but was not really definitive.
But I must ask, does 'in all circumstances' really mean total motor shutdown always?

I intend to run more tests tomorrow, with blackbox logging.

- land in POSHOLD
- land in ALTHOLD
- land in ANGLE
- hover in POSHOLD a few feet high, disarm switch should not have any effect (nav_overrides_motor_stop = ALL_NAV), kill switch should drop it to the ground (set disarm_kill_switch = ON)

I'm sorry that I do not have a clearer picture of exactly what happened.
I appreciate all your help and hope that with 6.1.1 and greater skill on my part all this can be left behind. 

Thanks again
Steve

oldrootbeer, 2023-06-30:
A sanity check is preferable in any rc link lost situation I guess. Regardless of any failsafe mode. Who knows what the craft is doing, where it is and how dangerous it is behaving.

breadoven, 2023-06-30:
`disarm_kill_switch` just determines if the normal disarm switch disarms regardless of throttle position (when set to ON, the default) or only if the throttle is low (set to OFF). So you don't want to try to disarm in Poshold 2 feet above the ground because it will just disarm and drop. Killswitch mode on the other hand will disarm always regardless of other settings or Failsafe being active. `disarm_kill_switch `and Killswitch mode are probably a bit confusing because although they are related to disarming they are fundamentally different things.

`nav_overrides_motor_stop `has nothing to do with disarming, it just affects whether the motor stops or not when the throttle stick is low during Nav modes using Auto throttle.

b14ckyy, 2024-03-21:
Should not be an issue anymore in 7.1 after https://github.com/iNavFlight/inav/pull/9681 
Please test and if this is still an issue, we can reopen.

closed 2024-03-21 by b14ckyy