iNavFlight/inav #8960: Add some kind of scaling for differential thrust on planes with multiple motors

Painless360, 2023-04-07:
## Current Behaviour
At he moment, the motors can be setup in the mixer and connected separately with a little rudder conrtol mixed into those two channels to provide a little differential thrust. However, there can be issues. At very low throttles, i.e. when coming into land and gliding, a little rudder input can spin up one motor enough and cause a yaw that stalls the aircraft into the ground.

## Desired Behaviour
It would be useful to have some kind of scaling for differential throttle so that when the throttle is below a certain threshold, the effect is reduced and ultimately turned off to avoid these issues. Or some kind of scaling related to throttle to help minimise the potential in flight for an issue.

## Suggested Solution
Ideally a setting in INAV to do this by default would be useful and adding the ability to turn on and off differential thrust via the programming framework would be nice too.

## Who does this impact? Who is this for?
All pilots with models with more than one motor who want to use differential thrust safely.

## Additional context
Thanks for considering this. Issues like the one covered above are why I (and my flying buddies) stopped using the feature in INAV.

Helipro, 2023-08-15:
That would be nice an on/off capable differential thrust via programing. This is necessary for water planes like my Avios Albatross and other float planes that I fpv fly.  The Albatross on the water requires a lot of differential thrust, with the rudders to keep it straight until the pontoons come out the water.  In the air I prefer to use the rudder mostly with some slight differential thrust.
I propose 3 check boxes next to the motor mixer.  1 box for each profile that can turn on/off differential thrust in that profile.   By switching profiles in flight you can turn differential thrust on and off easily.   

My work around involves edgetx to vary the amount of differential thrust for each mode.  I put the rudder on a separate channel not stabilized by inav so my rudder and differential thrust are not linked and I can turn the differential thrust on and off.

MrD-RC, 2023-08-15:
The checkboxes are a bit too restrictive. Having an active condition (as the servo mixer does) would allow much more flexibility with the motor mixes. Not just a change for differential thrust. 

With this, you could simply switch between motor mixes, and levels of differential thrust, via an active logic condition. This could be a switch, or even altitude triggered.

Helipro, 2023-08-15:
> The checkboxes are a bit too restrictive. Having an active condition (as the servo mixer does) would allow much more flexibility with the motor mixes. Not just a change for differential thrust. [...]

So true!  That would definitely be the best solution. I already use logic for the vtx power and override arming safety checks when craft flying.  I just love inav and the community  enjoying its growth so much over the years!  Thank you guys!

ChrisBalmforth, 2023-10-22:
My requirement is to enable differential thrust on the ground for taxiing and disable it in the air for flying. Is there a fundamental reason why the motor mixer can't have logic conditions applied like the servo mixer?