iNavFlight/inav #8961: Motors not working on Speedybee f405 v3

RGBTyp, 2023-04-07:
I tried installing inav 6.0 on this flight Controller. 

The first thing i tried is setting up the motor outputs. No matter what protocoll i choose none of them will spin. 
I have already set up another quad with inav so I am not completly new but i must have forgotten something because else i would not be the only one with this issiue. 
The motors work just fine in betaflight. 
Is there anything else that I need to do other than enabeling motor output and choosing the protocoll?
I have changed no other settings.

Demoskin, 2023-04-24:
Motors working but, with blheli_s only. With Blujay didn't start. Biderection not working too, Protocol Multishot and Oneshot125 only. Dshot300/600 not supported.

Trankilo0255, 2023-05-16:
> I tried installing inav 6.0 on this flight Controller.
>
> The first thing i tried is setting up the motor outputs. No matter what protocoll i choose none of them will spin. I have already set up [...]

I have exactly the same issue...didi you manage to solve it in the mean time?