iNavFlight/inav #8962: FR: Normal autolaunch with manual throttle in addition to nav_fw_launch_thr and the new manual throttle.

Thieu-s, 2023-04-07:
## Current Behavior
In a autolaunch situation, when "nav_fw_launch_manual_throttle = OFF", the throttle will go to "nav_fw_launch_thr" after launch is detected.
When "nav_fw_launch_manual_throttle = ON", the throttle will instantly go to "stick position throttle" after arming.

## Desired Behavior
Same as above but an extra option where the autolaunch behaves like the "nav_fw_launch_manual_throttle = OFF" but the throttle behaves like "nav_fw_launch_manual_throttle = ON" **_after_** the launch is detected.

## Suggested Solution
Add a option in "nav_fw_launch_manual_throttle" where above function is available.
Not just nav_fw_launch_manual_throttle = ON or OFF but something like HYBRID added.

(I think this is what needs to be added/changed in the code... I'm not a top programmer in c).
Add a function where this:
if (elapsedTimeMs > motorSpinUpMs) {
rcCommand[THROTTLE] = launchThrottle;
is replaced whith this: 
if (elapsedTimeMs > motorSpinUpMs) {
rcCommand[THROTTLE] = rcCommand[THROTTLE];

## Who does this impact? Who is this for?
Everybody who use autolauch and likes use all the built-in features like "Idle Throttle",  Idle Throttle Delay", "Motor Delay", "Motor Spinup Time" and "Launch Timeout" after the launch has detected instead of when raised the throttle like in "nav_fw_launch_manual_throttle = ON".

## Additional context
Depending on the wind and different battery sizes, sometimes I have to launch at full throttle, but only for the first couple of seconds, then I want the autolaunch to continue but on a lower throttle (lower the throttle stick).
On calm days or with a light weight battery, I can launch at a lower throttle from the start (throw).
This way I can still throw the wing overhead with two hands and don't have to care about throttle management and not have a prop at full throttle while putting down the radio, holding the wing with one hand (still at full throttle!) and then launch overhead, all before the timer ends.

MrD-RC, 2023-04-07:
I’m think a better option would be for autolaunch to automatically reduce the throttle to a specified climb rate. It would make more sense than a manual throttle. It may take some time to tune. But it would be more consistent when set up.

Also, this feature request is no different to https://github.com/iNavFlight/inav/issues/8891. Please don’t just create new issues to bump them. The original issue could have just been modified.

Thieu-s, 2023-04-07:
Sorry, I just watched your video on how to do a feature request, and wanted to do it like it supposed to.

Why not just best of both? Use all the good things of the old autolaunch AND the possibility to change the throttle during autolauch like the new manual throttle!
I expected this feature a very long time ago, simply because I think it's just the best throttle behaviour during launch.

MrD-RC, 2023-04-07:
I disagree. I think the best throttle behaviour during autolaunch would be automatic. Providing enough power to maintain a set climb rate at the launch angle. That would be the most efficient. Manual throttle would be less consistent.

Thieu-s, 2023-04-07:
People who jerk the plane before throwing don't really understand the benefits of all the settings for autolaunch I guess...
(yelled so many times to my screen when watching a video of a jerker, JUST THROW IT!)

Anyway, your solution sounds good, but what if you use different sizes battery (weight) on the same wing? When it's tuned to adjust the throttle to a specific climb rate with a heavy battery, is the tune still ok with a battery 1/3 the weight?
Thats why I simply thought to reduce the throttle manually (doesn't need to be precise, just not full throttle) till I can sit down and get my goggles on.

MrD-RC, 2023-04-07:
I would just set it up using the heavy battery. There won’t be a huge amount of difference in weight due to hitting CG.

Thieu-s, 2023-04-07:
Ok, I'll take it!

Little background story. I want this especially for a (very) long range wing. It's loaded with battery's way above the max takeoff weight to get the absolute maximum range. So I have to launch with full throttle to get it going, but want to preserve as much of the capacity for that range. So I want to make the launch as efficient as possible. But I do sometimes fly it with "normal sized" battery's as well.

breadoven, 2023-04-07:
> People who jerk the plane before throwing don't really understand the benefits of all the settings for autolaunch I guess... (yelled so many times to my screen when watching a video of a jerker, JUST THROW IT!)

Actually the whole point of the manual throttle autolaunch is simply to take advantage of the stabilised climb out whilst otherwise doing a normal launch. It deliberately avoids relying on the auto motor start which never worked well in my case. There's always too much of a delay between throwing and the motor producing decent thrust even if all timings are minimised. And if the detection fails it's likely to end in a crash. But obviously the manual launch is a choice which only really works when you launch with one hand and have the other hand free to work the throttle.

Jetrell, 2023-04-07:
>It deliberately avoids relying on the auto motor start which never worked well in my case.

If I remember correctly. Your case was using the side throw method.. 
I agree, that the launch detector can be hit and miss with that launch method.

Thieu-s, 2023-04-08:
>There's always too much of a delay between throwing and the motor producing decent thrust even if all timings are minimised. And if the detection fails it's likely to end in a crash.

Minimizing all timers can actually introduce a delay in starting the motor, especially with a big prop on a underpowered motor (for that prop). The motor has trouble starting if it tries to go from zero to say 80% instantly. It might even desync and not start at all.
A "Idle Throttle" of 20% and a "Motor Spinup Time" (not "Motor Delay") of 100ms prevents this and you don't even notice the delay, it feels instantaneously! I have literally done hundreds of autolaunces, side throw on smaller wings, overhead on bigger ones and football-throw style with normal planes, and never had a single fail since using this set of settings. I use them on all my planes, big or small, underpowered efficient setups or powerhouses. It just works!

> And if the detection fails it's likely to end in a crash.

You can lower the "nav_fw_launch_velocity" and "nav_fw_launch_accel" to match a slower throw.
You might have to increase the "nav_fw_launch_max_angle" value, if you throw outside this angle, autolauch fails. I just set it to 90.
 
>But obviously the manual launch is a choice which only really works when you launch with one hand and have the other hand free to work the throttle.

Indeed, and that's why I would like to see a hybrid between the manual and auto. Sometimes you need three hands...

Thieu-s, 2023-04-08:
Sorry, I'm a little stubborn...

I fixed it in programming...

[image]

If I put the throttle stick to full throttle whilst in autolaunch, the autolaunch will take care of the throttle.
As soon as the throttle stick is lowered (below 1800 in my case) I have manual control.
But only when in flight (above 555cm/s (20km/h)) to prevent the throttle from full when it's still on in my hands.

Just tested, works great! Exactly what I wanted!

closed 2024-03-21 by b14ckyy