iNavFlight/inav #8964: Very much needed feature!!

Lalbabu2001, 2023-04-08:
I build a quad running inav firmware and due to the yaw and roll sticks are more stiff there is a problem of controlling i want my quad to be flying smooth as it can,( and most of our community users doesn't want high maneuvers) whenever I push sticks and want little bit of change in direction ( in terms of value=150~200) the transmitter sends value to 500~600 which makes the quad very uncontrollable..

 i dont need that much value anytime i want when i push pitch stick to full forward , the flight controller limits it to 200(or a desired value) and the same for  roll and yaw.

## Suggested Solution
You may add a feature to trim the transmitter signal , for an example if i select 200 value for pitch the firmware set boundries for pitch from 1300 to 1700 and ofcourse 1500 will be the middle value. and during flight when I push the pitch stick full forward the transmitter will send the value 2000 but the firmware should limit it to 1700 and send 1700 value to output. in case of full backward of pitch stick it should be 1300 and the same configuration for yaw and roll just leave thottle as it is throttle should transmit full value from 1000-2000. YOU can add a tick mark button to enable or disable this feature under configuration tab or whatever you fit. so that in case of caliberation required let's say accelerometer trimming user can jump back into full stick configuration and after trimming they can get this feature back after enabling the tick option via configrator.

This feature will help all the users of any category, and specially who are new learners, and most for who makes their own diy transmitter (because a big part of budget goes into transmitter ) and repaired the old one because aftermarket joysticks are not precise enough.

Hope my suggested feature will bring increament in inav community! Please implement it.

MrD-RC, 2023-04-08:
This doesn’t need a feature. 

You can increase expo so that the mid stick is less sensitive. 

You can decrease rates so that the craft rotates slower. 

You can limit angles in angle mode to only move a set amount. 

You just need to set up INAV for how you want to fly.

Jetrell, 2023-04-08:
@Lalbabu2001 INAV 6.0 has `rc_filter_auto = ON` enabled by default..
Along with @MrD-RC  suggestions, you can also increase the `rc_filter_smoothing_factor` and/or reduce `rc_filter_lpf_hz` to provide smoother control over your quad.
The settings for this feature are in the Receiver Tab.

closed 2023-04-08 by DzikuVx

Lalbabu2001, 2023-04-08:
Hello Darren, nice to hear from you
May you elaborate please, i think this needs to be explained in a broader
way, as i didn't find good information regarding that. Only a little bit of
explain only is over internet.