iNavFlight/inav #8968: Automatic flaperons in certain flight modes

robustini, 2023-04-11:
## Current Behavior

It would be very interesting if flaperons were automatically tied to gps or air speed, i.e., they would go into action when, for example, gps speed is below a certain speed, but not in takeoff where perhaps they can be extended manually.

## Desired Behavior

They should be automatic only in some flight modes, escliuding Acro, Manual and Angle.
By setting, for example, 13 m/s ground speed the flaperons go into action below that speed.

## Suggested Solution

Two settable parameters:
1) speed below which the flaperons are automatically activated.
2) speed at which the flaperons extend (in degrees/second)

DzikuVx, 2023-04-11:
@robustini this can be done with Programming Framework without any code changes 
https://github.com/iNavFlight/inav/blob/master/docs/Programming%20Framework.md

robustini, 2023-04-11:
Top, thanks!