iNavFlight/inav #8973: Altitude Hold/Surface mode does not seem to use Benewake TFMini Pro readings

Chiefmikescott, 2023-04-11:
## Current Behavior
I am flying a Source One V3, Speedybee F405V3 stack with a Benewake TFMini Pro for the purposes of indoor altitude hold. The quad does not seem to perform any better with or without the Rangefinder and in ALTHOLD mode it oscillates between 0-3m. The Rangefinder is showing up in the sensor tab accurately. 

## Steps to Reproduce

1. I have tried to isolate whether the quad's behavior is utilizing the Rangefinder's data, so I am flying without GPS by disabling nav_extra_arming_safety.
2. I tried disabling the barometer to isolate the Rangefinder further, but ALTMODE did not show when the barometer was disabled.
3. I flew without the Rangefinder and with the Rangefinder, and the flight characteristics did not seem to be any different.

## Expected behavior
I expected the altitude / surface mode to hold the quad significantly better while utilizing the Rangefinder.

## Additional context
This might be a tuning issue, but it seems like it is not utilizing the Rangefinder sensor at all. Is there a way to tune the quad to perform better under ALTHOLD mode?

CLI dump paste here --> https://pastebin.com/6TgCQ3gb

---

 INAV/SPEEDYBEEF405V3 6.0.0 Mar 22 2023 / 11:16:01 (084101ac)
GCC-10.2.1 20201103 (release)

pogopo789, 2023-04-16:
Hi, I'm currently working with Altitude Mode and Position Mode so I can help you with some tips.
1. Even though ALTITUDE MODE is a combination between barometer, rangefinder, and magnetometer (https://github.com/iNavFlight/inav/blob/master/docs/Navigation.md#nav-althold-mode---altitude-hold)
but without a barometer, you can't be able to use ALTITUDE MODE. This is because the rangefinder is only working for close range on the ground and the magnetometer is just for supporting the altitude measurement. So you have to combine these 3 things if you want it to work smoothly.
2.  The SURFACE MODE needs to be turned on BEFORE the ALTITUDE MODE (https://github.com/iNavFlight/inav/issues/4776)
3.  The SURFACE MODE is support ALTITUDE MODE and POSITION MODE only so nothing happens with other modes when you enable it.
4.  The rangefinder's measurement would be joint to use in ALTITUDE MODE when it is below 2m (or 3m I don't remember ) range from the ground. You can extend that range with these 2 commands (I got it from here https://github.com/iNavFlight/inav/issues/5332)
set inav_max_surface_altitude = 1000: Max allowed altitude for surface following mode. [cm]
set nav_max_terrain_follow_alt = 1000: Max allowed above the ground altitude for terrain following mode
5.  The magnetometer needs to be isolated far from ESC and FC. Because it will be really noisy when getting close to those (believe me cause I mounted it on top of FC and ESC. It took me 2 months to try to hover the drone and when I isolated it, big difference)
6. The GPS also impacts ALTITUDE MODE cause it feedbacks the altitude from satellites to the FC. When you are indoors, GPS won't work correctly so if you testing your drone indoors, make sure you disable it.

Hope that these tips will help you.

Chiefmikescott, 2023-04-19:
Thanks for the detailed tips. Once I get the chance to put some more time into testing, I'll let you know how it went.

Chiefmikescott, 2023-04-29:
> Hi, I'm currently working with Altitude Mode and Position Mode so I can help you with some tips.
> 
> 1. Even though ALTITUDE MODE is a combination between barometer, rangefinder, and magnetometer (https://github.com/iNavFlight/inav/blob/master/docs/Navigation.md#nav-althold-mode---altitude-hold)
>    but without a barometer, you can't be able to use ALTITUDE MODE. This is because the rangefinder is only working for close range on the ground and the magnetometer is just for supporting the altitude measurement. So you have to combine these 3 things if you want it to work smoothly.
> 2. The SURFACE MODE needs to be turned on BEFORE the ALTITUDE MODE ([SURFACE mode doesn't work #4776](https://github.com/iNavFlight/inav/issues/4776))
> 3. The SURFACE MODE is support ALTITUDE MODE and POSITION MODE only so nothing happens with other modes when you enable it.
> 4. The rangefinder's measurement would be joint to use in ALTITUDE MODE when it is below 2m (or 3m I don't remember ) range from the ground. You can extend that range with these 2 commands (I got it from here [Problems with altitude hold #5332](https://github.com/iNavFlight/inav/issues/5332))
>    set inav_max_surface_altitude = 1000: Max allowed altitude for surface following mode. [cm]
>    set nav_max_terrain_follow_alt = 1000: Max allowed above the ground altitude for terrain following mode
> 5. The magnetometer needs to be isolated far from ESC and FC. Because it will be really noisy when getting close to those (believe me cause I mounted it on top of FC and ESC. It took me 2 months to try to hover the drone and when I isolated it, big difference)
> 6. The GPS also impacts ALTITUDE MODE cause it feedbacks the altitude from satellites to the FC. When you are indoors, GPS won't work correctly so if you testing your drone indoors, make sure you disable it.
> 
> Hope that these tips will help you.

@pogopo789  it worked great. Thanks for the great advice. Now I need to tune out the immediate jump in elevation when I activate position or altitude hold.

closed 2023-10-27 by DzikuVx