iNavFlight/inav #8975: Speedybee no conection MAMBA MK4 F740

stavross118, 2023-04-12:
Have a MAMBA MK4 F740 AIO fc
The target being MAMBAF722_2022B
Loaded with the latest 6.0.0 hex inav 
Won't connect on WiFi to speedybee and it is the WiFi version as it connects perfect in Betaflight so uart 4 ect...
All OK but on inav ...also tried to compile my own through
The inav resources as wanted to change the pin b03
Normally led_strip to servo has have cine whoop pavo 360
And need a landing servo leg but when compiling keep getting errors still new to this any help much apreciated
Please 🙏 any help much apreciated 
Steve

stavross118, 2023-05-20:
Just updated to 6.1.0 now baro not working set as MSP 
As well as speedy bee conection for the app not working ....might have to go back to betaflight for a while until its resolved cheers anyway guys would have been great to try inav

kirbylaroza, 2023-05-28:
After flashing the firmware, please remove the SDCard on initial try. It will proceed as usual and ask for default settings. Reinsert SDCard after.

closed 2023-06-20 by DzikuVx