iNavFlight/inav #8983: HC-100 doesn't work)

alezalez8, 2023-04-13:
Hi!
I run across to little problem: I have an ultrasonic HC-100 and an arduino board with hex "inav_rangefinder_us100_serial". But I don't understand how can I connected it to my FC (Matek H743-SLIM V3). I try to connect to DA1 and CL1, but I can't see any changes in the configurator (INAV configurator ver 6.0.0). What options I have to set and what pins of this FC I have to use? It's very important thing for me, because my quad needs to fly at the height of about 0.5 m upper terrain, and my quad  must follow the terrain. Lidar doesn't make the same behavior (depends on light and the surface of the terrain). 
Please, help me with this problem.
With my regards, Aleks

mmosca, 2023-04-13:
the hex say serial, so you probably need to connect to a uart. I suspect it uses msp.

You should probably check the documentation from wherever you got this hex for arduino.

alezalez8, 2023-04-14:
Thanks for your answer, my friend!
I tried to make this project: https://github.com/iNavFlight/inav-rangefinder  
But, when I connect to arduino board and FC by I2C, my configurator doesn't see the rangefinder...

mmosca, 2023-04-14:
I am afraid INAV no longer supports this sensor as a range finder. You may need to use an older version or pick a different sensor.

closed 2023-04-14 by mmosca

alezalez8, 2023-04-14:
I am afraid too, but it is the truth... INAV, probably, doesn't support this feature in the latest releases. 
Thanks a lot for your help!