iNavFlight/inav #8998: Support SpeedyBee F7 V3 Onboard Data Flash for Blackbox Logging

Mitch318, 2023-04-21:
## Current Behavior
Onboard Flash does not seem to be available for SpeedyBee F7V3 

## Desired Behavior
Support for onboard flash to allow blackbox logging

## Suggested Solution
Implementation of similar levels of support compared to other flight controllers

## Who does this impact? Who is this for?
All users of SpeedyBee F7V3 flight controllers that are interested in tuning and diagnostics

## Additional context

[image]

Mitch318, 2023-04-21:
[INAV_6.0.0_cli_RM_INAV_20230421_192930.txt](https://github.com/iNavFlight/inav/files/11298382/INAV_6.0.0_cli_RM_INAV_20230421_192930.txt)

Attaching the diff incase I've been silly.

mmosca, 2023-04-21:
@Mitch318

Can you flash this target and see if your blackbox shows up?

[inav_6.1.0_SPEEDYBEEF7V3_NOSD.hex.zip](https://github.com/iNavFlight/inav/files/11299000/inav_6.1.0_SPEEDYBEEF7V3_NOSD.hex.zip)

Mitch318, 2023-04-21:
Just flashed and getting the same issue. This time, however, I am now only able to select Serial from the blackbox logging device. drop down menu.

'get blackbox' in the CLI is showing

blackbox_device = SDCARD
Allowed values: SERIAL, SPIFLASH, SDCARD

I tried changing to SPI flash via the CLI and it didnt change anything.

I then put my config from a dump to see what happened and got one issue about "SD card inversion = on" or words to that effect.
[image]

mmosca, 2023-04-21:
Can you run flash_info on the cli in inav, do the same with betaflight?

I don't have a board to test with, so I made these changes based on what I could see from the bf configuration.

Mitch318, 2023-04-21:
Hi @mmosca 

So I flashed to BF and I noticed that Blackbox logging was working, I then reflashed to INAV and it is now showing on there and is now logging.

I will try to recreate the error and report back. But for now it appears to be working!

Appreciate your help!!

Mitch318, 2023-04-21:
It does still show that there is no dataflash chip found on the top right as shown before but I can live with that!

mmosca, 2023-04-21:
> It does still show that there is no dataflash chip found on the top right as shown before but I can live with that!

That is strange. Do you see the option for internal flash on the blackbox tab?

Try erasing the flash from there

Mitch318, 2023-04-22:
Yesterday when I was having the issues it only showed options for SD card or Serial. But today I checked and it now has an option for "_onboard_ SD card slot". 
Blackbox appears to be logging as normal.

To view the logs I need to go via the CLI with the "MSC" command.

Admittedly, I am new to INAV having just crossed over from BF. I am sure on BF there was an option to enter Mass Storage Device mode with a button click in the Configurator. Is this the same on INAV?

No option on INAV Blackbox tab to enter mass storage or to clear the flash.
[image]

mmosca, 2023-04-22:
> Yesterday when I was having the issues it only showed options for SD card or Serial. But today I checked and it now has an option for "_onboard_ SD card slot". Blackbox appears to be logging as [...]

Now I am confused, because the screenshot is showing the original target, not the target I attached. :)

Mitch318, 2023-04-22:
My mistake, I should have been clearer!
_"So I flashed to BF and I noticed that Blackbox logging was working, I then reflashed to INAV and it is now showing on there and is now logging."_
I should have stated that I reflashed 6.0 at that point!

closed 2023-04-23 by mmosca

mikecarr, 2024-03-07:
Any chance we can get a button in the UI that will essentially run the MSC command? Similar to Betaflight.

MrD-RC, 2024-03-07:
There already is on the CLI page.

jonathanoeijoeng, 2024-03-24:
i have same speedybee F7 v3 that has onboard flash but only get options serial and sd card. I even cannot delete or format flashchip to free up storage
[image]

alexkmarkov, 2024-06-29:
I have the same controller and with Betaflight found that Betaflight can't wright logs if it thinks it's full even when I deletet all logs, but format from PC helped. I am with INAV from yesterday - remain to be seen how INAV cup with it