iNavFlight/inav #9005: INAV 6.1 RC1 OSD Altitude Shown in Miles > 1000 ft

prelator, 2023-04-23:
## Current Behavior
Using INAV 6.1 RC1 with the OSD set to imperial units, altitude displays in feet below 1000 feet and in miles over 1000 feet. I have observed this in both the relative altitude display above your home location and the absolute MSL altitude display.
## Steps to Reproduce

1. Flash INAV 6.1 RC1
2. Set OSD to imperial units and turn on both relative and MSL altitude. My OSD mode was BF43COMPAT with a DJI O3 system and DJI Goggles V2, but I don't think this is unique to that.
3. Fly the model to over 1000 feet, or launch from an MSL altitude over 1000 feet.
4. Altitude displays are in miles over 1000 feet altitude.

## Expected behavior
 Altitude should always display in feet when using imperial units.

## Suggested solution(s)
Change all OSD altitude displays to always display in feet with imperial units. Only lateral distance should switch to miles.

## Additional context

set osd_video_system = BF43COMPAT
set osd_units = IMPERIAL
set osd_time_alarm = 0
set osd_alt_alarm = 0
set osd_dist_alarm = 0
set osd_neg_alt_alarm = 0
set osd_link_quality_alarm = 50
set osd_rssi_dbm_alarm = -100
set osd_switch_indicators_align_left = OFF

osd_layout 0 0 27 2 H
osd_layout 0 1 0 14 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 6 24 14 V
osd_layout 0 7 14 13 V
osd_layout 0 9 14 15 V
osd_layout 0 11 7 15 V
osd_layout 0 12 5 14 V
osd_layout 0 13 3 8 H
osd_layout 0 14 0 1 V
osd_layout 0 15 23 8 V
osd_layout 0 20 12 0 V
osd_layout 0 21 1 0 V
osd_layout 0 22 15 2 V
osd_layout 0 23 0 2 V
osd_layout 0 24 6 1 V
osd_layout 0 26 24 11 V
osd_layout 0 28 24 15 H
osd_layout 0 29 16 2 H
osd_layout 0 30 2 3 V
osd_layout 0 34 11 1 V
osd_layout 0 37 0 12 H
osd_layout 0 38 0 15 V
osd_layout 0 40 24 13 V
osd_layout 0 43 0 0 V
osd_layout 0 85 3 8 V
osd_layout 0 96 25 6 V
osd_layout 0 100 14 14 H
osd_layout 0 108 17 2 H
osd_layout 0 109 24 0 V
osd_layout 0 110 26 1 V
osd_layout 0 111 25 4 H
osd_layout 0 112 25 2 V
osd_layout 0 128 0 11 V

---

- FC Board name and vendor: INAV/MATEKF405TE_SD 6.1.0 Apr 21 2023 / 17:48:27 (2d7a057d)
- INAV version string: GCC-10.2.1 20201103 (release)

MrD-RC, 2023-04-24:
The BetaFlight font does not have a symbol for kft (thousand feet). This has probably been purposely set to differentiate between feet and thousand feet (which is what is being displayed).

The only options are:
1. Leave as it is and accept it's a limitation due to DJI being DJI.
2. Change all altitude readings to just use the ft symbol. This may be a little better, as the decimal will be there with kft.

Please complain to DJI. For their system to work correctly with all firmware. They need to implement MSP DisplayPort properly. This mean full support, including multiple page font, and custom fonts that can be uploaded to their goggles. Without this, you will all be stuck in a world of **?** and incorrect symbols.

prelator, 2023-04-24:
I'm not seeing it display in K feet, but miles. For example my home MSL altitude is 5800 feet. 6.1 displays 1.1, not 5.8. 6.0 properly displays 5800 feet. That's INAV, not DJI.

MrD-RC, 2023-04-24:
It won’t say kft as the BetaFlight font doesn’t have that symbol.

prelator, 2023-04-24:
The symbol isn't the issue. It's the actual numbers that are in miles.

MrD-RC, 2023-04-24:
What are you basing the numbers being in miles on, if not the symbol?

prelator, 2023-04-24:
Math.... 5800 feet = 1.1 miles. If it was in thousands of feet it should be 5.8.

MrD-RC, 2023-04-24:
I see the problem. O3 is calling the wrong function for displaying altitude. It's using **osdFormatDistanceSymbol** when it should be using **osdFormatAltitudeSymbol**. Probably because of all the messing about that needs to be done to perfectly working code to get this crap added! All because DJI are too lazy or ignorant to implement MSP DisplayPort properly. The BetaFlight font (which DJI also managed to screw up symbols on) doesn't have numbers with decimals. So another workaround had to be added. This is why it was probably added to one function. The problem is the values are handled differently. 

In short. Every other system works fine. It's only on DJI O3. It can be fixed. But please still complain to DJI. This is all their doing at the end of the day. Proper MSP DisplayPort would have just worked out the box.

prelator, 2023-04-24:
That makes sense, since it appeared to be behaving exactly like the distance from home display, which it's actually useful to have that switch to miles over 1000 feet. Figured it must be something simple like calling the wrong function somewhere.

And I don't know if it will do any good, but I posted on DJI's support forum asking for full MSP DisplayPort: https://forum.dji.com/forum.php?mod=viewthread&tid=289203&extra=page%3D1%26filter%3Dtypeid%26typeid%3D817%26typeid%3D817

MrD-RC, 2023-04-24:
Thanks for raising an issue with them. The more people who do. The more they may realise there’s a problem that _they_ need to fix. Whether they do or not is another story.

MrD-RC, 2023-04-24:
To be honest, they probably don’t even know they’re using a cut down MSP DisplayPort. They still keep calling it canvas mode 🤦🏼‍♂️

MrD-RC, 2023-04-24:
I'd added a PR which should fix the issue at https://github.com/iNavFlight/inav/pull/9007

I have tested the non-O3 side, and that's fine. I've also requested reviews from some other devs who have the O3 system.

MrD-RC, 2023-04-24:
If anyone with DJI O3 wishes to test, the targets can be downloaded at https://github.com/iNavFlight/inav/suites/12450805251/artifacts/663004480 This is 6.1 in it's latest form. So will need the 6.1 RC1 Configurator.

closed 2023-05-06 by MrD-RC

jeffalanmcclain, 2023-07-22:
I have replied to the post on DJI support pages as well begging to please support their customers for iNav and MSP Displayport.

PR #9007 (2023-04-25): Fix issue with DJI O3 showing miles instead of kFt
Fixes #9005

Tested on other systems via HITL. All fine. Should be tested by someone with the O3 system for final confirmation.
commit 5350e2f3 (2023-04-25): Fix issue with DJI O3 showing miles instead of kFt

Fixes #9005

Tested on other systems via HITL. All fine. Should be tested by someone with the O3 system for final confirmation.