iNavFlight/inav #9006: INAV 6.1 RC1 BFHDCOMPAT OSD AHI Not Centered

prelator, 2023-04-24:
## Current Behavior
Using the new BFHDCOMPAT OSD mode with the DJI O3 Air Unit and DJI Goggles V2 set to HD display mode, the artificial horizon indicator and its sidebars at not centered, but are way off to the left of the screen. Movable elements can be moved into the proper positions, but the fixed elements like the AHI cannot be positioned in the center of the screen.

## Steps to Reproduce

1.  Using INAV 6.1 RC1, set OSD mode to BFHDCOMPAT with a DJI O3 Air Unit and DJI Goggles V2 (not sure if happens with Goggles 2. Position save OSD settings.

## Expected behavior
Fixed OSD elements including the artificial horizon indicator and its sidebars should be positioned in the center of the screen in HD mode as they are in SD mode.

## Suggested solution(s)
Some of horizontal offset may be needed to ensure the fixed position OSD elements are centered in the HD canvas.

---
- FC Board name and vendor: INAV/MATEKF405TE_SD 6.1.0 Apr 21 2023 / 17:48:27 (https://github.com/iNavFlight/inav/commit/2d7a057d03a33e7cb6ff2c33673faf264f4775a7)
- INAV version string: GCC-10.2.1 20201103 (release)

mmosca, 2023-04-25:
Can you share your diff?

The elements are too far left to be a simple issue with alignment. Do you have any right aligned elements?

The image seems to sugest you are set to BF43COMPAT on INAV, but have the goggles set to HD canvas size.

prelator, 2023-04-25:
The picture above is confusing, because while it was set to BFHDCompat mode, I hadn't actually moved any of the elements from their regular BF43COMPAT mode positions. I could however move them to the right side of the screen and that would work fine. I didn't get a picture of it when I did this, but at one point I had all the moveable elements positioned around the perimeter of the screen but the AHI was still off to the left. Changing from regular BFCOMPAT to HD BFCOMPAT mode didn't make any difference where the AHI was positioned when the Goggles V2 were set to to HD display mode.

happylandings, 2023-04-26:
The HD canvas mode was added to Goggles 2. You try to apply it on goggles V2.

prelator, 2023-05-13:
Update: I upgraded to 6.1 stable and this issue is still happening. With BFHDCOMPAT mode in INAV and the HD display mode set in the Goggles V2, the AHI is still way off to the left, as pictured below. I can move other elements to the right positions (you can see my RSSI readouts on the far right side of the screen, which are the only ones I moved), but the AHI and other fixed elements are still way over on the left. If I change the offset settings for the AHI, it moves it up and down just fine, but there is no setting to move it side to side. One possible solution would be to add a horizontal AHI offset setting in future releases. Until this is fixed, HD mode will still not be usable for the O3 Air Unit and Goggles V2. Not sure if this happens with the Goggles 2 or not.

[image]

My complete diff is below:
`# diff all

# version
# INAV/MATEKF405TE_SD 6.0.0 Mar 22 2023 / 10:21:39 (084101ac)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 1 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 0 100 0 -1
smix 3 4 2 100 0 -1

# Outputs [servo]
servo 1 1000 2000 1670 -100
servo 2 1000 2000 1515 -100
servo 3 1000 2000 1595 -100
servo 4 1000 2000 1500 -100

# safehome

# features
feature BAT_PROF_AUTOSWITCH
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports
serial 3 33554432 115200 115200 0 115200
serial 4 2 115200 115200 0 115200
[... 254 lines cut ...]
set nav_fw_launch_thr = 1900
set nav_fw_launch_idle_thr = 1200

# battery_profile
battery_profile 2

set bat_cells = 4
set battery_capacity = 4000
set battery_capacity_warning = 1000
set battery_capacity_critical = 600
set throttle_idle =  5.000

# battery_profile
battery_profile 3

set throttle_idle =  5.000

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# `

dugdiamond-git, 2023-06-01:
I am also getting the same issue with 6.1. GogglesV2 and Mateksys H743WLITE.

Here is my DIFF:

# version
# INAV/MATEKH743 6.1.0 May 11 2023 / 13:29:16 (d6ce6e70)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 1 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 0 100 0 -1
smix 3 4 2 100 0 -1
smix 4 5 8 100 0 -1

# Outputs [servo]
servo 1 1000 2000 1500 -100
servo 2 1000 2000 1500 -100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 -100
servo 5 1000 2000 1500 -100

# safehome

# features
feature -TX_PROF_SEL
feature -BLACKBOX
feature THR_VBAT_COMP
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports
[... 220 lines cut ...]
set battery_capacity = 6500
set battery_capacity_warning = 2600
set battery_capacity_critical = 1950
set nav_fw_max_thr = 1800
set nav_fw_launch_thr = 1800
set nav_fw_launch_idle_thr = 1250

# battery_profile
battery_profile 2

set throttle_idle =  5.000

# battery_profile
battery_profile 3

set throttle_idle =  5.000

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

#

mmosca, 2023-06-09:
I tried to reproduce it, but no luck.

Did you guys change the video system on the cli? The configurator appears to move the AHI whenever it changes canvas size for me.

prelator, 2023-06-09:
I did it in the configurator.

Ifeo-A, 2023-07-29:
@prelator @mmosca 
I just got the DJI Goggles 2 and I've got the same issue. I'm on INAV Configurator 6.1.0

MrD-RC, 2023-07-29:
Did you reboot the flight controller after changes the video for at and reboot your DJI system?

dugdiamond-git, 2023-07-29:
> Did you reboot the flight controller after changes the video for at and reboot your DJI system?

Inav was not the issue.... It was that WTFOS that needed updating. It all works now as axpected. Thanks for your reply

MrD-RC, 2023-07-29:
I saw an email reply to this issue, which I can’t see here on the site saying:

> Inav was not the issue.... It was that WTFOS that needed updating. It all works now as axpected. Thanks for your reply

With WTFOS, don’t use BF43COMPAT or BFHDCOMPAT. You should use the WTF video type setting. You will be compromising your OSD by using the COMPAT types with WTF. They don’t support all elements and have a small area.

closed 2023-10-27 by DzikuVx

BigMacOne, 2024-08-06:
Just got goggles 3 and O3 Au. Matek 722 I nav 6.1.0. Goggles on HD canvas BTcomp setting... Same issue. AhI way off left. Initially all the fonts were a mess or not showing including question marks... After multiple re boots on things it is improving. Fonts mostly usable but horizon indicator stuck left! Old Air unit on goggles V2 with wtfos was luvly.

prelator, 2024-08-06:
I actually solved this issue on my Goggles V2 simply by power cycling both the goggles and flight controller. My mistake was thinking just changing OSD settings in the INAV configurator fully took effect without rebooting. Most OSD settings do but not switching to HD mode.

BigMacOne, 2024-08-06:
> I actually solved this issue on my Goggles V2 simply by power cycling both the goggles and flight controller. My mistake was thinking just changing OSD settings in the INAV configurator fully took effect without rebooting. Most OSD settings do but not switching to HD mode.

Early days with V3 goggles O3. I will try 'fully' re booting the kit making changes on OSD. Fingers crossed.

BigMacOne, 2024-08-07:
No luck with re boot scenario. On closer examination the entire display is left shifted ie values displayed in config screen to extreme right, appear at centre on goggles. Hence AhI is way left. Can't position anything in goggles to right of centre scree. As if 50% of display is missing! I like the goggles, light, comfortable and great display but this is going to be a head banger....

BigMacOne, 2024-08-07:
Suppose I will grow to love AhI on left eye... Like an Apache pilot monicle.

samsab1, 2024-08-18:
hello same pb here with speedybee f405 WING with inav 7.1.2 and airunit O3 + google2: air cross and horizon side bar not centred whate ever BFHD or BF43COMP. i do not  know what to do do

with F405 V4 speedybee  on a quad, no pb, O3 is centered

BigMacOne, 2024-08-18:
Happy to say I have fixed mine! Flashed FC to I nav 6.0.1. From 6.0.0. This small upgrade gives you another osd option in configurator... DJIHDCOMP. That was it. All centred. Can't comment on 7.1.2 but sure you will need the DJIHD option somewhere along the process. Good luck.

BigMacOne, 2024-08-18:
Should have added its a screen format problemo ie 4.3 or 16.9

BigMacOne, 2024-08-18:
In CLI enter    if it will accept it? 

set osd_video_system = DJIHDCOMPAT

samsab1, 2024-08-19:
# set osd_video_system = DJIHDCOMPAT
### ERROR: Invalid value. Allowed values: AUTO, PAL, NTSC, HDZERO, DJIWTF, AVATAR, BF43COMPAT, BFHDCOMPAT

BigMacOne, 2024-08-19:
That must be because of the I nav version installed...  Take it configurator and FC flash are same? There must be another command for V 7......  I can confirm on 7.i.0 that goggles V2 work fine on old air unit. Mine are wtf rooted. Check wiki for problem....

BigMacOne, 2024-08-19:
On goggles settings menu, check for canvas or HD mode... Set HD if you have that option?

samsab1, 2024-08-19:
i use inav configurator 7.1.1 and FC SPEEDYBEE F405 Wing flashed with INAV **7.1.2**; do you suggest to downgrad firmware to 7.1.1?

samsab1, 2024-08-19:
So, my last setting 

inav 7.1.2
uart5: MSP DISPLAY
OSD: BFHD

Google2 canvas: HD

don't ask why, **but now it works** as it should.....
thanks again for your help
i was struggeling since 48 hours, but now, it works, so strange.

BigMacOne, 2024-08-19:
BTW, I have a FW on I nav 7.1.0 with speedybee f405 and original air unit V2 goggles. Will be fitting O3 to it soon and goggles 3 let you know what happens re osd!