iNavFlight/inav #9009: Allow clearing trackback breadcrumbs in flight

aviphysics, 2023-04-24:
## Current Behavior
Trackback will follow return breadcrumbs for configured distance, which is sometimes undesired. Example would be that after exploring an area, RTH would trackback through the exploration, which might not be helpful.

## Desired Behavior
Trackback should allow the user to clear the breadcrumb history, so the quad will not attempt to trackback that path if RTH is triggered.

## Suggested Solution
Add "mode" option that will clear trackback breadcrumbs with an RC command.

## Who does this impact? Who is this for?
Anyone that uses trackback that may not want RTH to go back through an area. For example, a long range pilot that stops to investigate an area may want to clear trackback, so the quad will not try to return through that exploration.l if RTH is triggered.

## Additional context
I noticed this when I was testing RTH trackback and found my quad following a silly route in and around a bunch of trees that I had been checking out. If I had been in a real emergency RTH situation and the altitude was not accurate, the quad could have collided with the tree.

Jetrell, 2023-04-25:
I'm not sure if you've read the Wiki document on this feature... https://github.com/iNavFlight/inav/wiki/Navigation-Mode:-Return-to-Home

>so the quad will not attempt to trackback that path if RTH is triggered.

Trackback RTH can be cancelled using the RTH Altitude Control Override RIGHT ROLL command at which point RTH will revert to a normal RTH heading directly back home. `nav_rth_alt_control_override` needs to be ON for this to work.

aviphysics, 2023-04-26:
> I'm not sure if you've read the Wiki document on this feature... https://github.com/iNavFlight/inav/wiki/Navigation-Mode:-Return-to-Home
>
>
> Trackback RTH can be cancelled using the RTH Altitude [...]

That only works if you didn't have a failsafe RTH. If you lose RC link, you are at the mercy of the Trackback breadcrumbs.

It also overrides the Altitude, which can also be a problem.

There would still be value in allowing the pilot to clear the breadcrumbs.

MrD-RC, 2023-04-26:
I honestly don’t think anyone would remember to clear the trackback points. Well, not until after a FS where they had a wild trackback path. To be honest, there is always an element with FS that is in the hands of the foam gods. Whether that’s the path previously flown or the RTH settings.

breadoven, 2023-04-27:
https://github.com/iNavFlight/inav/pull/8698 allows you to toggle Trackback ON/OFF in flight.

However, how are you going to know if you want to use it before a FS happens ? In some cases it may be useful in others not. It was originally always intended as nothing more than a means of returning back to a previous position that had a good signal so you can regain control. There is little other reason for using it other than for a bit of fun.

aviphysics, 2023-04-27:
> #8698 allows you to toggle Trackback ON/OFF in flight.
>
> However, how are you going to know if you want to use it before a FS happens ? In some cases it may be useful in others not. It was originally [...]

The main use case I was thinking of for trackback was avoiding accidental RTH overflight of somewhere you shouldn't be flying over.

You would know to clear breadcrumbs if you just flew through an area you didn't want to fly back through, like in and around trees

b14ckyy, 2024-03-21:
A new feature is in the works that will prevent flying over forbidden areas during RTH or Failsafe. I agree with breadoven and MrD that a manual delete option does not make much sense. Ad mentioned, the main purpose of that feature is, to gain back control from a position where the craft already was with RC link and then you can take over. The amount of track back can also be reduced if the default 500m are too much. I think we can close this.

closed 2024-03-21 by b14ckyy