iNavFlight/inav #9033: INav OSD elements rendered non-center using Walksnail Avatar (4:3 mode - Skyzone o30 goggles)

AlexC176, 2023-05-04:
## Current Behavior
INav 6.0 configurator OSD screen (Avatar Profile) does not represent actual OSD rendering in goggles. Cannot place OSD elements in the top left or top right of goggle view. OSD Crosshair and artificial horizon do not render center. 

## Steps to Reproduce
1. Skyzone o30 goggles (4:3 native display) using Avatar VRX - AvatarSE_Gnd_32.37.10 firmware, Avatar Camera setting is 4:3
2. OSD - Video Format - Avatar
3. Place elements on the far left, top of the OSD
4. View goggle, elements are not rendered top left
5. Load the open source 'Walksnail OSD Tool' software and render the DVR. OSD elements are rendered over the recorded video as per the iNav Video format Avatar  

## Expected behavior
iNav elements render in the 4:3 goggle centered. Especially cross hair and artificial horizon.

## Suggested solution(s)
Something is mismatched between the iNav OSD handing and the Avatar OSD rendering related with 4:3 mode

## Additional context
Entering CLI Mode, type 'exit' to return, or 'help'

# version
# INAV/MATEKF405TE 6.0.0 Mar 22 2023 / 10:20:32 (084101ac)
# GCC-10.2.1 20201103 (release)

# dump

# version
# INAV/MATEK F405TE 6.0.0 Mar 22 2023 / 10:20:32 (084101ac)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 50 0 -1
smix 1 1 1 50 0 -1
smix 2 2 0 -50 0 -1
smix 3 2 1 50 0 -1

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1526 -100
servo 2 1000 2000 1496 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
[... 1632 lines cut ...]
set failsafe_throttle = 1000
set nav_mc_hover_thr = 1500
set nav_fw_cruise_thr = 1250
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1200
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

- FC Board name and vendor: Matek f405-wmn
- INAV version string: INAV/MATEKF405TE 6.0.0 Mar 22 2023 / 10:20:32 (084101ac)

AlexC176, 2023-05-04:
[image]
[image]

MrD-RC, 2023-05-04:
@AlexC176  Can you add the sidebars and AHI please. Then post up the images showing the VRx view of the OSD (not the rendered from the video tool). 

Have you moved the OSD in the VRx Display menu?

AlexC176, 2023-05-05:
In the Walksnail VRX display menu -> OSD position allowed me to centre the iNav OSD properly. Case closed !

closed 2023-05-05 by AlexC176

MrD-RC, 2023-05-05:
@AlexC176 When the OSD was lined up in the Display menu (with 16:9, we go all the way left). Were the guide lines in configurator in about the right place? If not a screen grab would still be useful to improve them.

AlexC176, 2023-05-06:
@MrD-RC I can confirm that the iNav OSD 4:3 lines up perfectly for placement of OSD elements. Recent firmware has added a 4:3-Full mode which fill the goggle view in the horizontal axis. In that case the a further box should allow +2 and -2 characters left and right, top and bottom are ok. Hope that helps.