iNavFlight/inav #9034: Compass-free RTH for multirotors

zoran-grujic, 2023-05-04:
## Current Behavior
Doesn't work

## Desired Behavior
Use GPS heading like fixed wing or like it is implemented in Betaflight. Functional RTH with no compass/magnetometer hardware.

## Suggested Solution
HOW it works in Betaflight?
In BetaFlight, when RTH is activated, the multirotor will go to angle mode, stop - try to hover, gain the  preset altitude, and then go in forward direction whatever it is. After some distance it will learn heading from its GPS, correct itself, and go to the home position.

HOW to improve?
All multirotors have gyros and they can use GPS to learn (when velocity is above some level) proper RTH heading. Then until the next good GPS info heading info can be maintained with gyro data only. In this way the multirotor can have sufficiently good home heading to start from and latter while moving fast to HOME it will be corrected with new GPS data. The minimum RTH speed should be set above some reasonable level.

## Who does this impact? Who is this for?
Multirotor

## Additional context

It seems like compass is not useful at all. Also, in general it is NOT reliable. It requires frequent re-calibration that most people doesn't like to loose their time into, and it is usually first sensor to give false info. Also, if pilot add/change payload must/should re-calibrate. Boring! If Betaflight has RTH without compass then INAV must do it better!

Jetrell, 2023-05-04:
>It seems like compass is not useful at all. Also, in general it is NOT reliable

INAV has more automated functionality than Betaflight.
`Position Hold` will not work without a compass.. INAV is also capable of performing Waypoint missions. 
A gyro will drift over time. So its not a reliable means of performing true automated navigation, of which RTH is only one part.

MrD-RC, 2023-05-05:
If compass is unreliable. I would say it is more and issue with the installation or calibration.

zoran-grujic, 2023-05-05:
Yes, [MrD-RC](https://github.com/MrD-RC), yes, it is always user error or bad hardware. But it happens with magnetometer more often than with any other sensor. And several people I had contact with simply do not wish to depend on it. RTH is must have function and there is no god reason why it MUST use compass/magnetometer. Making a bad step before getting the heading with GPS is better than letting so many compass-less people with no option to come home in emergency. It doesn't have to land in the 10cm circle, just come back close to home. +/-20m is sufficiently good in most cases.

Me personally, I have one multirotor with INAV and compass and, yes it does work properly - but will it work every time?

Jetrell, 2023-05-06:
@zoran-grujic More often than not, the user places the magnetometer in a location that is exposed to a weak magnetic field.
Placement is important.. Ensure the compass is always mounted above or below the motors, never between or in-line with them. 

After a calibration, the accuracy should be checked on the setup page in the configurator, to ensure **all axis's move correctly in every orientation**.

MrD-RC, 2023-05-06:
So you want this change because BetaFlight does it. Even though you have never had any problems yourself. 

The BetaFlight system is not perfect either.

fermiums22, 2023-06-09:
I fully support zoran-grujic. One of my copter could not return in a critical situation and I lost it about half a year ago. And what infuriates itself in this situation, is that inav seeing a large departure by gps, apparently continued to take the copter in the wrong direction by compass... Now I have a new drone and I check compass 3 times a day. But it's really scary.

rts18, 2023-06-11:
>  inav seeing a large departure by gps, apparently continued to take the copter in the wrong direction by compass.

The base navigation engine INAV uses, was implemented over 5 years ago. That developer is no longer part of the INAV development team. So presently, there are no developers with the time or ability to implement your request.

You dudes need to be aware.  INAV only has a small development team. With less people to implement such major code changes, in comparison to Betaflight or Ardupilot.
It would seem there is only one developer making any significant changes to INAV's navigation code. With his time obviously being limited.

>Now I have a new drone and I check compass 3 times a day.

That's the safest way. The software is free, but the hardware is definitely not. So use caution.

zoran-grujic, 2023-06-11:
> The base navigation engine INAV uses, was implemented over 5 years ago. That developer is no longer part of the INAV development team. So presently, there are no developers with the time or ability to implement your request.
....

I appreciate your straight honest answer! I hope, in time, this will be properly addressed.

sdellava, 2023-07-03:
The solutions of the RTH issues are related to the compass calibration that can be improved, and all automatic navigagion modes will be addvantaged.

I've already subminted  a suggestion on how to improve the compass calibration here: https://github.com/iNavFlight/inav/issues/8004

zoran-grujic, 2023-07-03:
Yes, but again people who do not have, or do not wish to have a compass will switch to BetaFlight. There should be compass-free RTH in INAV.

sdellava, 2023-07-03:
If you flight without using advanced INAV features (that requires
magnetometer), why you want to use INAV?

zoran-grujic, 2023-07-03:
Just because I wish to use advanced INAV features without magnetometer. To be precise, those that should or could work without of magnetometer.

This discussion is pointless. It is like "The Story of the Donkey and the Tiger".

sdellava, 2023-07-03:
Actually RTH without (a well-calibrated magnetometer) is not very reliable, especially for a quadrocopter.

The direction given by the GPS as heading may be wrong due to the wind causing the drone to drift. So every so many times the heading needs to be corrected by calculating the distance between expected and obtained heading.

This is obviously not a problem if you are flying a few tens of meters away from the home point, but if you are 4 km away, the problem can be very serious. 

This problem must be avoided because you cannot predict when an RTH will come into operation and therefore how much remaining battery capacity you can use.

In addition, a direct RTH supported by both BF and INAV is not reliable. If the LOS was lost because the drone ended up behind a building or a mountain, if it came back directly it would surely crash unless it had predicted the eventuality beforehand and defined the altitude at which to climb to avoid it, which nonrmally does not happen and in any case consumes a lot of aut.

So the safest thing is to retrace one's steps and then return to LOS in very little time, leaving the pilot to decide what to do next.

To turn back, it is critical that the path is accurate, and doing so without magnetormeter can lead to dozens of meters of error.

sdellava, 2023-07-03:
A magnetometer can be a very tiny device that can be added to almost any drone that own a GPS. So the real problem is that the magnetometer need to be calibrated any time you move from where you calibrated last time and this is very anoying

zoran-grujic, 2023-07-03:
From your reply I see you do not understand the function and performance of sensors. And YES this conversation is pointless with you. Grass is blue. If you check community requests in this repository you will find dozens of requests equivalent to this. So yes, it is useful feature to go home with out of compass. Check my replies to you nonsense below:

> Actually RTH without (a well-calibrated magnetometer) is not very reliable, especially for a quadrocopter.

Only advantage of a compass/magnetometer over GPS is potentially reliable heading determination at near zero ground speed. PERIOD.

> The direction given by the GPS as heading may be wrong due to the wind causing the drone to drift. So every so many times the heading needs to be corrected by calculating the distance between expected and obtained heading.

WRONG. The GPS derives heading according to previously obtained locations. This means its heading is always correct no matter how wind blows (if ground speed is not near zero). But in case of a strong wind this heading my not be the same as forward orientation of the flying craft. In that case the craft will not fly towards the home point but slightly off until it appears to be downwind of the home point (please do a simulation and you will confirm this happens as explained). Then, assuming the craft is faster than the wind, it will arrive exactly to the home point - just following heading given by GPS.

> This is obviously not a problem if you are flying a few tens of meters away from the home point, but if you are 4 km away, the problem can be very serious.

If the distance is 20-50 m then RTH is not needed. But indeed, in windy situation, when craft needs to slow down approaching the home point, the compass reading might become useful again, as GPS heading  looses accuracy. In that case landing anywhere in a circle of 50 m from the home point is acceptable (assuming pilot doesn't take over the control). If you are 4 km away the GPS heading is all you need to come home. Even if you do not go on the straight line it will finally, as described above, reach home after minor downwind excursion. If it works for winged crafts it will for multirotors equivalently. 

> This problem must be avoided because you cannot predict when an RTH will come into operation and therefore how much remaining battery capacity you can use.

There is no perfect system. It should attempt to do it and every meter it comes closer to home point will increase probability of recovery.

> In addition, a direct RTH supported by both BF and INAV is not reliable. 

Really? So what? Then you should drop it from INAV completely. That is your logic?

> If the LOS was lost because the drone ended up behind a building or a mountain, if it came back directly it would surely crash unless it had predicted the eventuality beforehand and defined the [...]

Normally how RTH is configured is to clime up to a predetermined altitude and then go back to the home point. This logic is sufficiently good to avoid such scenario. It is job of a pilot to set claiming pattern and altitude - AND to predict and avoid such situations. Your argument doesn't hold again.   

> So the safest thing is to retrace one's steps and then return to LOS in very little time, leaving the pilot to decide what to do next.

This sounds nice, and it is an advanced RTH option that should be available only with operational compass. This is a complex maneuver where GPS might fail to provide useful info in a sharp turn back to inverse trajectory... No compass no inverse trajectory. Fine for all. Just basic, go up and back to home. PERIOD.

> To turn back, it is critical that the path is accurate, and doing so without magnetormeter can lead to dozens of meters of error.

What is dozens of meters compared to 4 km? So what if it lands 50 m away? It will be recovered in most of cases, or pilot can take over the control.

zoran-grujic, 2023-07-03:
> A magnetometer can be a very tiny device that can be added to almost any drone that own a GPS. So the real problem is that the magnetometer need to be calibrated any time you move from where you calibrated last time and this is very anoying

Yes! Majority of pilots do not wish to deal with it. Waiting for a GPS lock is already annoyance, but must be. Magnetometer calibration is often forgotten. If un-calibrated it will probably lead to disorientation and will do more harm than without of it. Plus, it is not really required for RTH, might be helpful, but not crucial in 90% of cases.

Jetrell, 2023-12-09:
Fixed by #9387

closed 2024-03-20 by b14ckyy